alexa-pi/AlexaPi #215: AlexaPi.service don't start

1stTimerRick, 2017-04-18:
Output of the audio debugging script.

http://www.alsa-project.org/db/?f=e27bdc7d2c2d65f96075ed222877d10d89bd2b15
Your OS (including version) where you are running AlexaPi:

PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian

Your hardware platform and model you are running on:
Raspberry pi 3
Python release (python2 --version):
Python 2.7.9

Descrition of problem:
Does not start when I run sudo systemctl start AlexaPi.service
When I run sudo systemctl staus AlexaPi.service
I get Unit AlexaPi.service entered failed state.

Expected:

Problem-relevant config.yaml entries:

Steps to reproduce:
1.  Followed the Audio setup & debugging guidance at https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script 
Traceback (if applicable):

Additional info:
Please help.  I spent hours trying to figure out the audio piece with no luck.  It only to 15 minutes to setup and install everything!  Patience I must have it!

Thanks in advance for your support.

renekliment, 2017-04-24:
Look at the logs and post it here. https://github.com/alexa-pi/AlexaPi/wiki/Debugging#logs
Also, if you haven't set up PulseAudio, don't even ask here.

harimau99, 2017-05-17:
I have the same problem and i already went to the wiki as you shared. But it seems it didn't work. 
Plus, the debug seems listening very well.

Using Raspbian (Jessie Patch). 

Thanks in advance.
Regards.

EmerickH, 2017-06-10:
Hello @1stTimerRick and @harimau99,
We need more informations to help you...
Can you copy paste of the debug log?
```
sudo systemctl stop AlexaPi.service
/opt/AlexaPi/src/main.py -d
```
(Ctrl+C to exit when it finished loading, copy all the lines that appears, do it in SSH to directly copy/paste)

Emerick

EmerickH, 2017-06-14:
@1stTimerRick and @harimau99 You don't replied so I suppose that your problem is fixed.
If you still have the issue, reopen the issue and provide more information please.

Emerick

closed 2017-06-14 by EmerickH

harimau99, 2017-06-15:
For me, the problem already fixed. I already solve it with by looking at others fixing problem few days ago and i forgot to return here.
It seems that my boot was unable to accept and interpret on their own. Due to changes i made on the code. 
But right now, everything is working already. 

Thanks @EmerickH for the awesome support.

Thumbsup**