alexa-pi/AlexaPi #216: Bluetooth toggle

WeCreate180, 2017-04-19:
Alright, so I'm not sure if this is the right place to post but here I go.
Does anyone know of a way I could run a command with an Alexa skill? Like
Alexa, tell pi to toggle Bluetooth 
(To start my Bluetooth speaker thingy)
I tried belle an but the only part that works on that is the Alexa part. Not the wit.ai part.

Thanks in advanced!

WeCreate180, 2017-04-20:
Ayy!
I got it working! (Which is surprising because I don't know Python)
After reading the come and having some common sense, it looked exactly like Java. Just a few minor differences. (And bash)
Still need the wit.ai token.
I made it read the response that was stored in a variable (in plain text) response["_text"]
And did this:
if response["_text"] == "test":
   offline_speak("It works.")
   wit_ai_received = True

Sure enough, I say pie, then test, and bam. I get the output. I'll be posting this on my github so other people can clean it up and use it. I basically commented out everything that used to be there. It's quite messy.

maso27, 2017-04-20:
Cool! Google "beagleman Alexa" if you're looking for ideas, too. He forked the original sammachin Alexa code and integrated wit.ai for custom commands. It's also where I got the early pocketsphinx stuff for this project.

He passes the audio to wit.ai and Alexa simultaneously, then drops Alexa's response if wit.ai recognizes an intent.

WeCreate180, 2017-04-21:
Yes! I struggled with that dang pocketsphinx and I found that page by a mistake...
His wit.ai stopped working (or I wasn't doing it right) it always said my intents never existed. 
It would be cool to see something like this patched into the alexapi though...

WeCreate180, 2017-04-21:
I say that because Beagleman seems to stop recognizing if you don't speak to it after about 39 seconds. It's kind of annoying.

maso27, 2017-04-21:
I had troubles with the intents as well. Asked wit.ai about it and they said they had changed the way they did things. I didn't look into it any further after that

WeCreate180, 2017-04-21:
@maso27 this is off topic, but:
When I turn on my Bluetooth receiver on my pie (systemctl start bt_speaker) the Alexa stops outputting and the Bluetooth does not output.
Any ideas?

maso27, 2017-04-21:
Unfortunately I haven't had any experience with bluetooth receivers.
I don't actually know how AlexaPi would handle failovers when a device appears or disappears.

You might want to see if anyone on the gitter channel has any ideas too: https://gitter.im/alexa-pi/Lobby

WeCreate180, 2017-04-22:
Actually I fixed that. The only last problem is alexapi doesn't listen . It runs though.

WeCreate180, 2017-04-22:
@maso27 https://github.com/WeCreate180/beagleman-fixed
There you go, that's my fixed version of beagleman; I cleaned it up a little.

renekliment, 2017-10-08:
I'm not sure what this is. Closing.

Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment

jabulonboaz, 2018-02-16:
Please excuse my next to zero level of coding experience @maso27 and WeCreate180 but what is the code to do this:
How can i make Alexa with the top project or Google Assistant with the second project to locally trigger/execute a script .sh file i have in my Pi with a voice command  ?
E.g  “Alexa (or hey Google) turn on projector” (or e.g. "trigger fan") and then a script I have called say ‘projector_on.sh’ is executed in the Pi locally?
 And i guess i have to add any code for this to config.yaml right?