alexa-pi/AlexaPi #219: AlexaPi keeps crashing after couple of minutes

pejotigrek, 2017-04-22:
hi!

I have latest AlexaPi installed on latest OpenHABian. everything configured on RaspberryPi3, working fine via Ethernet and also via WiFi. right now I'm using a webcam as mic [waiting for usb-mic to come, just ordered it] and a regular speaker.

system works fine for couple of minutes, and it fails..
for testing, started Alexa with  **/usr/bin/python /opt/AlexaPi/src/main.py** and that's the output:

```
2017-04-22 19:27:24 INFO: Checking Internet Connection ...
2017-04-22 19:27:25 INFO: Connection OK
2017-04-22 19:27:25 INFO: AVS token: Requesting a new one
2017-04-22 19:27:26 INFO: AVS token: Obtained successfully
```

few questions and even a song sung by Alexa ;) later - and this is the failed output:

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 70, in thread
    _, buf = inp.read()
ALSAAudioError: Input/output error [hw:CARD=U0x46d0x825,DEV=0]
```

when restarted - again, it works for some time and fails.

because of last line - input/output error, I thought that maybe webcam is failing due to not enough power or something, so I've entered **max_usb_current=1** to config.txt, rebooted and... still nothing.

is it me, or it's something in the code? unfortunately I do not know anything about python so it's kinda difficult for me to crack this riddle ;)

cheers!

EmerickH, 2017-06-02:
Hello,
can you copy/paste the output of `arecord -L` and `arecord -l`?
hw:CARD=U0x46d0x825,DEV=0 seems strange...
Are you shure AlexaPi is not running twice?
Check before starting it manually:
`sudo service AlexaPi status`
and do:
`sudo service AlexaPi stop`
if you see that AlexaPi is running in background

Emerick

EmerickH, 2017-06-14:
@pejotigrek do you still have the issue?

pejotigrek, 2017-06-21:
@EmerickH hi!
excuse my long delay - I'm still kinda far away from the device [bouncing from place to place], hope to be back and have some time at home in couple of days, so I'll start another heavy tests - especially that I know my new usb microphone dongle arrived last week.

cheers!

pejotigrek, 2017-06-29:
@EmerickH hi! I'm back for a weekend, so I did try again few things.
arecord -L output is in [big_L.txt](https://github.com/alexa-pi/AlexaPi/files/1113034/big_L.txt) attachement, and arecord -l is in [small_L.txt](https://github.com/alexa-pi/AlexaPi/files/1113033/small_L.txt) file.

maybe it looks strange, because it's a webcam? ;)

I've tried my new usb-dongle-mic also, but it gives no device number :( just something like: 
`CARD=Device,DEV=0`
`    USB PnP Sound Device, USB Audio`
`    Hardware device with all software conversions`
in every line - so I guess it's not raspberrypi-compatible :(

but I guess it's all irrelevant, because now Alexa fails to start anyway..
that's what I get while checking status:
[image]

cheers!

renekliment, 2017-10-08:
We use PyAudio now. You can give that a shot.

Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment