alexa-pi/AlexaPi #220: AlexaPi crashing on start-up, but fine if I manually restart.

AKibbers, 2017-04-25:
Hi,

I've had a few issues with my AlexaPi, originally I had choppy audio so I went through the Running through Root wiki and managed to get it working. Now, however, when I start up my Pi I hear Alexa say 'Hello' then it crashes. I checked the journal and one line states;
ALSAAudioError: Device or resource busy.

I was wondering if anyone had any ideas what could be the issue here? I will say, once it's crashed, I can manually restart AlexaPi and it will work totally fine with no crashes, it's just on start up.

(So, in the past I have changed from vlc to sox in the config, and I changed to run through root. These could be my issues.)

EmerickH, 2017-06-14:
Hello,
You don't have to run AlexaPi as root!
Please copy/paste your audio settings in your `config.yaml` file and the output of `aplay -L`

Emerick

renekliment, 2017-10-08:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment