alexa-pi/AlexaPi #222: Custom Wake Word?

AntonioSwift77, 2017-04-28:
Hello Devs and Forum members.

I was wondering If you can set a custom wake word other than: "Alexa", "Echo", "Amazon" or "Computer"

I know with the set options for wake words are only those 4.
However, I know you can activate the listening function by pushing a button on the actual  Alexa/Echo that will cause the device to listen for a command/question.

keeping that in mind, can you make a small extension that will accept A custom wake word you set, which will trigger the equivalent of pushing the button on the device? (e.g. The custom wake word extension basically pushing the button for you instead of you physically pushing the button.)

I have little to no information about programming however I know this should be possible and is in the realm of possibility. I mean, it's just a voice activated virtual button push

I know that some people would really like this function to customize their alexapi. I would certainly like a more futuristic name for my alexapi other than those 4 (e.g. Ikaros* Nymph* or Iris)

I know that programming voice recognition software programs must be difficult but it shouldn't be too hard with a team right......?

Devs, Programmers and Modders I hope you take this request into consideration

Forum Members, I hope you guys will go crazy over this or at least take some interest to push this idea to the Programmers and Devs.
 
* Sora no Otoshimono reference, look it up

*btw If you haven't noticed already Iris is actually Siri backwards

pejotigrek, 2017-04-28:
hi,
I'm not a dev, but found this setting in /etc/opt/AlexaPi/config.yaml [if I remember the path correctly]. there's a line where you can wrote whatever you want :)

AntonioSwift77, 2017-04-28:
heyyyy thanks for responding! 
can you go more into detail into how that works and how to do that? 
I am kind of a noob because I just got my first Pi (RPi3) a couple of days ago...

I am afraid you might have to explain this to me from square one..... :(

AntonioSwift77, 2017-04-28:
Okay thank you so much! 
its nice to see that the community is friendly to noobs like me lol!

pejotigrek, 2017-04-28:
don't worry, I'm on that level too :D even if I use RPi since the first version [I've got pi + pi2 + 2 pi3 ;)] most of my "advanced" pi usage is tutorials and ctrl-c/ctrl-v from forums and pages like this ;)

I'm away from the device, so it's kinda freestyle based on my weak memory ;)
login to terminal, stop alexa and edit configuration, go:

```
sudo systemctl stop AlexaPi.service
sudo nano /etc/opt/AlexaPi/config.yaml
```

find the text fragment I wrote, using cursor-keys, change the name from alexa to something you like, press CTRL-X [it's "close"], then "y" [like "yes, overwrite" - because it asked too], then press enter to confirm name, and done :)

restart alexa with `sudo systemctl start AlexaPi.service` and enjoy custom wake-word :)

[if I'm being wrong, please someone smarter correct me ;)]

AntonioSwift77, 2017-04-28:
ummmmm......... 
when I pasted the
 sudo systemctl stop AlexaPi.service
sudo nano /etc/opt/AlexaPi/config.yaml
into my terminal it just brought me a text editor...... what do I do?

I also seem to be having an issue with the wake word working itself.... its not working.

Help?

maso27, 2017-04-29:
Use the text editor to edit the file. You are editing _config.yaml_, located in _/etc/opt/AlexaPi_.

That's where all the settings are for AlexaPi. There are suggestions along the bottom of the screen for functions, like _Ctrl-W_ to search, _Ctrl-X_ to exit, etc.

My suggestion would be to use _Ctrl-W_ to search for "alexa". That would lead you to the wake word for the program.  Change it to something of your choosing, then exit with _Ctrl-X_.
"Yes" you want to save the file.
"Enter" to save it with the same name.

Then you can reboot your Pi (_sudo reboot_) or else just start AlexaPi up again (_sudo systemctl start AlexaPi.service_) and see how it works.

closed 2017-05-20 by renekliment

tkamaal, 2017-10-11:
@AntonioSwift77 Were you successful in customizing the wake word? Please let me know, because this will provide direction for my project.

HLowe4, 2017-11-16:
So I am thinking of buying the Echo Show and was curious of the same thing is it possible to customize the wake word as well as change Alexas voice to lets say a nonspecific British Male AI...?

onyxcode, 2020-01-29:
@HLowe4 Dunno

Beastlynoob, 2020-06-18:
Hello, I am also a noob with pi. I am creating a smart/magic mirror using a monitor and a raspberry pie 4, and I now wanted to have my Alexa behind my smart mirror. I was hoping to customize my Alexa so that it's wake word would be "mirror mirror on the wall". I noticed that they were instructions on how to do this above, but I was wondering how I connect my Alexa to my raspberry pie. Thank you Github community!