alexa-pi/AlexaPi #223: Process: 1200 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)

rorobig, 2017-04-29:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie

**Your hardware platform and model you are running on:**
Raspberry pi 3 model B

**Python release (`python2 --version`):**
```
python 2.7.9
```

**Description of problem:**
```
My mic works fine installation went fine but when i run this "sudo systemctl status AlexaPi.service" to see logs i get this error message 
```
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Sat 2017-04-29 15:21:02 UTC; 25min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 1200 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 1200 (code=exited, status=1/FAILURE)

Apr 29 15:21:02 raspberrypi python[1200]: surround50:CARD=U0x46d0x825,DEV=0
Apr 29 15:21:02 raspberrypi python[1200]: surround51:CARD=U0x46d0x825,DEV=0
Apr 29 15:21:02 raspberrypi python[1200]: surround71:CARD=U0x46d0x825,DEV=0
Apr 29 15:21:02 raspberrypi python[1200]: iec958:CARD=U0x46d0x825,DEV=0
Apr 29 15:21:02 raspberrypi python[1200]: dmix:CARD=U0x46d0x825,DEV=0
Apr 29 15:21:02 raspberrypi python[1200]: dsnoop:CARD=U0x46d0x825,DEV=0
Apr 29 15:21:02 raspberrypi python[1200]: hw:CARD=U0x46d0x825,DEV=0
Apr 29 15:21:02 raspberrypi python[1200]: plughw:CARD=U0x46d0x825,DEV=0
Apr 29 15:21:02 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
Apr 29 15:21:02 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.

**Expected:**
```
So when i use the wakeword alexa nothings happens because obviously the service is not running.
```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

honeybadger01, 2017-05-05:
I am having the same problem, but in my case, the error is in process: 2044

rorobig, 2017-05-06:
im using this now u can try it 
https://github.com/alexa/alexa-avs-sample-app <https://github.com/alexa/alexa-avs-sample-app>

honeybadger01, 2017-05-06:
Thanks mate! I'l give it a try to this one.

renekliment, 2017-05-20:
Check out https://github.com/alexa-pi/AlexaPi/wiki/Debugging#logs ... after that, you'll see the message that you have to set up your `input_device` properly.

@honeybadger01 
> I am having the same problem, but in my case, the error is in process: 2044
> UPDATE: Every time I reboot the RPi, I have a different process error code... :-(

This is irrelevant.

closed 2017-05-20 by renekliment