alexa-pi/AlexaPi #226: USB Ear-Microphone not working

Aviatior, 2017-05-02:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
```

**Your hardware platform and model you are running on:**
```

```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
```

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

EmerickH, 2017-05-30:
Hello,
if you want to use pulseaudio you don't have to use hw:... but:
```
  input_device: "pulse"
  allow_unlisted_input_device: false
  playback_handler: "sox" #or vlc
  output: "pulseaudio" #or just pulse if you are using vlc
  output_device: "default" 
```
And if your device is not selected by default, use pavucontrol to select it (you must have a graphical interface or use pactl command instead)
```
sudo apt-get install pavucontrol
pavucontrol
```

Emerick

EmerickH, 2017-06-14:
@Aviatior have you solved your issue?

renekliment, 2017-10-08:
We use PyAudio now. Try that.

Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment