alexa-pi/AlexaPi #228: Could not import runpy module

ryjo1026, 2017-05-06:
**Your OS (including version) where you are running AlexaPi:**
```
Raspian Jessie Lite April 2017
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi 2 Model B
```

**Python release (`python2 --version`):**
```
Python 2.7.9
```

**Description of problem:**
```
When installing through the shell script I get the error "Could not import runpy module" then a traceback follows which ends in "FileNotFoundError: [Errno 2] No such file or directory" (full traceback included below). I attempted first on Raspian Jessie with Pixel and then did a clean install of Raspian Jessie Lite and followed installation instruction step-by-step. Further up in the output are two groups of four "sh: 0: getcwd() failed: No such file or directory" errors; I'm pretty sure this is the root of the error. There are no other errors that I can see but its possible I missed them. 
```

**Traceback (if applicable):**
```
Could not import runpy module
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2222, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 2164, in _find_spec
  File "<frozen importlib._bootstrap>", line 1940, in find_spec
  File "<frozen importlib._bootstrap>", line 1911, in _get_spec
  File "<frozen importlib._bootstrap>", line 1879, in _path_importer_cache
FileNotFoundError: [Errno 2] No such file or directory
```

**Additional info:**
```
AlexaPi.service status:

AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Sat 2017-05-06 19:17:18 UTC; 54min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 703 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 703 (code=exited, status=1/FAILURE)

May 06 19:17:14 raspberrypi systemd[1]: Started Alexa client for all your devices.
May 06 19:17:18 raspberrypi python[703]: CRITICAL: Can not find configuration file. Exiting...
May 06 19:17:18 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
May 06 19:17:18 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
```

renekliment, 2017-05-07:
Hey. This seems like a broken python install.
Check out https://github.com/rg3/youtube-dl/issues/523#issuecomment-10780325

EmerickH, 2017-06-14:
@ryjo1026 have you solved your issue?

renekliment, 2017-10-08:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment