alexa-pi/AlexaPi #229: ALexaPi error, speaker and mic on USB

claudioita, 2017-05-07:
I receive this error
```

pi@raspberrypi:~ $ sudo systemctl start AlexaPi.service
pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Sun 2017-05-07 17:24:36 CEST; 5s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 2372 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILU                                                                         RE)
 Main PID: 2372 (code=exited, status=1/FAILURE)

May 07 17:24:36 raspberrypi python[2372]: surround50:CARD=USB,DEV=0
May 07 17:24:36 raspberrypi python[2372]: surround51:CARD=USB,DEV=0
May 07 17:24:36 raspberrypi python[2372]: surround71:CARD=USB,DEV=0
May 07 17:24:36 raspberrypi python[2372]: iec958:CARD=USB,DEV=0
May 07 17:24:36 raspberrypi python[2372]: dmix:CARD=USB,DEV=0
May 07 17:24:36 raspberrypi python[2372]: dsnoop:CARD=USB,DEV=0
May 07 17:24:36 raspberrypi python[2372]: hw:CARD=USB,DEV=0
May 07 17:24:36 raspberrypi python[2372]: plughw:CARD=USB,DEV=0
May 07 17:24:36 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status...LURE
May 07 17:24:36 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.
pi@raspberrypi:~ $ ^C
pi@raspberrypi:~ $

```

**Your OS (including version) where you are running AlexaPi:**
```
```raspberry pi3 with hassbian

**Your hardware platform and model you are running on:**
``` raspberry pi3 with Jabra 510 USB speaker/mic

```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
``` Receive the above error at start. The Google Home on pi3 (is installed in env environment as per google instructions is working fine.
I also changed this in the  config file I put tty/USB0 ... not sure is correct, now I also do not remember how was the original file
/etc/opt/AlexaPi/config.yaml

``` serial:
    port: "/dev/ttyUSB0"
    baudrate: 9600
    messages:
      trigger: ""
      playback_start: "g"
      playback_end: "x"
      recording_start: ""
      recording_end: ""
      processing_start: ""
      processing_end: ""
      failure: ""
      success: ""

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

claudioita, 2017-05-07:
I guess the Google Assistant did not install so I did this
cd/opt
python -m pip install google-assistant-sdk[samples]

hope is not too wrong

maso27, 2017-05-08:
The _serial_ portion of config.yaml is only used when you select _serial_ as the platform.  You should be using _raspberrypi_ instead.  So the application won't even look at that setting.

The initial settings of config.yaml can be found in _/opt/AlexaPi/src/config.template.yaml_ so you can always look there for a way back to where you started.

As for the failure, the report has been shortened, so it's hard to tell what the error is.  You might find more clues if you use the _-l_ option as it suggests.

Meanwhile I'm suspecting that you don't have the audio devices configured correctly.  If I'm right, try using the `aplay -l` command to list what your output options are, and `arecord -l` to list your input devices.  Copy the best fit (probably the one that says _plughw_) into config.yaml under _input_device_ and _output_device_.

For example I'd suggest `output_device: 'plughw:CARD=USB,DEV=0'` based on what I see above.

See if that helps.

clanmcclard, 2017-05-17:
I'm having a similar issue but I don't seem to be able to save changes on the config.yaml file when I am editing it from the editor app from the RPI desktop.   Seems like a permissions issue but I am signed in as 'pi'.  Any suggestions for how I can successfully edit config.yaml?

maso27, 2017-05-17:
I'm not so familiar with the RPi desktop, but you can edit it from the terminal.  
Open a terminal, then use `sudo nano /etc/opt/AlexaPi/config.yaml`.
This will edit it as root, with appropriate permissions.

clanmcclard, 2017-05-19:
Thanks, that worked.  I thought I had done that before but I kept getting a
new file opening like it wasn't there.

Pat

EmerickH, 2017-05-30:
If you solved your problem, can you close this issue please?

Emerick

EmerickH, 2017-06-10:
Hello, just go on the webpage of the issue ([here](https://github.com/alexa-pi/AlexaPi/issues/229))
and click on **Close Issue** on the bottom of the page.

Emerick

closed 2017-06-14 by EmerickH

guigobass, 2018-02-12:
Hello...
My problem is. I say "Alexa" and she answer yes? But after this she not answer more...
I execute systemctl status AlexaPi.service and i get error 403.

Any solution?

jabulonboaz, 2018-02-15:
@guigobass https://github.com/alexa-pi/AlexaPi/issues/376