alexa-pi/AlexaPi #230: AlexaPi seems to kill googlevoicehat until reboot

rickwookie, 2017-05-07:
Trying to get AlexaPi to work with the new googlevoicehat, and I can hear the audio output through the 'speaker if I run:
`sudo sox /opt/AlexaPi/src/resources/hello.mp3 -t alsa`
But whenever I start AlexaPi, I hear no "Hello" message, and any further attempt to play sound results in silence, but with no errors reported. I can only restore sound by rebooting.

`sudo /opt/AlexaPi/src/main.py -d`
outputs:
2017-05-07 19:15:24 DEBUG: Setting up playback handler: VlcHandler
2017-05-07 19:15:24 INFO: Checking Internet Connection ...
2017-05-07 19:15:24 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-05-07 19:15:25 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2017-05-07 19:15:25 INFO: Connection OK
2017-05-07 19:15:25 INFO: AVS token: Requesting a new one
2017-05-07 19:15:25 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-05-07 19:15:26 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 978
2017-05-07 19:15:26 INFO: AVS token: Obtained successfully
2017-05-07 19:15:26 DEBUG: Stopping audio play
2017-05-07 19:15:26 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-05-07 19:15:26 DEBUG: Player State: State.Opening
2017-05-07 19:15:26 DEBUG: Player State: State.Playing
2017-05-07 19:15:26 DEBUG: Started play.
2017-05-07 19:15:29 DEBUG: Player State: State.Ended
2017-05-07 19:15:29 DEBUG: Finished play.

In `/etc/opt/AlexaPi/config.yaml` I have:
input_device: "plughw:CARD=sndrpigooglevoi,DEV=0"
output: "alsa"
output_device: "plughw:CARD=sndrpigooglevoi,DEV=0"

`aplay -L` gives me:
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
softvol
micboost
default
sysdefault:CARD=sndrpigooglevoi
    snd_rpi_googlevoicehat_soundcar,
    Default Audio Device
dmix:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct sample mixing device
dsnoop:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct sample snooping device
hw:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct hardware device without any conversions
plughw:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Hardware device with all software conversions

maso27, 2017-05-08:
Two thoughts:
1.) Have you used `arecord -L` to make sure that's how the input device needs to be configured?  It may not come up with the same device information.
2.) You've got PulseAudio on your system.  That has historically caused various issues with the audio configuration for AlexaPi.  You could try running PulseAudio in system-wide mode, as outlined on [this page](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging).  Always the easiest, though, is to run Raspbian Lite instead of the Desktop version.

I'm interested to hear how googlevoicehat works! Report back how it goes.

rickwookie, 2017-05-08:
1.) Yes, I initially ran it with OOTB setting for AlexaPi and I was getting an error like:
CRITICAL: Your input_device 'plughw:CARD=Device,DEV=0' is invalid. Use one of the following:
null
pulse
softvol
micboost
default
sysdefault:CARD=sndrpigooglevoi
dmix:CARD=sndrpigooglevoi,DEV=0
dsnoop:CARD=sndrpigooglevoi,DEV=0
hw:CARD=sndrpigooglevoi,DEV=0
plughw:CARD=sndrpigooglevoi,DEV=0

Hence why I looked at /etc/opt/AlexaPi/config.yaml and saw the reference to using arecord -L, which gave me:
`arecord -L`
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
softvol
micboost
default
sysdefault:CARD=sndrpigooglevoi
    snd_rpi_googlevoicehat_soundcar,
    Default Audio Device
dmix:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct sample mixing device
dsnoop:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct sample snooping device
hw:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct hardware device without any conversions
plughw:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Hardware device with all software conversions

That's why I changed the config line from `input_device: "plughw:CARD=Device,DEV=0"` to `input_device: "plughw:CARD=sndrpigooglevoi,DEV=0"`
Out of interest, why is the default config like that, and not `input_device: "default"` (which also seems to execute without errors) the same as for the output_device entry?

Anyway, perhaps you could grab the RPi image used with the kit from https://dl.google.com/dl/aiyprojects/voice/aiyprojects-latest.img.xz (it's Raspbian based apparently) and see if you can find anything obvious wrong? You'd need to modify /boot/config.txt to comment out the dtoverlay=googlevoicehat-soundcard (and uncomment dtparam=audio=on if you're using the inbuilt audio output)

I did of course disable the voice-recognizer.service from the kit software thinking that it was an audio hardware sharing issue, but that's not it (as far as I can see). 

2.) Pulse is there because the kit image includes the Pixel Desktop I think, but yes I did follow that guide and set up running PulseAudio in system-wide mode, but no luck.

I also added root to the audio group since I expect AlexaPi needs to run as root, but tbh I see no error output from AlexaPi even without having root in the audio group.

maso27, 2017-05-08:
You're running this in an untested configuration.  

The default settings within _config.yaml_ work when users start with vanilla Raspbian and plug speakers into the headphone jack.  That's why the _input_device_ line reads as it does.  I'm not sure what has been done for the aiyprojects image, so I can't speak to how that works.

But as far as adding _root_ to the groups is concerned--that shouldn't be necessary.  AlexaPi creates a new user _alexapi_ at install and runs things from there when it's running as a service.  Starting it at the command line with `sudo` is different, but there shouldn't be any issues.

So are you trying to run both the google project and the alexa one at the same time?  I think you'll have to prepare yourself for some trial-and-error.

First off, it looks like your playback test is using the _sox_ handler.  You could try changing config.yaml from _vlc_ to _sox_ and see if it works any better.

You could see if combinations of that and engaging _pulseaudio_ instead of _alsa_ works at all.  If you're getting no love there, you could try 'running as root' as outlined in the wiki as well.

rickwookie, 2017-05-08:
So on a vanilla Raspbian with no audio input device, the config line `input_device: "plughw:CARD=Device,DEV=0` doesn't throw an error?

I'm not trying to run the google project at the same time (for now), hence why I've disabled the voice-recognizer.service from startup. If I can get AlexaPi to work, then I'll probably look at https://github.com/xtools-at/AssistantPi as a way to run both.

Is there a command I can run to test the vlc handler independently?
Setting `playback_handler: "sox"` in the config gets me:
2017-05-08 20:59:07 DEBUG: Setting up playback handler: SoxHandler
2017-05-08 20:59:07 INFO: Checking Internet Connection ...
2017-05-08 20:59:07 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-05-08 20:59:07 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2017-05-08 20:59:07 INFO: Connection OK
2017-05-08 20:59:07 INFO: AVS token: Requesting a new one
2017-05-08 20:59:07 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-05-08 20:59:09 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 978
2017-05-08 20:59:09 INFO: AVS token: Obtained successfully
2017-05-08 20:59:09 DEBUG: Stopping audio play
2017-05-08 20:59:09 DEBUG: Stopped play. No SoX process to stop
2017-05-08 20:59:09 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-05-08 20:59:09 DEBUG: Started play. sox -q /opt/AlexaPi/src/resources/hello.mp3 -t alsa default vol -6 dB pad 0 0
2017-05-08 20:59:12 DEBUG: Finished play.
^C2017-05-08 20:59:28 DEBUG: Cleaning up playback handler
2017-05-08 20:59:28 DEBUG: Stopped play. [Errno 3] No such process

But I still hear nothing and the sound is dead again requiring a reboot.

After a reboot, I can run the command exactly the same as in the AlexaPi debug output: `sox -q /opt/AlexaPi/src/resources/hello.mp3 -t alsa default vol -6 dB pad 0 0` and I hear the sound "Hello". Then run `sudo /opt/AlexaPi/src/main.py -d`, exit with ^C, try the same sox test again, no sound!

Something is happening to the soundcard/driver before it tries to output. I thought maybe it had something to do with the input side getting incorrectly initialized and breaking the soundcard driver, but even if I set the config to `input_device: "null"`, I get the same result - so sound output after running AlexaPi.

Noiprox6, 2017-05-28:
I had the same problem using the Google AIY Kit and AlexaPi.  I finally tracked it down to this:

https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#pulseaudio

Specifically the  "External DAC Support" section.  What wasn't explained by Google with the kit was that the Google Voice Hat uses an External DAC which uses GPIO pins to communicate with the Pi.  So when AlexaPi starts it attempts to setup optional buttons on GPIO pins and conflicts (kills) the Voice Hat audio.

I did as explained in that artile and used the "dummy" platform (device: "dummy" in the config file) instead of "raspberrypi" as you would expect.  Now I can run both AlexaPi and the original AIY project at the same time and both work perfectly.  AlexaPi even uses the Speaker from the AIY kit however I didn't get it working with the Voice Hat MIC, I used a usb MIC and Alexa will respond to her name and Google to the button press from the AIY Kit.

Can't remember what I used for audio devices in the config files but I think defaults should work once you change the AlexaPi platform.

rickwookie, 2017-05-29:
Brilliant work Noiprox6, that makes perfect sense. I didn't even consider that the raspberrypi setting in the config was to use GPIO.

I wonder why the mic wouldn't work though once you use the dummy interface?

Noiprox6, 2017-05-29:
I did end up getting the mic working, I hadn't been bothered by it as I also had a usb mic and when I attached it Alexapi would use it and google would use the voicehat mic.  For fun (and because I broke the google aiy install trying to get hotwords working) I wiped the sd card and reinstalled the aiy kit base sd card image and started from scratch.  I did things in this order:

-Flash sd card
-configure aiy kit as usual until working
-"git pull" in the aiy kit install folder (voice-recognizer-raspi) to update to the current master branch state
-"git fetch" to bring in pull request #64 (uses the Google Assistant Library to enable hotwords "ok google" and "hey google")
-switched to branch 64
-ran install script for dependencies and then installed the Google Assistant Library

So at this point I have the aiy kit working with both the button and by just saying "ok google" at any time (always listening)

Then I installed AlexaPi as usual
-after install and configuration I changed the platform to "dummy" in config file
-tested both Alexa and Google and both are working using the voicehat speaker but Alexa still needs the usb mic to hear me
-couldn't get the mic to work using the correct hardware name in the config file (config.yaml)

Here is the list I tried:

default, used the usb mic no problem - input_device: "plughw:CARD=Device,DEV=0"
what should work for the voicehat mic - input_device: "plughw:CARD=sndrpigooglevoi,DEV=0"
what actually worked (wasn't even listed as an option - input_device: "default"

So now I'm good, both Google and Alexa listen all the time and respond to their correct hotwords, "Alexa" or "ok google"

My only problem (and it's very annoying) is Alexa is at like 200% volume when she talks, even her "hello" on boot is super loud.  Google volume is fine but Alexa doesn't seem to respond to any volume setting changes, OS or config file.

Once I sort that little issue out I'll be all set with my AlexaGooglePi :)

drigz, 2017-05-30:
> My only problem (and it's very annoying) is Alexa is at like 200% volume when she talks, even her "hello" on boot is super loud. Google volume is fine but Alexa doesn't seem to respond to any volume setting changes, OS or config file.

Does it work if you use these settings?

```
input_device: "plug:micboost"
output_device: "plug:softvol"
```

Noiprox6, 2017-05-31:
Hello Drigz,

Thank you, I'm new to all of this and had no idea those were even options!  I tried those and the AlexaPi service failed but checking the service status also listed all available options:

softvol
micboost
default
sysdefault:CARD=sndrpigooglevoi
dmix:CARD=sndrpigooglevoi,DEV=0
dsnoop:CARD=sndrpigooglevoi,DEV=0
hw:CARD=sndrpigooglevoi,DEV=0
plughw:CARD=sndrpigooglevoi,DEV=0

So I tried the following in the  /etc/opt/AlexaPi/config.yaml file:

input_device: "default"
playback_handler: "vlc"
output: "alsa"
output_device: "softvol"

The "softvol" wouldn't work with the "plug:" in front of the output device but it worked as above.  Also the input_device wouldn't work with any of the listed options other than "default" including "micboost" with and without the "plug:" in front of it but "default" seems to work great.  

Now it's all working, Google and Alexa are both responding to changes in volume using alsamixer command!  Google is slightly lower volume compared to Alexa which is strange and the start volume in the config.yaml file for Alexa doesn't change anything but as I adjust the alsamixer they both go up and down, still a bit out of sync for volume level.  

Is there maybe a way to adjust the base volume of Google AIY up a bit so it lines up with Alexa's base volume? 

Thanks again for the help, troubleshooting while Alexa yelled at me was getting bad after two days ;)

EmerickH, 2017-06-14:
I think solved your issue?
If I am wrong, you can reopen the issue.

Emerick

closed 2017-06-14 by EmerickH

Noiprox6, 2017-06-14:
Yes, issue is completely resolved, thanks again!

shadowscott, 2018-03-14:
@Noiprox6 Can you help? I'm trying to do something similar. I installed GassistPi to get Google Assistant working on my Pi without the MagPi image which works fine (despite not realizing it doesn't have all the play features I was hoping for). However I'm not sure how you got AlexaPi to use the AIY Hat installation. Did you just install it and then change information in the config.yaml for the settings to be 'default' to use the built in speaker?