alexa-pi/AlexaPi #231: Can't play Amazon Music on this device?

Colin1964, 2017-05-08:
Have managed to get Alexa-AVS-Sample-App working fine on my RPI3. Just can't seem to get Amazon music working. My Prime and Music subscription shows up in the PC based config utility of the app and if I click on an album or track in PC app the output is loud and clear from the Pi. When I try "Alexa, play some music from prime music" I get the response "This is not possible on this device"

Alexa version and instructions were from here:
https://github.com/alexa/alexa-avs-sample-app/wiki/Raspberry-Pi

renekliment, 2017-05-08:
Hey.

1. AlexaPi is a different thing from Amazon's sample app. This is not the right repo.

2. In this case, it is however the same for both apps - https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#does-alexapi-support-amazon-music

closed 2017-05-08 by renekliment

Gru-Preet, 2019-07-28:
how to play amazon music on alexapi