alexa-pi/AlexaPi #232: AlexaPi crash immediatelly after Alexa's "yes" response.

aki29, 2017-05-12:
Hi 
I run the latest AlexaPi on Action S500 RoseApplePi platform with debian OS.
But it crash after Alexa say "yes" (I say "Hello" to Alexa)
Any suggestion please.

root@roseapple:/opt/AlexaPi/src# lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 8.7 (jessie)
Release:	8.7
Codename:	jessie

root@roseapple:/opt/AlexaPi/src# ./main.py 
2017-05-12 09:51:44 INFO: Checking Internet Connection ...
2017-05-12 09:51:45 INFO: Connection OK
2017-05-12 09:51:45 INFO: AVS token: Requesting a new one
2017-05-12 09:51:46 INFO: AVS token: Obtained successfully
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "./main.py", line 542, in trigger_process
    alexa_speech_recognizer(audio_stream)
  File "./main.py", line 347, in alexa_speech_recognizer
    resp = requests.post(url, headers=headers, data=data)
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 110, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 56, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/adapters.py", line 443, in send
    for i in request.body:
  File "./main.py", line 326, in alexa_speech_recognizer_generate_data
    for audio_chunk in audio:
  **File "/opt/AlexaPi/src/alexapi/capture.py", line 105, in _listen
    isSpeech = self._vad.is_speech(data, self.VAD_SAMPLERATE)**
  File "/usr/local/lib/python2.7/dist-packages/webrtcvad.py", line 27, in is_speech
    return _webrtcvad.process(self._vad, sample_rate, buf, length)
Error: Error while processing frame

rsuhendro, 2017-05-12:
I also experience similar problem...it crashed after "Yes". I'm running on latest raspbian jessie lite. 

2017-05-12 10:49:48 DEBUG: Start recording
2017-05-12 10:49:48 DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com
2017-05-12 10:49:50 DEBUG: Start sending speech to Alexa Voice Service
2017-05-12 10:50:53 DEBUG: End recording
2017-05-12 10:50:53 DEBUG: Finished sending speech to Alexa Voice Service
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/AlexaPi/src/main.py", line 542, in trigger_process
    alexa_speech_recognizer(audio_stream)
  File "/opt/AlexaPi/src/main.py", line 347, in alexa_speech_recognizer
    resp = requests.post(url, headers=headers, data=data)
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 112, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 58, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 518, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 639, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/adapters.py", line 469, in send
    r = low_conn.getresponse(buffering=True)
  File "/usr/lib/python2.7/httplib.py", line 1111, in getresponse
    response.begin()
  File "/usr/lib/python2.7/httplib.py", line 444, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python2.7/httplib.py", line 408, in _read_status
    raise BadStatusLine(line)
BadStatusLine: ''

rsuhendro, 2017-05-14:
The exception crash messages were gone after rebooting my wifi access point...weird. 
But I then stuck with 'null response':
2017-05-13 10:22:25 DEBUG: Start sending speech to Alexa Voice Service
2017-05-13 10:22:33 DEBUG: End recording
2017-05-13 10:22:33 DEBUG: Finished sending speech to Alexa Voice Service
2017-05-13 10:22:34 DEBUG: Processing Request Response...
2017-05-13 10:22:34 DEBUG: Request Response is null (This is OKAY!)

The system audio works fine because I've checked it with arecord and then aplay.  However checking AlexaPi sound record 'recording.wav' revealed hissing sound or no sound at all.

Last resort I setup PulseAudio as described in the documentation and replaced 'input_device: "plughw:CARD=Device,DEV=0"' with 'input_device: pulse'. And tada, it worked. I got the response from Alexa. 

Just out of curiosity, why I have to use PulseAudio? I use raspbian jessie lite no GUI. 

Thanks.

EmerickH, 2017-06-14:
@rsuhendro I don't know if in `plughw:CARD=Device,DEV=0` you have replaced **Device** by your device name (do `aplay -L` to see the names).
But I advise you to keep pulse because Alsa is very buggy with AlexaPi (the library has been changed in the `dev` branch but not actually in the `master` one)

Emerick

closed 2017-06-14 by EmerickH