alexa-pi/AlexaPi #236: Alexa Default Location

garc7481, 2017-05-17:
Having a problem with AlexaPi. When asking for weather or local things to my location. Alexa keeps defaulting to Seattle. I changed my location on the google developer site. But it keeps on defaulting. Any idea why. Geo location it's not being capture by Alexa. Are there any setting on the config files or the code that we need to edit.

maso27, 2017-05-17:
You have to use the Alexa app to change your location.  Install it on an ios or android device, then change you location.
This is how Amazon manages it, and to my knowledge it's the same way you do it on an echo.

renekliment, 2017-05-20:
It's also covered in our FAQ: https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#how-do-i-change-my-location-time-zone-units-

closed 2017-05-20 by renekliment