alexa-pi/AlexaPi #237: Debug Input device

teddylambert, 2017-05-20:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**
[image]

**Your OS (including version) where you are running AlexaPi:**
```
Linux Raspbian 3-02-2017 release
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi 2B with JBL speaker and Playstation eye camera/mic
```

**Python release (`python2 --version`):**
```
2.7.9
```

**Description of problem:**
```
I ran the debug, and followed the instructions to copy the plughw into the "input-device" field. Yet I still don't have audio, and when I re-run the debug (with the new config) it gives the same error. Any ideas?
```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

renekliment, 2017-05-20:
Do what the script tells you. Set the `input_device` to one from the list - probably the last one.

teddylambert, 2017-05-20:
That's what I originally did (I pasted the last one), and doing so threw out the error above. Maybe I'm copy/pasting wrong?

teddylambert, 2017-05-20:
Update:
When I tried switching ``` input_device:``` to one of the other values (I used ```surround40:CARD=CameraB409241,DEV=0``` as a test value), when running ```/opt/AlexaPi/src/main.py -d``` I still get this-
[image]

This is my ```config.yaml```, is there anything that I'm missing here?-
[image]

probststefan, 2017-05-22:
Try it with: `input_device: "plughw:CARD=CameraB409241,DEV=0"`

That worked for me

closed 2017-05-28 by teddylambert

harimau99, 2017-05-28:
But i thought the solution was either way around.. But changing to respective listener hardware plugin. or input devices..