alexa-pi/AlexaPi #239: alexa-pi crash(Segmentation fault) when player(vlc) status change

frank1546, 2017-05-20:
**Python release (`python2 --version`):**
Python 2.7.9

**Description of problem:**
There is very strange issue and very easy to reproduce.
When Alexa-pi start playing hello.mp3, vlc(libvlc.so) will call resisted callback function(see vlc.py event_attach function). But it always crash(Segmentation fault) after calling(crash point at event.c listener_cached->pf_callback( p_event, listener_cached->p_user_data );)

log show as below
/usr/lib/python2.7/webrtcvad.py:1: UserWarning: Module webrtcvad was already imported from /usr/lib/python2.7/webrtcvad.py, but /usr/lib/python2.7/site-packages/webrtcvad-2.0.10-py2.7-linux2.egg is being added to sys.path
  import pkg_resources
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/lib/python2.7/site-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone
-allphone_ci		no		no
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/lib/python2.7/site-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict
-feat			1s_c_d_dd	1s_c_d_dd
-featparams
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7.000000e-29
-fwdtree		yes		yes
-hmm					/usr/lib/python2.7/site-packages/pocketsphinx/model/en-us
-input_endian		little		little
-jsgf
-keyphrase				one
-kws
-kws_delay		10		10
-kws_plp		1e-1		1.000000e-01
-kws_threshold		1		1.000000e-10
-latsize		5000		5000
-lda
-ldadim			0		0
-lifter			0		22
-lm
-lmctl
-lmname
-logbase		1.0001		1.000100e+00
-logfn
-logspec		no		no
-lowerf			133.33334	1.300000e+02
-lpbeam			1e-40		1.000000e-40
-lponlybeam		7e-29		7.000000e-29
-lw			6.5		6.500000e+00
-maxhmmpf		30000		30000
-maxwpf			-1		-1
-mdef
-mean
-mfclogdir
-min_endfr		0		0
-mixw
-mixwfloor		0.0000001	1.000000e-07
-mllr
-mmap			yes		yes
-ncep			13		13
-nfft			512		512
-nfilt			40		25
-nwpen			1.0		1.000000e+00
-pbeam			1e-48		1.000000e-48
-pip			1.0		1.000000e+00
-pl_beam		1e-10		1.000000e-10
-pl_pbeam		1e-10		1.000000e-10
-pl_pip			1.0		1.000000e+00
-pl_weight		3.0		3.000000e+00
-pl_window		5		5
-rawlogdir
-remove_dc		no		no
-remove_noise		yes		yes
-remove_silence		yes		yes
-round_filters		yes		yes
-samprate		16000		1.600000e+04
-seed			-1		-1
-sendump
-senlogdir
-senmgau
-silprob		0.005		5.000000e-03
-smoothspec		no		no
-svspec					0-12/13-25/26-38
-tmat
-tmatfloor		0.0001		1.000000e-04
-topn			4		4
-topn_beam		0		0
-toprule
-transform		legacy		dct
-unit_area		yes		yes
-upperf			6855.4976	6.800000e+03
-uw			1.0		1.000000e+00
-vad_postspeech		50		50
-vad_prespeech		20		20
-vad_startspeech	10		10
-vad_threshold		2.0		2.000000e+00
-var
-varfloor		0.0001		1.000000e-04
-varnorm		no		no
-verbose		no		no
-warp_params
-warp_type		inverse_linear	inverse_linear
-wbeam			7e-29		7.000000e-29
-wip			0.65		6.500000e-01
-wlen			0.025625	2.562500e-02

INFO: feat.c(715): Initializing feature stream to type: '1s_c_d_dd', ceplen=13, CMN='batch', VARNORM='no', AGC='none'
INFO: acmod.c(166): Using subvector specification 0-12/13-25/26-38
INFO: mdef.c(518): Reading model definition: /usr/lib/python2.7/site-packages/pocketsphinx/model/en-us/mdef
INFO: mdef.c(531): Found byte-order mark BMDF, assuming this is a binary mdef file
INFO: bin_mdef.c(336): Reading binary model definition: /usr/lib/python2.7/site-packages/pocketsphinx/model/en-us/mdef
INFO: bin_mdef.c(516): 42 CI-phone, 137053 CD-phone, 3 emitstate/phone, 126 CI-sen, 5126 Sen, 29324 Sen-Seq
INFO: tmat.c(149): Reading HMM transition probability matrices: /usr/lib/python2.7/site-packages/pocketsphinx/model/en-us/transition_matrices
INFO: acmod.c(117): Attempting to use PTM computation module
INFO: ms_gauden.c(127): Reading mixture gaussian parameter: /usr/lib/python2.7/site-packages/pocketsphinx/model/en-us/means
INFO: ms_gauden.c(242): 42 codebook, 3 feature, size:
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(127): Reading mixture gaussian parameter: /usr/lib/python2.7/site-packages/pocketsphinx/model/en-us/variances
INFO: ms_gauden.c(242): 42 codebook, 3 feature, size:
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(304): 222 variance values floored
INFO: ptm_mgau.c(476): Loading senones from dump file /usr/lib/python2.7/site-packages/pocketsphinx/model/en-us/sendump
INFO: ptm_mgau.c(500): BEGIN FILE FORMAT DESCRIPTION
INFO: ptm_mgau.c(563): Rows: 128, Columns: 5126
INFO: ptm_mgau.c(595): Using memory-mapped I/O for senones
INFO: ptm_mgau.c(838): Maximum top-N: 4
INFO: phone_loop_search.c(114): State beam -225 Phone exit beam -225 Insertion penalty 0
INFO: dict.c(320): Allocating 138824 * 20 bytes (2711 KiB) for word entries
INFO: dict.c(333): Reading main dictionary: /usr/lib/python2.7/site-packages/pocketsphinx/model/cmudict-en-us.dict
INFO: dict.c(213): Dictionary size 134723, allocated 1016 KiB for strings, 1679 KiB for phones
INFO: dict.c(336): 134723 words read
INFO: dict.c(358): Reading filler dictionary: /usr/lib/python2.7/site-packages/pocketsphinx/model/en-us/noisedict
INFO: dict.c(213): Dictionary size 134728, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -225, delay 10)
2017-05-20 04:45:19,840 DEBUG: Setting up playback handler: VlcHandler
Type "a" to trigger Alexa: 2017-05-20 04:45:25,719 DEBUG: Stopping audio play
2017-05-20 04:45:25,923 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
media.c libvlc_media_new_from_input_item 317
event.c libvlc_event_attach 393 0x1641718
media.c libvlc_media_set_state 559
media.c libvlc_media_set_state 562 1
event.c libvlc_event_send 220 0x1641718
Segmentation fault

It look like porting issue(platform dependency issue), does anyone can give me some suggestion?

EmerickH, 2017-06-14:
Hello @frank1546, VLC is very buggy with AlexaPi, be shure it is up-to-date and try:
1. [Updating AlexaPi to the last dev branch](https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#how-to-try-the-dev-branch)
2. Using SoX instead:
    `sudo apt-get install sox`
    and change the playback handler in your config file from `vlc` to `sox`

Emerick

EmerickH, 2017-06-20:
@frank1546 have you solved your issue?

renekliment, 2017-10-08:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment