alexa-pi/AlexaPi #240: Where's Alexa's Voice

mistergrumpy, 2017-05-20:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
`Debian``
```

**Your hardware platform and model you are running on:**
`Raspberry Pi 3 running with a LAN cable``

```

**Python release (`python2 --version`):**
``2.7`

```

**Description of problem:**
I have a RPi 3 all updated and after lots of playing around with my Alexa install, I finally have something, but not everything.
There is no response from the Alexa voice. I have 3 times had Alexa say "Yes" pretty randomly but there's no pattern to it.
I can ask "Alexa. Play Classic FM/Heart Radio" or other simple named radio stations and the speaker will stutter for 2 or 3 seconds and then play fine.
I can ask "Alexa. What's new?" and again the speaker will stutter and I will get an American news radio broadcast for a short while.
Other than that though I'm not getting much else. Where's Alexa's voice? Why can I not ask "Alexa. Play some AC DC" or "Alexa. Tell me the date/the time/a joke"?
Am I missing something?
To add, I did ask "Alexa. What's the weather like today"? and got the usual choppy speaker but then a little bit of the weather report.

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```I attached a LAN cable which seemed to help between getting absolutley nothing but a stuttery speaker and what I have now.

```

mistergrumpy, 2017-05-20:
I've been talking to myself and am getting somewhere slowly. It seems like the more I ask Alexa, the more it seems to work. It's very slow to get going though, starts off a bit choppy and can crash. It almost seems to me like there's an internet speed problem. Like buffering. I have Virgin Media broadband. Is there any known speed issues or requirements for the AlexaPi?

mistergrumpy, 2017-05-21:
So I installed VLC player and the everything improved. Except now it's randomly crashing.

EmerickH, 2017-05-30:
Seems to be the same issue as [here](https://github.com/alexa-pi/AlexaPi/issues/239) for VLC crashes
Can you try to do:
```
sudo service AlexaPi stop
sudo /opt/AlexaPi/src/main.py -d
```
And copy/paste the log?

Emerick

mistergrumpy, 2017-05-30:
`[NAME]                  [DEFLT]         [VALUE]
-agc                    none            none
-agcthresh              2.0             2.000000e+00
-allphone
-allphone_ci            no              no
-alpha                  0.97            9.700000e-01
-ascale                 20.0            2.000000e+01
-aw                     1               1
-backtrace              no              no
-beam                   1e-48           1.000000e-48
-bestpath               yes             yes
-bestpathlw             9.5             9.500000e+00
-ceplen                 13              13
-cmn                    live            batch
-cmninit                40,3,-1         41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen             no              no
-debug                                  0
-dict                                   /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase               no              no
-dither                 no              no
-doublebw               no              no
-ds                     1               1
-fdict
-feat                   1s_c_d_dd       1s_c_d_dd
-featparams
-fillprob               1e-8            1.000000e-08
-frate                  100             100
-fsg
-fsgusealtpron          yes             yes
-fsgusefiller           yes             yes
-fwdflat                yes             yes
-fwdflatbeam            1e-64           1.000000e-64
-fwdflatefwid           4               4
-fwdflatlw              8.5             8.500000e+00
-fwdflatsfwin           25              25
-fwdflatwbeam           7e-29           7.000000e-29
-fwdtree                yes             yes
-hmm                                    /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us
-input_endian           little          little
-jsgf
-keyphrase                              alexa
-kws
-kws_delay              10              10
-kws_plp                1e-1            1.000000e-01
-kws_threshold          1               1.000000e-10
-latsize                5000            5000
-lda
-ldadim                 0               0
-lifter                 0               22
-lm
-lmctl
-lmname
-logbase                1.0001          1.000100e+00
-logfn
-logspec                no              no
-lowerf                 133.33334       1.300000e+02
-lpbeam                 1e-40           1.000000e-40
-lponlybeam             7e-29           7.000000e-29
-lw                     6.5             6.500000e+00
-maxhmmpf               30000           30000
-maxwpf                 -1              -1
-mdef
-mean
-mfclogdir
-min_endfr              0               0
-mixw
-mixwfloor              0.0000001       1.000000e-07
-mllr
-mmap                   yes             yes
-ncep                   13              13
[... 75 lines cut ...]
INFO: dict.c(358): Reading filler dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/noisedict
INFO: dict.c(213): Dictionary size 134728, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -225, delay 10)
2017-05-30 17:02:23 DEBUG: Setting up playback handler: VlcHandler
2017-05-30 17:02:24 INFO: Checking Internet Connection ...
2017-05-30 17:02:24 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-05-30 17:02:25 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2017-05-30 17:02:25 INFO: Connection OK
2017-05-30 17:02:25 INFO: AVS token: Requesting a new one
2017-05-30 17:02:25 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-05-30 17:02:25 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 976
2017-05-30 17:02:25 INFO: AVS token: Obtained successfully
2017-05-30 17:02:25 DEBUG: Stopping audio play
2017-05-30 17:02:25 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-05-30 17:02:25 DEBUG: Player State: State.Opening
2017-05-30 17:02:25 DEBUG: Player State: State.Playing
2017-05-30 17:02:25 DEBUG: Started play.
2017-05-30 17:02:28 DEBUG: Player State: State.Ended
2017-05-30 17:02:28 DEBUG: Finished play.
`

EmerickH, 2017-05-31:
Ok there's nothing "bad" in log
so try to use sox instead of vlc, modify:
```
  playback_handler: "sox"
```
in your config file
and `sudo apt-get install sox`

If it's not working try with pulseaudio:
`sudo apt-get install pulseaudio`
and modify in your config file
```
  playback_handler: "vlc"
  output: "pulse"
  output_device: "default"
```
and if it's not working with sox (see [here](https://github.com/alexa-pi/AlexaPi/issues/218#issuecomment-304689484))

Emerick

harimau99, 2017-05-31:
@EmerickH hi. I have the same question as well. I'm using the pulseaudio. It work awhile, but after rebooted it didn't workout.. as the alexapi muted after say hi.
Already try and check with hardware.. but no avail to solve  that.
Any idea?

EmerickH, 2017-05-31:
@harimau99 can you copy/paste the ouptut of
```
sudo service AlexaPi stop
sudo /opt/AlexaPi/src/main.py -d
```
(it will start AlexaPi in debug mode)
And your config file?

Emerick

mistergrumpy, 2017-06-05:
Well bizzarely, it seems the more I've used and crashed the thing, it's actually started working better and rarely crashes. It doesn't make any sense at all but I don't want to touch it any further at the minute. 
I'm not going to close the issue whilst harimau99 has added to it but that aside, I'm working well now.

EmerickH, 2017-06-14:
@harimau99 have you followed [the wiki](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#running-pa-in-system-wide-mode-recommended)?

Emerick

renekliment, 2017-10-14:
You can try on latest OS with latest AlexaPi.
Closing due to inactivity for now.

Feel free to reopen.

closed 2017-10-14 by renekliment