alexa-pi/AlexaPi #241: Alexa won't respond

JesusGarc, 2017-05-21:
It won't respond, i think is because of the mic I'm using (it's the ps3 camera), hope someone can help me. Thanks!

pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Sun 2017-05-21 18:50:26 UTC; 23s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 1382 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 1382 (code=exited, status=1/FAILURE)

May 21 18:50:26 raspberrypi python[1382]: surround50:CARD=CameraB409241,DEV=0
May 21 18:50:26 raspberrypi python[1382]: surround51:CARD=CameraB409241,DEV=0
May 21 18:50:26 raspberrypi python[1382]: surround71:CARD=CameraB409241,DEV=0
May 21 18:50:26 raspberrypi python[1382]: iec958:CARD=CameraB409241,DEV=0
May 21 18:50:26 raspberrypi python[1382]: dmix:CARD=CameraB409241,DEV=0
May 21 18:50:26 raspberrypi python[1382]: dsnoop:CARD=CameraB409241,DEV=0
May 21 18:50:26 raspberrypi python[1382]: hw:CARD=CameraB409241,DEV=0
May 21 18:50:26 raspberrypi python[1382]: plughw:CARD=CameraB409241,DEV=0
May 21 18:50:26 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, stat...LURE
May 21 18:50:26 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.

EmerickH, 2017-05-23:
Hello,
You have to find your device's PCM with:
`arecord -L`
You must find theses lines:
```
plughw:CARD=[YOUR CAMERA NAME],DEV=0
    Webcam [YOUR CAMERA NAME], USB Audio (or something like that)
    Hardware device with all software conversions
```
Edit the AlexaPi config file:
`sudo nano /etc/opt/AlexaPi/config.yaml `
and change this line:
`  input_device: "default"`
with:
`  input_device: "plughw:CARD=[YOUR CAMERA NAME],DEV=0"`
And finally reboot your PI or restart AlexaPi daemon

I don't know why theses steps are not on the Installation wiki page...

Emerick

lap-tran, 2017-05-26:
I have PS3 eye too and initially had the same problem. It turned out that the record was bad, it was just random noise. I fixed by modify one line in src/alexapi/capture.py.

```
			# now do VAD
			while (force_record and force_record[0]()) \
					or (do_VAD and (thresholdSilenceMet is False) and ((time.time() - start) < self.MAX_RECORDING_LENGTH)):
				length, data = inp.read()
				if length > 0:  # changed line here, ignore overrun frame
					yield data
```

Explanation: usually one period in a record is overrun, which returns negative length and one-byte buffer. Original code writes that one byte to audio stream and destroy all the subsequent good frames. The solution is to skip that bad period. Luckily it happens only once in a long series, so the overall audio isn't significantly affected.

renekliment, 2017-06-13:
@JesusGarc See the full log & docs: https://github.com/alexa-pi/AlexaPi/wiki
You need to set your input device name properly.

EmerickH, 2017-06-14:
@JesusGarc have you solved your issue?

renekliment, 2017-10-08:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment