alexa-pi/AlexaPi #242: Alexa not working with Alsa on RaspberryPi

EmerickH, 2017-05-23:
I'm running the latest version of **AlexaPi**
The PulseAudio solution was discovered [here](https://github.com/alexa-pi/AlexaPi/issues/232) but there's still no solution to use Alsa

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**
[Here](http://www.alsa-project.org/db/?f=8e002a500eee0c885a43946414cc82c2ef2c3ada)

**Your OS (including version) where you are running AlexaPi:**
```
Raspbian GNU/Linux 8 (jessie)
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi B+
```

**Python release (`python2 --version`):**
```
Python 2.7.9
```

**Description of problem:**
```
When using Alsa:
pocketsphinx not detecting "Alexa" trigger and when triggering manualy (with GPIO), AlexaPi says "yes", listen to my question and stop the microphone (I can see it with the LED on the micro) but there's no reply...
My microphone is working fine with arecord>aplay.
The only solution is to install PulseAudio and configure it with AlexaPi
```

**Problem-relevant `config.yaml` entries:**
```
sound:
  input_device: "plughw:CARD=C110,DEV=0"
  allow_unlisted_input_device: false
  playback_handler: "vlc"
  output: "alsa"
  output_device: "default"
  default_volume: 80
  media_output:
  media_output_device:
  media_default_volume:
  playback_padding: 0
```
[...]
```
triggers:
  platform:
    enabled: true
    voice_confirm: true
    event_type: "oneshot-vad"
    long_press:
      command: ""
      duration: 10
      audio_file: ""

  pocketsphinx:
    enabled: true #I also tried to disable it to test with platform
    voice_confirm: true
    phrase: "alexa"
    threshold: 1e-50
```

**Traceback (if applicable):**
```
2017-05-23 12:12:37 DEBUG: Setting up playback handler: VlcHandler
2017-05-23 12:12:38 INFO: Checking Internet Connection ...
2017-05-23 12:12:38 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-05-23 12:12:39 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2017-05-23 12:12:39 INFO: Connection OK
2017-05-23 12:12:39 INFO: AVS token: Requesting a new one
2017-05-23 12:12:39 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-05-23 12:12:40 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 978
2017-05-23 12:12:40 INFO: AVS token: Obtained successfully
2017-05-23 12:12:40 DEBUG: Stopping audio play
2017-05-23 12:12:40 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-05-23 12:12:40 DEBUG: Player State: State.Opening
2017-05-23 12:12:41 DEBUG: Player State: State.Playing
2017-05-23 12:12:41 DEBUG: Started play.
2017-05-23 12:12:43 DEBUG: Player State: State.Ended
2017-05-23 12:12:43 DEBUG: Finished play.
2017-05-23 12:12:46 DEBUG: Button pressed!
2017-05-23 12:12:46 DEBUG: Stopping audio play
2017-05-23 12:12:46 DEBUG: Playing audio: /opt/AlexaPi/src/resources/alexayes.mp3
2017-05-23 12:12:46 DEBUG: Player State: State.Opening
2017-05-23 12:12:46 DEBUG: Player State: State.Playing
2017-05-23 12:12:46 DEBUG: Started play.
2017-05-23 12:12:47 DEBUG: Button released.
2017-05-23 12:12:47 DEBUG: Player State: State.Ended
2017-05-23 12:12:47 DEBUG: Finished play.
2017-05-23 12:12:47 DEBUG: Setting up recording
2017-05-23 12:12:47 DEBUG: Start recording
2017-05-23 12:12:47 DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com
2017-05-23 12:12:47 DEBUG: Button pressed!
2017-05-23 12:12:48 DEBUG: Button released.
2017-05-23 12:12:48 DEBUG: Start sending speech to Alexa Voice Service
2017-05-23 12:12:56 DEBUG: End recording
2017-05-23 12:12:56 DEBUG: Finished sending speech to Alexa Voice Service
2017-05-23 12:12:57 DEBUG: Processing Request Response...
2017-05-23 12:12:57 DEBUG: Request Response is null (This is OKAY!)
```
Thank you,
Emerick

EmerickH, 2017-06-10:
Hello @ezar,
You have two solution:

1. you can use pulseaudio instead of alsa.
Install it:
`sudo apt-get install pulseaudio`
and put in your config file:
`input_device: "pulse"`
2. install [the last developement version](https://github.com/alexa-pi/AlexaPi/tree/dev)
`git clone -b dev https://github.com/alexa-pi/AlexaPi.git`
PyAudio has been replaced by Pyalsaaudio, it should works better (if you try this solution can you say me if it's working?)

Emerick

ezar, 2017-06-11:
Nothing.. I test using webcam mic, an external usb mic, ... 
Magic Word "Alexa" is not detected.

I try:
1. Pulse audio.
2. PyAudio. Latest development version.

Can you share your config file?

EmerickH, 2017-06-11:
Hello,
strange...
My config is:
```
  input_device: "pulse"
  allow_unlisted_input_device: false
  playback_handler: "sox"
  output: "pulseaudio"
  output_device: "default"
```
I'm using Sox, but it doesn't change anything with input device
(if you want to use Sox, do `sudo apt-get install pulseaudio libsox-fmt-pulse`)

Maybe pulseaudio is not correctly configured... I don't know how to configure it with pacmd but there's packages to do it easier.

- If you have a graphical interface, install `pavucontrol`:
`sudo apt-get install pavucontrol`
and launch it with `pavucontrol` command.
You can now check if the input devices are correctly configured.

- If you don't have one, try `pamixer`
Try:
`sudo apt-get install pamixer`
if it's not working, install it manually:
```
git clone https://github.com/cdemoulins/pamixer.git
cd pamixer
make
./pamixer
```

Emerick

ezar, 2017-06-11:
Hi,
I use your config... doesn't works!
Using "pavucontrol" I verify that mic is detected... seems a problem with AlexaPi because arecord works correctly.
:(

ezar, 2017-06-11:
Do you user trigger "Alexa" or use button?
Regards,

EmerickH, 2017-06-13:
Hello,
I'm using **pocketshpinx** ("Alexa" trigger).
Can you copy/paste your log?
See [here](https://github.com/alexa-pi/AlexaPi/wiki/Debugging#debug-mode) in the **Debug mode** category to start in debug mode to see the complete log

Emerick

ezar, 2017-06-13:
2017-06-13 21:41:38 DEBUG: Initialising Magic Mirror platorm
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]                  [DEFLT]         [VALUE]
-agc                    none            none
-agcthresh              2.0             2.000000e+00
-allphone                               
-allphone_ci            no              no
-alpha                  0.97            9.700000e-01
-ascale                 20.0            2.000000e+01
-aw                     1               1
-backtrace              no              no
-beam                   1e-48           1.000000e-48
-bestpath               yes             yes
-bestpathlw             9.5             9.500000e+00
-ceplen                 13              13
-cmn                    live            batch
-cmninit                40,3,-1         41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17-compallsen             no              no
-debug                                  0
-dict                                   /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase               no              no
-dither                 no              no
-doublebw               no              no
-ds                     1               1
-fdict                                  
-feat                   1s_c_d_dd       1s_c_d_dd
-featparams                             
-fillprob               1e-8            1.000000e-08
-frate                  100             100
-fsg                                    
-fsgusealtpron          yes             yes
-fsgusefiller           yes             yes
-fwdflat                yes             yes
-fwdflatbeam            1e-64           1.000000e-64
-fwdflatefwid           4               4
-fwdflatlw              8.5             8.500000e+00
-fwdflatsfwin           25              25
-fwdflatwbeam           7e-29           7.000000e-29
-fwdtree                yes             yes
-hmm                                    /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us
-input_endian           little          little
-jsgf                                   
-keyphrase                              alexa
-kws                                    
-kws_delay              10              10
-kws_plp                1e-1            1.000000e-01
-kws_threshold          1               1.000000e-10
-latsize                5000            5000
-lda                                    
-ldadim                 0               0
-lifter                 0               22
-lm                                     
-lmctl                                  
-lmname                                 
-logbase                1.0001          1.000100e+00
-logfn                                  
-logspec                no              no
-lowerf                 133.33334       1.300000e+02
-lpbeam                 1e-40           1.000000e-40
-lponlybeam             7e-29           7.000000e-29
-lw                     6.5             6.500000e+00
-maxhmmpf               30000           30000
-maxwpf                 -1              -1
-mdef                                   
-mean                                   
-mfclogdir                              
-min_endfr              0               0
-mixw                                   
-mixwfloor              0.0000001       1.000000e-07
-mllr                                   
[... 93 lines cut ...]
2017-06-13 21:41:42 INFO: AVS token: Obtained successfully
2017-06-13 21:41:42 DEBUG: Starting Magic Mirror platform HTTP Server
2017-06-13 21:41:42 DEBUG: Starting Magic Mirror heartbeat with 5 second interval
2017-06-13 21:41:42 DEBUG: Sending MM Heatbeat
2017-06-13 21:41:42 DEBUG: Response: {"status":"success"}2017-06-13 21:41:42 DEBUG: Stopping audio play
2017-06-13 21:41:43 DEBUG: Stopped play. No SoX process to stop
2017-06-13 21:41:43 DEBUG: Indicate Stop Playing
2017-06-13 21:41:43 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
2017-06-13 21:41:43 DEBUG: Response: {"status":"success"}2017-06-13 21:41:43 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-06-13 21:41:43 DEBUG: Started play. sox -q /opt/AlexaPi/src/resources/hello.mp3 -t pulseaudio default vol -6 dB pad 0 0
2017-06-13 21:41:43 DEBUG: Indicate Start Playing
2017-06-13 21:41:43 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=playback
2017-06-13 21:41:43 DEBUG: Response: {"status":"success"}2017-06-13 21:41:45 DEBUG: Finished play. 
2017-06-13 21:41:45 DEBUG: Indicate Stop Playing
2017-06-13 21:41:45 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
2017-06-13 21:41:46 DEBUG: Response: {"status":"success"}2017-06-13 21:41:46 DEBUG: Indicating Success
2017-06-13 21:41:46 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=success
2017-06-13 21:41:46 DEBUG: Response: {"status":"success"}2017-06-13 21:41:47 DEBUG: Sending MM Heatbeat
2017-06-13 21:41:47 DEBUG: Response: {"status":"success"}^C2017-06-13 21:41:48 DEBUG: Cleaning up Magic Mirror platform
2017-06-13 21:41:48 DEBUG: Cleaning up playback handler
2017-06-13 21:41:48 DEBUG: Stopped play. [Errno 3] No such process
2017-06-13 21:41:48 DEBUG: Indicate Stop Playing
2017-06-13 21:41:48 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
2017-06-13 21:41:49 DEBUG: Response: {"status":"success"}INFO: cmn_live.c(88): Update from < 41.00 -5.29 -0.12  5.09  2.48 -4.07 -1.37 -1.78 -5.08 -2.05 -6.45 -1.42  1.17 >
INFO: cmn_live.c(105): Update to   < 44.82 -11.86 -7.02 -2.72 -7.96  0.93  3.32 -5.41 -9.14  0.32 -1.12 -0.98  2.14 >

ezar, 2017-06-13:
# See https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging
# to get a better idea of how to set things in here

# Level of log output
# default: INFO
# set to DEBUG for verbose logging
logging: INFO

sound:
  # Name of your ALSA microphone/soundcard in arecord -L
  #input_device: "default"
  input_device: "pulse"
  # This is here if none of the listed device: "magicmirror"
  # This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false

  #playback_handler: "vlc"
  playback_handler: "sox"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  #output: "alsa"
  output: "pulseaudio"
  # output device: "magicmirror"
  # currently supports ALSA only
  output_device: "default"

  # if left empty, won't be passed to the underlying library / command if it supports it
  # if it doesn't 100% will be used
  default_volume: 80

  # Means the same as above, but is used for media (playing music for example)
  # Leave blank to use the same setting as for speech content
  media_output:
  media_output_device:
  media_default_volume:

  # Some device: "magicmirror"
  # value is the number of seconds of silence to pad the beginning and end of the audio. Only
  # the sox handler supports this configuration option.
  playback_padding: 0

# Amazon Alexa settings
### removed

# See https://github.com/alexa-pi/AlexaPi/wiki/Triggers for the description of these options
triggers:
  platform:
    enabled: true
    voice_confirm: false

    # one of: oneshot-vad, continuous, continuous-vad
    event_type: "oneshot-vad"

    # only for "continuous" event_types
    long_press:
      # system command to run after pressing the button for _duration_ number of seconds
      # empty to disable
      command: ""

      # duration in seconds
      duration: 10

      # CURRENTLY DISABLED
      # optional audio file to play before executing the command
      # you might wanna use full path to make sure AlexaPi finds it
      # you can use {resources_path} for the path to the AlexaPi "resources" directory
      audio_file: ""

  pocketsphinx:
    enabled: true
    voice_confirm: true

    phrase: "alexa"
    threshold: 1e-10

# Commands to run before and after an interaction begins. Leave empty to disable.
event_commands:

  # Example used to lower shairport-sync volume during interaction with alexa
  #     startup: "amixer set shairportvolume 100%"
  #     pre_interaction: "amixer set shairportvolume 10%-""
  #     post_interaction: "amixer set shairportvolume 10%+""
  # Note this assumes you have redirected shairport-syncs audio to a seperate softdevice
  # with control "shairportvolume". Example asound.conf avalible in the scripts folder

  # Command to run at the start of AlexaPi
  startup: ""

  # Command to run after a trigger is detected but before an interaction begins
  pre_interaction: ""

  # Command to run after the interaction has completed
  post_interaction: ""

  # Command to run during a clean shutdown of AlexaPi
  shutdown: ""

platform:
  # Name of your platform, e.g. raspberrypi, orangepi, desktop
  device: "magicmirror"

platforms:
  # common config for all platforms (if they support the options)
  # can be overriden in the the respective platform sections
  common:

  chip:
    # GPIO Pin with button connected
    button: XIO-P1
    # GPIO Pin for the playback/activity light
    plb_light: XIO-P3
    # GPIO Pin for the recording light
    rec_light: XIO-P5

  raspberrypi:
    # GPIO Pin with button connected
    button: 18
    # GPIO Pin for the playback/activity light
    plb_light: 24
    # GPIO Pin for the recording light
    rec_light: 25

  orangepi:
    # GPIO Pin with button connected
    button: 20
    # GPIO Pin for the playback/activity light
    plb_light: 9
    # GPIO Pin for the recording light
    rec_light: 8

  desktop:
    min_seconds_to_record: 3

  dummy:

  magicmirror:
    # Magic Mirror - github.com/MichMich/MagicMirror
    # Integrates with MMM-AlexaPi module (github.com/dgonano/MMM-AlexaPi) to display listening, playback, etc status
    # Hostname of the AlexaPi HTTP server, leave empty for any name
    hostname: ""
    # Port the ALexaPi HTTTP server binds to
    port: 5055
    # Hostname of the Magic Mirror
    mm_hostname: "localhost"
    # Port of the Magic Mirror
    mm_port: "8080"
    # How ofter the Hearbeat fires in seconds
    hb_timer: 5

  hyperion:
    # Hostname of the Hyperion node's JSON server
    hyperion_json_host: "127.0.0.1"
    # Port of the Hyperion node's JSON server
    hyperion_json_port: 19444
    # Visual status as static 'color' or 'effect'
    # Overridden by 'hyperion_flash_state_xxx'
    hyperion_mode: "effect"
    # Running speed when visualizing as 'effect'
    hyperion_effect_speed: 1
    hyperion_priority: 1
    color_playback: [80, 213, 35]
    color_processing: [35,169,213]
    color_recording: [168, 35, 213]
    flash_duration: 500
    flash_frequency: 8
    # Use Strobe effect instead of default effect or static color
    flash_state_playback: true
    flash_state_processing: false
    flash_state_recording: false
    # Visualize playback state
    indicate_playback: true
    #Also print json request/response when in debug mode
    verbose: false

  serial:
    port: "/dev/ttyACM0"
    baudrate: 9600
    messages:
      trigger: ""
      playback_start: "g"
      playback_end: "x"
      recording_start: ""
      recording_end: ""
      processing_start: ""
      processing_end: ""
      failure: ""
      success: ""

ezar, 2017-06-13:
Config.yaml
https://pastebin.com/Sk0Gz8VL

ezar, 2017-06-13:
I install python 3, updated pip libs, ... nothing!
Alexa trigger doesn't work. Can I trigger using URL . Seems that using magic mirror integration I can... how?

EmerickH, 2017-06-14:
Hello,
You don't precise that you use `magicmirror`...
It's written on your config:
```
# output device: "magicmirror"
# currently supports ALSA only
```

So your config must be:
```
sound:
  # Name of your ALSA microphone/soundcard in arecord -L
  input_device: "default"
  # This is here if none of the listed device: "magicmirror"
  # This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false
 
  playback_handler: "vlc"
or
  playback_handler: "sox"
 
  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "alsa"
  # output device: "magicmirror"
  # currently supports ALSA only
  output_device: "default"
or try
  output_device: "magicmirror"
```

Emerick

ezar, 2017-06-14:
Nothimg! Doesn't works. I try your config but Alexa magic word don't wake the system.

I hear hello using Alsa and Pulseaudio. 

The problem it's wake up. Every configuration doesn't works. I reinstall Jessie Lite many times.

Regards

EmerickH, 2017-06-14:
@ezar what do you use like input device in your config?
You can list them with `arecord -L`
(uses only `plughw:` devices)

Emerick

ezar, 2017-06-15:
`default
    Playback/recording through the PulseAudio sound server
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
sysdefault:CARD=Device
    USB PnP Sound Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Hardware device with all software conversions`

EmerickH, 2017-06-15:
@ezar try to install the last dev version: https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#how-to-try-the-dev-branch

ezar, 2017-06-15:
Updated to dev branch.

Config.yaml
```
sound:
  # Name of your microphone device in PortAudio
  # leave empty for default
  input_device: ""

  #playback_handler: "vlc"
  playback_handler: "sox"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "alsa"
  # output device for the audio output specified above,
  # currently supports ALSA only
  output_device: "default"

  # if left empty, won't be passed to the underlying library / command if it supports it
  # if it doesn't 100% will be used
  default_volume: 80

  # Means the same as above, but is used for media (playing music for example)
  # Leave blank to use the same setting as for speech content
  media_output:
  media_output_device:
  media_default_volume:

  # Some device: "magicmirror"
  # value is the number of seconds of silence to pad the beginning and end of the audio. Only
  # the sox handler supports this configuration option.
  playback_padding: 0
```

New logs:
https://pastebin.com/e5fNBJfK

EmerickH, 2017-06-15:
Ok now try changing your device.
see my reply in your issue https://github.com/alexa-pi/AlexaPi/issues/256#issuecomment-308051900

ezar, 2017-06-15:
2017-06-15 11:02:37 CRITICAL: Your input_device 'plughw:CARD=Device,DEV=0' is invalid. Use one of the following:
pulse
default

ezar, 2017-06-17:
Using this seems to works Alexa trigger: https://github.com/dolanmiu/MMM-awesome-alexa/

renekliment, 2017-10-14:
We use PyAudio now. Closing.

closed 2017-10-14 by renekliment