alexa-pi/AlexaPi #243: Hacked Sox to work, PulseAudio in User Mode, remote sinks and sources

KillerKelvUK, 2017-05-23:
So this isn't a bug as I've managed to get AlexaPi working but I had to make 1 minor code modification and want to share the info to test how robust my *hack* is and to maybe help others.  Please point me in the right direction if here isn't the right place for this.

CAUTION:  I'm not a developer!

**TL;DR**
I amended line 46 in the `AlexaPi/src/alexapi/playback_handlers/soxhandler.py` file from `sox_cmd = ['sox', '-q', audio_file]` to `sox_cmd = ['sox', '-q', audio_file, '-d']` 

**The Detail**
New RPi3 user, running Raspbian jessie 8.0 with all stock packages.  PulseAudio is running as installed by the main package which is in User Mode, however, I have no directly connected sinks/speakers or sources/microphones attached to the RPi3.  Instead, I use zeroconf (avahi) to share the audio devices of my ubuntu server with the RPi3 on my local network.  So basically the RPi3 is connected via wifi and plays audio via the speakers which are physically connected to my server, same goes with the microphone to record.  I had this all set up and working before I attempted AlexaPi.

For AlexaPi I just walked the path you guys laid out with the installation script but in addition I noted the alexapi user had its home directory set as `/var/lib/AlexPi` and in here you have a `.config/pulse` directory so I needed to put in their the PulseAudio security cookie from my server.  After that the I amended the `/etc/opt/AlexaPi/config.yaml` file with only the following changes...

```
  input_device: "pulse"
  playback_handler: "vlc"
  output: ""
  output_device: ""
```
Now, this all works but I had a strange issue with Alexa's output where the playback clipped twice at the start of the output but then was fine for the remaining duration.  So asking Alexa to "play my flash briefing" I noticed in debug there are about 3 or 4 separate play operations and this clipping/clicking at the front of each which was really annoying.

I quickly dived into the wiki e.g. [here](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-%26-debugging) and tried all of the sensible-for-me suggestions i.e. installing the full vlc package but nothing resolved this.  I then came across the `playback_handler` setting in the config file and looked up [this](https://github.com/alexa-pi/AlexaPi/issues/218) thread and so gave sox a try, however, I hit the same error message others in this thread have reported...

> Error attempting play. sox FAIL sox: missing filename

Now the debug output helpfully includes the command used to wield sox so I gave this a try outside of AlexaPi to figure it out.  `sox -q /opt/AlexaPi/src/resources/alexayes.mp3 vol -6 dB pad 0 0` just fails at the command prompt and lists the command help text for sox, if this isn't proof enough sox is installed I even checked it properly...

```
pi@blackpi:/var/lib/AlexaPi $ dpkg -l sox
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                Version                Architecture           Description
+++-===================================-======================-======================-===========================================================================
ii  sox                                 14.4.1-5               armhf                  Swiss army knife of sound processing
```
Reading the man page for sox it seems the sox devs include handly little sym-linked executables to make it simple for simpletons like me i.e. play and rec.  These worked first time at the command prompt so this had to be something in the AlexaPi code.  Anyways for me amending the sox command to 
include a `-d` directly following the filename worked...

So changing `sox -q /opt/AlexaPi/src/resources/alexayes.mp3 vol -6 dB pad 0 0` to `sox -q /opt/AlexaPi/src/resources/alexayes.mp3 -d vol -6 dB pad 0 0`

Quoting the sox man page...

>        -d, --default-device
>               This  can  be  used  in  place of an input or output filename to specify that the default audio device (if one has been
>               built into SoX) is to be used.  This is akin to invoking rec or play (as described above).

To incorporate this into AlexaPi I did my best to understand all your wizardry and located the `AlexaPi/src/alexapi/playback_handlers/soxhandler.py` file.  Line 46 reads `sox_cmd = ['sox', '-q', audio_file]` so I amended this to be `sox_cmd = ['sox', '-q', audio_file, '-d']`.  I was pretty surprised when it worked, sox now played back the audio and there is no clipping or clicking whatsoever!

I realise the audio aspect to AlexaPi is complex and that I don't fully understand it so wanted to offer this up in case I'm heading for a wreck or whether just possibly its okay and useful to others.

renekliment, 2017-06-13:
@KillerKelvUK

You need to have `sound/output` set to something in the config as it sets SoX's `-t` option: https://github.com/alexa-pi/AlexaPi/blob/master/src/alexapi/playback_handlers/soxhandler.py#L24

I'd try `alsa` or `pulse` depending on whether you have the necessary module installed.

SalSuleman, 2017-06-14:
@renekliment 
I'm new to programming so I'm not too sure how so set pulse as the audio output.
would it be like this?
`		if self.__config['sound']['output']:
			self.parameters_speech.extend(['-t', self.__config['sound']['pulse']])
			if self.__config['sound']['output_device']:
				self.parameters_speech.append(self.__config['sound']['output_device'])`

KillerKelvUK, 2017-06-14:
@renekliment

Yes, I tried several settings in the configuration file but none worked.  However according to the sox manual using the parameter "-t" sets the file type and doesn't relate to the output device.

```
-t, --type FILE-TYPE
              Gives the type of the audio file.  For both input and output files, this option is commonly used to inform SoX of the type a `headerless' audio file (e.g. raw, mp3) where  the  actual/desired
              type cannot be determined from a given filename extension.  For example:
                 another-command | sox -t mp3 - output.wav

                 sox input.wav -t raw output.bin
              It  can  also  be  used to override the type implied by an input filename extension, but if overriding with a type that has a header, SoX will exit with an appropriate error message if such a
              header is not actually present.

              See soxformat(7) for a list of supported file types.
```

EmerickH, 2017-06-14:
@KillerKelvUK it's `pulseaudio` not `pulse`.
From [SoX documentation](http://sox.sourceforge.net/soxformat.html):
```
PulseAudio driver; supports both playing and recording of audio. PulseAudio is a cross platform networked sound server. If a file name is specified with this driver, it is ignored. Examples:
     sox infile −t pulseaudio
     sox infile −t pulseaudio default
```

KillerKelvUK, 2017-06-14:
@EmerickH 

Neither "pulse", "pulseaudio" or "pulseaudio default" work, see below from AlexaPi debug log...

```
2017-06-14 17:22:16 DEBUG: Started play. sox -q /opt/AlexaPi/src/resources/hello.mp3 -t pulseaudio default default vol -6 dB pad 0 0
2017-06-14 17:22:16 DEBUG: Error attempting play. sox FAIL formats: no handler for given file type `pulseaudio default'
```
Unless I have a bad configuration, please advise?

```
sound:
  # Name of your ALSA microphone/soundcard in arecord -L
  input_device: "pulse"
  # This is here if none of the listed device: "raspberrypi"
  # This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false

  playback_handler: "sox"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "pulseaudio default"
  # output device: "raspberrypi"
  # currently supports ALSA only
  output_device: "default"

  # if left empty, won't be passed to the underlying library / command if it supports it
  # if it doesn't 100% will be used
  default_volume: 80

  # Means the same as above, but is used for media (playing music for example)
  # Leave blank to use the same setting as for speech content
  media_output:
  media_output_device:
  media_default_volume:

  # Some device: "raspberrypi"
  # value is the number of seconds of silence to pad the beginning and end of the audio. Only
  # the sox handler supports this configuration option.
  playback_padding: 0
```

EmerickH, 2017-06-14:
@KillerKelvUK you must have `libsox-fmt-pulse`:
```
sudo apt-get install libsox-fmt-pulse
```

KillerKelvUK, 2017-06-14:
Thank you @EmerickH, installing that package and changing my config to...

`output: "pulseaudio"`

resolved/corrected the issue :-)

However for me Sox did work by removing the -t parameter and replacing it with -d but I understand everyone's install is going to be different.  So rather than my poor attempt at coding the install script needs updating to grab the extra package and the config help comments need changing to say "pulseaudio".  Simple fix and very much appreciated cheers.

closed 2017-06-14 by KillerKelvUK

EmerickH, 2017-06-14:
Added this to **Audio setup & debug** guide

renekliment, 2017-06-14:
Awesome!