alexa-pi/AlexaPi #245: Alexa won't respond because of corrupted record

lap-tran, 2017-05-26:
I have PS3 eye, Pi2 on Raspbian Jesse Lite. Everything works fine until Alexa needs to respond. She almost never does.

Debug and play recording.wav file, I found the audio stream was corrupted, just random noise and contained an odd number of bytes. That's weird because each frame are 2 bytes.

Further debugged, I saw in most records, there was one time - quite early - alsaaudio.read() returned negative length. From pyalsaaudio documentation, that period is overrun. In that case, the returned buffer is only one byte. When we write that single byte to stream, all subsequent good frames are shifted in a half frame, result in corrupted data.

Solution: ignore overrun period.

lap-tran, 2017-05-26:
Proposed fix
In src/alexapi/capture.py, change a single line as below (ignore overrun period)

```
	# now do VAD
	while (force_record and force_record[0]()) \
			or (do_VAD and (thresholdSilenceMet is False) and ((time.time() - start) < self.MAX_RECORDING_LENGTH)):
		length, data = inp.read()
		if length > 0:  # ignore overrun period
			yield data

```

harimau99, 2017-05-26:
Great. Will try your method...

harimau99, 2017-05-31:
In my side. It didn't workout. Didn't have error just muted.

renekliment, 2017-06-13:
@lap-tran Hey. Thanks for the debugging. Currently we use PyAudio in the `dev` branch. Can you give it a try and see if the issue persists?

lap-tran, 2017-07-15:
The dev branch works fine, isn't affected by this bug.

harimau99, 2017-10-03:
Yep.. its work perfectly.. @renekliment , what was the issue?

renekliment, 2017-10-08:
I don't know. Linux audio is pure magic. Anyways, the stuff is in `master` now. Closing.

closed 2017-10-08 by renekliment