alexa-pi/AlexaPi #246: Airplay

katherinebui, 2017-05-30:
Has anyone been successful using Airplay? It says my device is connected, but no sound is being produced. I have ran the audio tests and my audio does work. Thoughts?

maso27, 2017-05-30:
Airplay works great for me; we use it all the time in my house.
No real suggestions except this: Have you made sure the volume is turned up on your phone after connecting to Airplay? I think ios maintains multiple volume settings for different audio configurations.

katherinebui, 2017-05-30:
Yeah- I made sure the volume is on. For some reason, it doesn't show up on Airplay now. :(

maso27, 2017-05-30:
The Airplay support is accomplished by simply installing shairport-sync during the install.
Hopefully the information at the github page can help you:
https://github.com/mikebrady/shairport-sync

There are a ton of configuration options for it.

As for getting it to show up again--if it has crashed, it should start again after a reboot, or you can get it going with `sudo systemctl start shairport-sync`. 

Sorry you're having trouble!

SpeedRacerX, 2017-05-31:
@katherinebui , I had a similar issue, which I resolved by modifying the audio settings:

I needed to modify the output_device setting in /etc/opt/AlexaPi/config.yaml to match my speaker setup.  
1. First, I ran aplay -L, to get the output device associated with my board.  There are several values returned, but since I'm using the 3.5mm audio device, I used plughw:CARD=ALSA,DEV=0 as the output device.
2. Then I modified the config.yaml file, setting the output_device to: "plughw:CARD=ALSA,DEV=0"

After doing this, I rebooted the pi and it worked for me.  Good luck!

EmerickH, 2017-06-14:
@katherinebui have you tried @SpeedRacerX's solution?
If it's working, please close this issue.

Emerick

renekliment, 2017-10-14:
Closing due to inactivity.

closed 2017-10-14 by renekliment