alexa-pi/AlexaPi #247: No Response From AlexaPI

hankp46, 2017-05-30:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
```

**Your hardware platform and model you are running on:**
```

```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
Had Clean install of AlexaPI. Originally heard "Hello" but could not input any requests. Followed debugging advice and went to Running PA in system-wide mode . Made all changes. On reboot all I hear is 2 bursts of static. Here is the terminal output:
pi@raspberrypi:~ $ /opt/AlexaPi/src/main.py
2017-05-30 17:14:20 INFO: Checking Internet Connection ...
2017-05-30 17:14:20 INFO: Connection OK
2017-05-30 17:14:20 INFO: AVS token: Requesting a new one
2017-05-30 17:14:20 INFO: AVS token: Obtained successfully
[6bf004d8] alsa audio output error: cannot estimate delay: Input/output error

pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; disabled)
   Active: inactive (dead)
     Docs: https://github.com/alexa-pi/AlexaPi/wiki

May 30 16:44:12 raspberrypi systemd[1]: Starting Alexa client for all your devices...
May 30 16:44:12 raspberrypi systemd[1]: Started Alexa client for all your devices.
May 30 16:44:26 raspberrypi python[549]: INFO: Checking Internet Connection ...
May 30 16:44:27 raspberrypi python[549]: INFO: Connection OK
May 30 16:44:27 raspberrypi python[549]: INFO: AVS token: Requesting a new one
May 30 16:44:27 raspberrypi python[549]: INFO: AVS token: Obtained successfully
May 30 16:44:29 raspberrypi python[549]: [6bf004d8] alsa audio output error: cannot estimate delay: Input/output error
May 30 17:03:59 raspberrypi systemd[1]: Stopping Alexa client for all your devices...
May 30 17:03:59 raspberrypi systemd[1]: Stopped Alexa client for all your devices.

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```sound:
  # Name of your ALSA microphone/soundcard in arecord -L
#  input_device: "plughw:CARD=Device,DEV=0"
  input_device: "pulse"
 # This is here if none of the listed device: "raspberrypi"
  # This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false

  playback_handler: "vlc"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "pulse"
  # output device: "raspberrypi"
  # currently supports ALSA only
  output_device: "pulse"

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

hankp46, 2017-05-30:
I got it to work but I need to be right on top of microphone for it to pickup

closed 2017-05-30 by hankp46

maso27, 2017-05-30:
You can adjust the volume of your mic using `sudo alsamixer`, then save your settings with `sudo alsactl store`.
You can also adjust the sensitivity of the trigger word detection within _/etc/opt/AlexaPi/config.yaml_.

Maybe one of those will make it more receptive.

Good luck!