alexa-pi/AlexaPi #248: Best Microphone for AlexaPi

hankp46, 2017-05-30:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
```

**Your hardware platform and model you are running on:**
```

```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
I finally go AlexaPi to work with some juggling of YAML settings:
input_device:"default"
output:"alsa"
output_device:"speaker"
Why this worked, don't know but it does.
My problem now is that I need to be right next to mic for it to pick up. I have ALSAmixer capture to max and also same on Pi audio device settings.
Any suggestion on other software fixes?
If not, can  anyone recommend a decent mini microphone that can pickup from about 6 -10 feet?

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

maso27, 2017-05-30:
I just replied to your other issue.  Have you tried debugging your input with the "Audio setup & debugging" section in the wiki?  If you record audio and play it back you should at least get a sense of how effectively you're picking up the sounds in the room.

You can adjust the settings within _/etc/opt/AlexaPi/config.yaml_ to see if adjusting the sensitivity helps, or if perhaps changing the trigger word is more effective for you.

As for mics, I've found that webcams work well, and even the little [kinovo mic from amazon](https://smile.amazon.com/Kinobo-Microphone-Desktop-Recognition-Software/dp/B00IR8R7WQ/ref=sr_1_2) works well enough.

hankp46, 2017-05-31:
Where do you adjust the sensitivity in yaml?  Changing the trigger word helped some. Thanks!

maso27, 2017-05-31:
It's called "threshold".

hankp46, 2017-06-01:
Answered

closed 2017-06-01 by hankp46