alexa-pi/AlexaPi #249: KeyError: 'language'

mcosti, 2017-05-31:
**Your OS (including version) where you are running AlexaPi:**
Raspbian

**Your hardware platform and model you are running on:**
Raspberry

**Python release (`python2 --version`):**
```Python 2.7.9

```

**Description of problem:**
```
Alexa is not starting

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. python /opt/AlexaPi/src/main.py -d
2.  That's all

**Traceback (if applicable):**
```Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 646, in <module>
    triggers.setup()
  File "/opt/AlexaPi/src/alexapi/triggers/__init__.py", line 20, in setup
    trigger.setup()
  File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 32, in setup
    ps_config.set_string('-hmm', os.path.join(get_model_path(), self._tconfig['language']))
KeyError: 'language'

```

**Additional info:**
```

```

EmerickH, 2017-05-31:
Hello,
It seems to be a missing option,
can you copy your config file?
Try to reset it (you can find the default file [here](https://github.com/alexa-pi/AlexaPi/blob/master/src/config.template.yaml))

Emerick

mcosti, 2017-05-31:
```
pi@raspberrypi:~ $ sudo cat /opt/AlexaPi/src/config.template.yaml
# See https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging
# to get a better idea of how to set things in here

# Level of log output
# default: INFO
# set to DEBUG for verbose logging
logging: INFO

sound:
  # Name of your ALSA microphone/soundcard in arecord -L
  input_device: "pulse"
  # This is here if none of the listed devices works for you and you need to use a value like "plughw:1"
  # This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false

  playback_handler: "vlc"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "pulse"
  # output device for the audio output specified above,
  # currently supports ALSA only
  output_device: ""

  # if left empty, won't be passed to the underlying library / command if it supports it
  # if it doesn't 100% will be used
  default_volume: 80

  # Means the same as above, but is used for media (playing music for example)
  # Leave blank to use the same setting as for speech content
  media_output:
  media_output_device:
  media_default_volume:

  # Some devices clip off the beginning/end of audio files being played back. The
  # value is the number of seconds of silence to pad the beginning and end of the audio. Only
  # the sox handler supports this configuration option.
  playback_padding: 0

  # Google Assistant SDK Audio Adjustments
  # If Assistant audio is choppy, try adjusting the sound device's block size [...]:
  #   If using a USB speaker or dedicated soundcard, set block size to `0` to automatically adjust the buffer size.
  #   If using the line-out 3.5mm audio jack on the Raspberry Pi, set block size to a value larger than the `ConverseResponse` audio payload size.
  #   e.g. 3200
  # If Assistant audio is truncated, try adjusting the sound device's flush size [...]
  # --> see https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/troubleshooting
  # (licensed under the Creative Commons Attribution 3.0 License, code samples are licensed under the Apache 2.0 License)
  #
  # Test for your self and use the best block size value in the settings below:
  # /opt/alexaPi/env/bin/python -m googlesamples.assistant --audio-block-size=<test_value>
  assistant:
    block_size: ""
    flush_size: ""

# Amazon Alexa settings
alexa:
  Client_ID:
  Client_Secret:
  Device_Type_ID:
  Security_Profile_Description:
  Security_Profile_ID:
  refresh_token:

# See https://github.com/alexa-pi/AlexaPi/wiki/Triggers for the description of these options
triggers:
  platform:
    enabled: false
    voice_confirm: false
[... 123 lines cut ...]
    flash_duration: 500
    flash_frequency: 8
    # Use Strobe effect instead of default effect or static color
    flash_state_playback: true
    flash_state_processing: false
    flash_state_recording: false
    # Visualize playback state
    indicate_playback: true
    #Also print json request/response when in debug mode
    verbose: false

  serial:
    port: "/dev/ttyACM0"
    baudrate: 9600
    messages:
      trigger: ""
      playback_start: "g"
      playback_end: "x"
      recording_start: ""
      recording_end: ""
      processing_start: ""
      processing_end: ""
      failure: ""
      success: ""
```

mcosti, 2017-05-31:
I did replace config.template.yaml with what you've given me, same error.

EmerickH, 2017-05-31:
Ha sorry I did not specify, the config is located in /etc/opt/AlexaPi/config.yaml in your Raspberry, config.template.yaml is just a template.

mcosti, 2017-05-31:
Yeah, tought about it, but I couldn't find the completed config.py :D 
Here it is:
`pi@raspberrypi:/opt/AlexaPi/src/alexapi $ sudo cat /etc/opt/AlexaPi/config.yaml
# See https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging
# to get a better idea of how to set things in here

# Level of log output
# default: INFO
# set to DEBUG for verbose logging
logging: INFO

sound:
  # Name of your ALSA microphone/soundcard in arecord -L
  #input_device: "plughw:CARD=Device,DEV=0"
  input_device: "pulse"
  # This is here if none of the listed device: "raspberrypi"
  # This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false

  playback_handler: "vlc"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "pulse"
  # output device: "raspberrypi"
  # currently supports ALSA only
  output_device: ""

  # if left empty, won't be passed to the underlying library / command if it supports it
  # if it doesn't 100% will be used
  default_volume: 80

  # Means the same as above, but is used for media (playing music for example)
  # Leave blank to use the same setting as for speech content
  media_output:
  media_output_device:
  media_default_volume:

  # Some device: "raspberrypi"
  # value is the number of seconds of silence to pad the beginning and end of the audio. Only
  # the sox handler supports this configuration option.
  playback_padding: 0

# Amazon Alexa settings
alexa:
  Client_ID: "amzn1.application-oa2-client.29399085c36142e893d5b9a19d495bfa"
  Client_Secret: "replaced_for_reasons"
  Device_Type_ID: "raspi_mc"
  Security_Profile_Description: "Raspi"
  Security_Profile_ID: "amzn1.application.83ebaddc9a8b45cdbae939a0f72f87c9"
  refresh_token: "replaced_for_reasons"

# See https://github.com/alexa-pi/AlexaPi/wiki/Triggers for the description of these options
triggers:
  platform:
    enabled: true
    voice_confirm: false

    # one of: oneshot-vad, continuous, continuous-vad
    event_type: "oneshot-vad"

    # only for "continuous" event_types
    long_press:
      # system command to run after pressing the button for _duration_ number of seconds
      # empty to disable
      command: ""

      # duration in seconds
      duration: 10

[... 101 lines cut ...]
    flash_duration: 500
    flash_frequency: 8
    # Use Strobe effect instead of default effect or static color
    flash_state_playback: true
    flash_state_processing: false
    flash_state_recording: false
    # Visualize playback state
    indicate_playback: true
    #Also print json request/response when in debug mode
    verbose: false

  serial:
    port: "/dev/ttyACM0"
    baudrate: 9600
    messages:
      trigger: ""
      playback_start: "g"
      playback_end: "x"
      recording_start: ""
      recording_end: ""
      processing_start: ""
      processing_end: ""
      failure: ""
      success: ""
`

mcosti, 2017-06-02:
Anyone? :(

maso27, 2017-06-02:
If you look at your existing _config.yaml_ and compare it with the template, you'll see that there's a "language" field missing.

Was this by any chance an upgrade, where you opted to keep your old config file?

Regardless, you can fix this particular issue by adding that line to your _config.yaml_.

I suspect that you may find further issues though, and would recommend copying the config template into its place and putting your particular license information and config settings into it.

mcosti, 2017-06-02:
I am looking at the template and I can't seem to find the field "language"
I tried to add language: "english" on the first indentation of the config yaml, but nothing.

maso27, 2017-06-02:
Huh. Yeah, you're right.
So what did you post yesterday, before your actual _config.yaml_ file? It looks like you've deleted the post.

I noticed the language line in that, listed as 'en_us' and figured that was it.

Sorry.

mcosti, 2017-06-02:
That was me copy pasting the config.template from my local directory :D

mcosti, 2017-06-02:
Oh. I added language: "en-US" under pocketsphinx and now it gives me the same error, but now with dictionary.

maso27, 2017-06-02:
Yeah, something's not matching in your setup. That config template you ppsted had a "dict" line that would fix that. Is this an upgrade or something like that? Adjustments have been made to the structure over time. 

That config template you originally posted has all this stuff in it.  Might compare the two to see what else is missing.

mcosti, 2017-06-02:
I added dictionary under pocketsphinx as 
dictionary: "en-US"

Now it gives me this. I checked my template, I don't have either dict / dictionary / language.
```
Current configuration:
[NAME]                  [DEFLT]         [VALUE]
-agc                    none            none
-agcthresh              2.0             2.000000e+00
-allphone
-allphone_ci            no              no
-alpha                  0.97            9.700000e-01
-ascale                 20.0            2.000000e+01
-aw                     1               1
-backtrace              no              no
-beam                   1e-48           1.000000e-48
-bestpath               yes             yes
-bestpathlw             9.5             9.500000e+00
-ceplen                 13              13
-cmn                    live            live
-cmninit                40,3,-1         40,3,-1
-compallsen             no              no
-debug                                  0
-dict                                   /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-US
-dictcase               no              no
-dither                 no              no
-doublebw               no              no
-ds                     1               1
-fdict
-feat                   1s_c_d_dd       1s_c_d_dd
-featparams
-fillprob               1e-8            1.000000e-08
-frate                  100             100
-fsg
-fsgusealtpron          yes             yes
-fsgusefiller           yes             yes
-fwdflat                yes             yes
-fwdflatbeam            1e-64           1.000000e-64
-fwdflatefwid           4               4
-fwdflatlw              8.5             8.500000e+00
-fwdflatsfwin           25              25
-fwdflatwbeam           7e-29           7.000000e-29
-fwdtree                yes             yes
-hmm                                    /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-US
-input_endian           little          little
-jsgf
-keyphrase
-kws                                    /opt/AlexaPi/src/keyphrase.list
-kws_delay              10              10
-kws_plp                1e-1            1.000000e-01
-kws_threshold          1               1.000000e+00
-latsize                5000            5000
-lda
-ldadim                 0               0
-lifter                 0               0
-lm
-lmctl
-lmname
-logbase                1.0001          1.000100e+00
-logfn
-logspec                no              no
-lowerf                 133.33334       1.333333e+02
-lpbeam                 1e-40           1.000000e-40
-lponlybeam             7e-29           7.000000e-29
-lw                     6.5             6.500000e+00
-maxhmmpf               30000           30000
-maxwpf                 -1              -1
-mdef
-mean
-mfclogdir
-min_endfr              0               0
-mixw
-mixwfloor              0.0000001       1.000000e-07
-mllr
-mmap                   yes             yes
-ncep                   13              13
-nfft                   512             512
-nfilt                  40              40
-nwpen                  1.0             1.000000e+00
-pbeam                  1e-48           1.000000e-48
-pip                    1.0             1.000000e+00
-pl_beam                1e-10           1.000000e-10
-pl_pbeam               1e-10           1.000000e-10
-pl_pip                 1.0             1.000000e+00
-pl_weight              3.0             3.000000e+00
-pl_window              5               5
-rawlogdir
-remove_dc              no              no
-remove_noise           yes             yes
-remove_silence         yes             yes
-round_filters          yes             yes
-samprate               16000           1.600000e+04
-seed                   -1              -1
-sendump
-senlogdir
-senmgau
-silprob                0.005           5.000000e-03
-smoothspec             no              no
-svspec
-tmat
-tmatfloor              0.0001          1.000000e-04
-topn                   4               4
-topn_beam              0               0
-toprule
-transform              legacy          legacy
-unit_area              yes             yes
-upperf                 6855.4976       6.855498e+03
-uw                     1.0             1.000000e+00
-vad_postspeech         50              50
-vad_prespeech          20              20
-vad_startspeech        10              10
-vad_threshold          2.0             2.000000e+00
-var
-varfloor               0.0001          1.000000e-04
-varnorm                no              no
-verbose                no              no
-warp_params
-warp_type              inverse_linear  inverse_linear
-wbeam                  7e-29           7.000000e-29
-wip                    0.65            6.500000e-01
-wlen                   0.025625        2.562500e-02

INFO: feat.c(715): Initializing feature stream to type: '1s_c_d_dd', ceplen=13, CMN='live', VARNORM='no', AGC='none'
ERROR: "acmod.c", line 83: Folder '/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-US' does not contain acoustic model definition 'mdef'
Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 646, in <module>
    triggers.setup()
  File "/opt/AlexaPi/src/alexapi/triggers/__init__.py", line 20, in setup
    trigger.setup()
  File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 49, in setup
    self._decoder = Decoder(ps_config)
  File "/usr/local/lib/python2.7/dist-packages/pocketsphinx/pocketsphinx.py", line 228, in __init__
    this = _pocketsphinx.new_Decoder(*args)
RuntimeError: new_Decoder returned -1

```

mcosti, 2017-06-02:
You were right.
There was a language in my config, and a dictionary.
https://github.com/xtools-at/AssistantPi/blob/master/src/config.template.yaml

Thank you for your help!

closed 2017-06-02 by mcosti