alexa-pi/AlexaPi #250: Running AlexaPi at Startup

hankp46, 2017-06-01:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
```

**Your hardware platform and model you are running on:**
```

```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
`Cannot get AlexaPI to run at startup. It works fine when I  start manually.

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
``Here's what shows at startup
pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─usergroup-root.conf
   Active: active (running) since Wed 2017-05-31 22:04:09 EDT; 52s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 536 (python)
   CGroup: /system.slice/AlexaPi.service
           └─536 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

May 31 22:04:42 raspberrypi python[536]: File "/opt/AlexaPi/src/alexapi/trig...d
May 31 22:04:42 raspberrypi python[536]: inp = alsaaudio.PCM(alsaaudio.PCM_C...)
May 31 22:04:42 raspberrypi python[536]: ALSAAudioError: Connection refused ...]
May 31 22:04:43 raspberrypi python[536]: ALSA lib pcm.c:2239:(snd_pcm_open_n...r
May 31 22:04:43 raspberrypi python[536]: [74100fa8] alsa audio output error:...y
May 31 22:04:43 raspberrypi python[536]: [74100fa8] core audio output error:...d
May 31 22:04:43 raspberrypi python[536]: [74100fa8] core audio output error:...:
May 31 22:04:43 raspberrypi python[536]: No such file or directory.
May 31 22:04:43 raspberrypi python[536]: [74100fa8] core audio output error:...l
May 31 22:04:43 raspberrypi python[536]: [73d12a00] core decoder error: fail...t

Warning: Unit file changed on disk, 'systemctl daemon-reload' recommended.
Hint: Some lines were ellipsized, use -l to show in full.
pi@raspberrypi:~ $ /opt/AlexaPi/src/main.py
2017-05-31 22:05:46 INFO: Checking Internet Connection ...
2017-05-31 22:05:46 INFO: Connection OK
2017-05-31 22:05:46 INFO: AVS token: Requesting a new one
2017-05-31 22:05:47 INFO: AVS token: Obtained successfully`
It then works perfectly
```

renekliment, 2017-06-13:
> ALSAAudioError: Connection refused

Did you follow the PulseAudio guide? https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#pulseaudio

renekliment, 2017-10-08:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment