alexa-pi/AlexaPi #252: How do I execute a custom command when Alexa is triggered?

mcosti, 2017-06-04:
Hello. I have some wireless lights that have python libraries to control them with.
Can you please tell me how do I trigger the light when Alexa is ready to listen to me?

Thank you very much!

EmerickH, 2017-06-10:
Hello,
Create first Python scripts to on/off your lights
Then in your config file there's a category to start commands before/after trigger.
Example:
```
# Commands to run before and after an interaction begins. Leave empty to disable.
event_commands:

  # Command to run at the start of AlexaPi
  startup: ""

  # Command to run after a trigger is detected but before an interaction begins
  pre_interaction: "python /my/folder/startlight.py"

  # Command to run after the interaction has completed
  post_interaction: "python /my/folder/stoplight.py"

  # Command to run during a clean shutdown of AlexaPi
  shutdown: ""
```

Emerick

closed 2017-10-08 by renekliment

asktoby, 2017-11-12:
I don't see in your code snippet where you would tell Alexa what the watchword is?
It looks like it will start the lights before *every* Alexa interaction and stop the lights after *every* interaction.

- How would you get it to respond to "Alexa, Lumos!" by turning on the lights?
- And then separately "Alexa, Morsmorde!" to turn them off again?