alexa-pi/AlexaPi #253: ERROR: [6bf004d8] alsa audio output error:...r

aguiducci, 2017-06-04:
Good morning!

I made yesterday a fresh installation of Raspian and AlexaPi but i can't make it works, following error:

ALSA INFO AT: http://www.alsa-project.org/db/?f=ff14c276b1d87c116b56c193c3545ba724d37bdd

**Your OS (including version) where you are running AlexaPi:**
```
Raspbian Jessie with PIXEL 4.4

**Your hardware platform and model you are running on:**
```
Raspberry P3

**Description of problem:**
```
Audio issue, Alexa service starts but with following error: [6bf004d8] alsa audio output error:...r
Alexa doesn't speaks

- start service:
sudo systemctl start AlexaPi.service

- check status:
sudo systemctl status AlexaPi.service

pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: active (running) since Sun 2017-06-04 09:04:17 UTC; 6min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 586 (python)
   CGroup: /system.slice/AlexaPi.service
           └─586 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

Jun 04 09:04:17 raspberrypi systemd[1]: Starting Alexa client for all your d....
Jun 04 09:04:17 raspberrypi systemd[1]: Started Alexa client for all your de....
Jun 04 09:04:26 raspberrypi python[586]: INFO: Checking Internet Connection ...
Jun 04 09:04:27 raspberrypi python[586]: INFO: Connection OK
Jun 04 09:04:27 raspberrypi python[586]: INFO: AVS token: Requesting a new one
Jun 04 09:04:28 raspberrypi python[586]: INFO: AVS token: Obtained successfully
Jun 04 09:04:30 raspberrypi python[586]: [6bf004d8] alsa audio output error:...r
Hint: Some lines were ellipsized, use -l to show in full.

```
System audio works fine because I've checked it with arecord and then aplay. 
aplay /usr/share/sounds/alsa/Front_Center.wav

arecord -f S16_LE -r 16000 -D pulse test.wav
aplay -f S16_LE -r 16000 test.wa

**Problem-relevant `config.yaml` entries:**
```
input_device: "pulse"
output: "pulse"
output_device: ""

Hope someone can help and I posted all the relevant and usefull info.
Thank you all!

EmerickH, 2017-06-10:
Hello,
can you copy/paste the output of:
`alsa -l` and `alsa -L`
And the full log:
```
sudo systemctl stop AlexaPi.service
/opt/AlexaPi/src/main.py -d
```
(you can exit with Ctrl+C)
Are you using VLC or SOX?

Emerick

aguiducci, 2017-06-16:
pi@raspberrypi:~ $ /opt/AlexaPi/src/main.py -d
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone				
-allphone_ci		no		no
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7.000000e-29
-fwdtree		yes		yes
-hmm					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us
-input_endian		little		little
-jsgf					
-keyphrase				alexa
-kws					
-kws_delay		10		10
-kws_plp		1e-1		1.000000e-01
-kws_threshold		1		1.000000e-10
-latsize		5000		5000
-lda					
-ldadim			0		0
-lifter			0		22
-lm					
-lmctl					
-lmname					
-logbase		1.0001		1.000100e+00
-logfn					
-logspec		no		no
-lowerf			133.33334	1.300000e+02
-lpbeam			1e-40		1.000000e-40
-lponlybeam		7e-29		7.000000e-29
-lw			6.5		6.500000e+00
-maxhmmpf		30000		30000
-maxwpf			-1		-1
-mdef					
-mean					
-mfclogdir				
-min_endfr		0		0
-mixw					
-mixwfloor		0.0000001	1.000000e-07
[... 80 lines cut ...]
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -225, delay 10)
2017-06-16 16:24:34 DEBUG: Setting up playback handler: VlcHandler
2017-06-16 16:24:35 INFO: Checking Internet Connection ...
2017-06-16 16:24:35 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-06-16 16:24:36 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2017-06-16 16:24:37 INFO: Connection OK
2017-06-16 16:24:37 INFO: AVS token: Requesting a new one
2017-06-16 16:24:37 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-06-16 16:24:38 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 976
2017-06-16 16:24:38 INFO: AVS token: Obtained successfully
2017-06-16 16:24:38 DEBUG: Stopping audio play
2017-06-16 16:24:38 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-06-16 16:24:38 DEBUG: Player State: State.Opening
2017-06-16 16:24:38 DEBUG: Player State: State.Playing
2017-06-16 16:24:38 DEBUG: Started play. 
[6bf004d8] alsa audio output error: cannot estimate delay: Input/output error
INFO: cmn_live.c(88): Update from < 41.00 -5.29 -0.12  5.09  2.48 -4.07 -1.37 -1.78 -5.08 -2.05 -6.45 -1.42  1.17 >
INFO: cmn_live.c(105): Update to   < 19.86 -1.02 -3.42  1.28  2.80  1.39 -3.45  4.01  3.22  0.58 -1.62  2.71  2.08 >
2017-06-16 16:25:11 DEBUG: Player State: State.Ended
2017-06-16 16:25:11 DEBUG: Finished play.

EmerickH, 2017-06-17:
Hello,
> **can you copy/paste the output of:
> `alsa -l` and `alsa -L`**
> [...]
> **Are you using VLC or SOX?**

And what output device are you using in your `config.yaml`?

Emerick

PeterH888, 2017-06-17:
I too am having the exact same problem.
alsa -l
-bash:: command not found

EmerickH, 2017-06-17:
@PeterH888 yes sorry it's `aplay -L` not alsa

PeterH888, 2017-06-17:
no problems. here are the results from aplay -l and aplay -L
I have also included the results of arecord -l and arecord -L as my microphone is working but I get no response when using the wake word.

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@raspberrypi:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
default
    Playback/recording through the PulseAudio sound server
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions

pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: U0x46d0x89d [USB Device 0x46d:0x89d], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

pi@raspberrypi:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
default
    Playback/recording through the PulseAudio sound server
sysdefault:CARD=U0x46d0x89d
    USB Device 0x46d:0x89d, USB Audio
    Default Audio Device
front:CARD=U0x46d0x89d,DEV=0
    USB Device 0x46d:0x89d, USB Audio
    Front speakers
surround21:CARD=U0x46d0x89d,DEV=0
    USB Device 0x46d:0x89d, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=U0x46d0x89d,DEV=0
    USB Device 0x46d:0x89d, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=U0x46d0x89d,DEV=0
    USB Device 0x46d:0x89d, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=U0x46d0x89d,DEV=0
    USB Device 0x46d:0x89d, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=U0x46d0x89d,DEV=0
    USB Device 0x46d:0x89d, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=U0x46d0x89d,DEV=0
    USB Device 0x46d:0x89d, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=U0x46d0x89d,DEV=0
    USB Device 0x46d:0x89d, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=U0x46d0x89d,DEV=0
    USB Device 0x46d:0x89d, USB Audio
    Direct sample mixing device
dsnoop:CARD=U0x46d0x89d,DEV=0
    USB Device 0x46d:0x89d, USB Audio
    Direct sample snooping device
hw:CARD=U0x46d0x89d,DEV=0
    USB Device 0x46d:0x89d, USB Audio
    Direct hardware device without any conversions
plughw:CARD=U0x46d0x89d,DEV=0
    USB Device 0x46d:0x89d, USB Audio
    Hardware device with all software conversions

Thanks in advance for checking these out

EmerickH, 2017-06-18:
Ok @PeterH888 Alsa seems to detect with no problems your devices.
Try this: https://github.com/alexa-pi/AlexaPi/issues/242#issuecomment-307614824

PeterH888, 2017-06-18:
Thanks EmerickH, I'll give that try.

PeterH888, 2017-06-20:
Hi EmerickH, I have removed Raspian Pixel and installed Raspian Lite. Installed Alexapi from git-hub instructions, configured alsa to my usb mic and 3.5 speaker jack and on boot I hear "hello". When I use the wake word "Alexa", Alexa replies with "yes". When I ask a question I get no response but after about 10 secs I can again use the wake word and Alexa replies with "yes". What am I missing here.
These are the aplay -l and arecord -l 

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Microphone [Logitech USB Microphone], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

PeterH888, 2017-06-20:
This is the result I get from   
/opt/AlexaPi/src/main.py -d

2017-06-20 22:23:09 DEBUG: Stopping audio play
2017-06-20 22:23:09 DEBUG: Playing audio: /opt/AlexaPi/src/resources/alexayes.mp3
2017-06-20 22:23:09 DEBUG: Player State: State.Opening
2017-06-20 22:23:09 DEBUG: Player State: State.Playing
2017-06-20 22:23:09 DEBUG: Started play.
2017-06-20 22:23:09 DEBUG: Player State: State.Ended
2017-06-20 22:23:09 DEBUG: Finished play.
2017-06-20 22:23:09 DEBUG: Setting up recording
2017-06-20 22:23:09 DEBUG: Start recording
2017-06-20 22:23:09 DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com
2017-06-20 22:23:10 DEBUG: Start sending speech to Alexa Voice Service
2017-06-20 22:23:18 DEBUG: End recording
2017-06-20 22:23:18 DEBUG: Finished sending speech to Alexa Voice Service
2017-06-20 22:23:19 DEBUG: Processing Request Response...
2017-06-20 22:23:19 DEBUG: Request Response is null (This is OKAY!)

EmerickH, 2017-06-20:
Hello @PeterH888,
what `input_device` are you using in your config?
Can you copy/paste the output of: `arecord -L` ?

Emerick

PeterH888, 2017-06-20:
here is the result or  arecord -L
pi@raspberrypi:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=Microphone
    Logitech USB Microphone, USB Audio
    Default Audio Device
sysdefault:CARD=Microphone
    Logitech USB Microphone, USB Audio
    Default Audio Device
front:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    Front speakers
surround21:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    Direct sample mixing device
dsnoop:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    Direct sample snooping device
hw:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    Hardware device with all software conversions

EmerickH, 2017-06-20:
So you must put `input_device: "plughw:CARD=Microphone,DEV=0"` in you config file

PeterH888, 2017-06-20:
thank you, I'll do that now.

# See https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging
# to get a better idea of how to set things in here

# Level of log output
# default: INFO
# set to DEBUG for verbose logging
logging: INFO

sound:
  # Name of your ALSA microphone/soundcard in arecord -L
  input_device: "plughw:CARD=Microphone,DEV=0"
  # This is here if none of the listed device: "raspberrypi"
  # This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false

  playback_handler: "vlc"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "alsa"
  # output device: "default"
  # currently supports ALSA only
  output_device: "default"

  # if left empty, won't be passed to the underlying library / command if it supports it
  # if it doesn't 100% will be used
  default_volume: 80

  # Means the same as above, but is used for media (playing music for example)
  # Leave blank to use the same setting as for speech content
  media_output:
  media_output_device:
  media_default_volume:

  # Some device: "raspberrypi"
  # value is the number of seconds of silence to pad the beginning and end of the audio. Only
  # the sox handler supports this configuration option.
  playback_padding: 0

# Amazon Alexa settings
alexa:
  Client_ID: "amzn1.application-oa2-client.3050cd75678b4833b2ded8003a6bb5e6"
  Client_Secret: "7c1b143d77c23c7cfb214254eb066ffa9abc89e8631d5160ad758f3b44acf125"
  Device_Type_ID: "Alexapi"
  Security_Profile_Description: "Alexapi"
  Security_Profile_ID: "amzn1.application.f3a7aa3ebe0c41fd9d18f9b157d001fe"
  refresh_token: "Atzr|IwEBILumV9ae48i2pr_hvJK0wCB26d_FlHumTaB9vgZsjz_Dcu2NaeGEnyp6IAiYRyXcFwMXXYwgBIyMFXKiCun0kXEqWwaqVl9hX4KKUfQsj0VE_BTg2bwwghko-X2UdLRW3S8LbuOq3U6_2hUcVo29OxMU6Cnl2i9$

# See https://github.com/alexa-pi/AlexaPi/wiki/Triggers for the description of these options
triggers:
  
this config gives me the same result  

2017-06-20 22:59:22 DEBUG: Stopping audio play
2017-06-20 22:59:22 DEBUG: Playing audio: /opt/AlexaPi/src/resources/alexayes.mp3
2017-06-20 22:59:22 DEBUG: Player State: State.Opening
2017-06-20 22:59:22 DEBUG: Player State: State.Playing
2017-06-20 22:59:22 DEBUG: Started play.
2017-06-20 22:59:22 DEBUG: Player State: State.Ended
2017-06-20 22:59:22 DEBUG: Finished play.
2017-06-20 22:59:22 DEBUG: Setting up recording
2017-06-20 22:59:22 DEBUG: Start recording
2017-06-20 22:59:22 DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com
2017-06-20 22:59:24 DEBUG: Start sending speech to Alexa Voice Service
2017-06-20 22:59:25 DEBUG: End recording
2017-06-20 22:59:25 DEBUG: Finished sending speech to Alexa Voice Service
2017-06-20 22:59:26 DEBUG: Processing Request Response...
2017-06-20 22:59:26 DEBUG: Request Response is null (This is OKAY!)

PeterH888, 2017-06-20:
sudo nano /etc/opt/AlexaPi/config.yaml

 Name of your ALSA microphone/soundcard in arecord -L
  input_device: "plughw:CARD=Microphone,DEV=0"

 playback_handler: "vlc"

 output: "alsa"

output_device: "default"

still gets me no response

2017-06-20 22:59:22 DEBUG: Stopping audio play
2017-06-20 22:59:22 DEBUG: Playing audio: /opt/AlexaPi/src/resources/alexayes.mp3
2017-06-20 22:59:22 DEBUG: Player State: State.Opening
2017-06-20 22:59:22 DEBUG: Player State: State.Playing
2017-06-20 22:59:22 DEBUG: Started play.
2017-06-20 22:59:22 DEBUG: Player State: State.Ended
2017-06-20 22:59:22 DEBUG: Finished play.
2017-06-20 22:59:22 DEBUG: Setting up recording
2017-06-20 22:59:22 DEBUG: Start recording
2017-06-20 22:59:22 DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com
2017-06-20 22:59:24 DEBUG: Start sending speech to Alexa Voice Service
2017-06-20 22:59:25 DEBUG: End recording
2017-06-20 22:59:25 DEBUG: Finished sending speech to Alexa Voice Service
2017-06-20 22:59:26 DEBUG: Processing Request Response...
2017-06-20 22:59:26 DEBUG: Request Response is null (This is OKAY!)

PeterH888, 2017-06-20:
This is what I get with: arecord

pi@raspberrypi:~ $ arecord -f S16_LE -r 16000 -D plughw:CARD=Microphone,DEV=0 test.wav
arecord: main:722: audio open error: Device or resource busy

EmerickH, 2017-06-20:
@PeterH888 have you closed the service before?
`sudo service AlexaPi stop`

PeterH888, 2017-06-20:
not that I can remember
I have used  sudo systemctl stop AlexaPi.service to stop before degugging

PeterH888, 2017-06-20:
does the last line here make any sense to you. It appears as if AlsaAudio is not picking up my microphone.

2017-06-21 00:00:41 DEBUG: Setting up playback handler: VlcHandler
2017-06-21 00:00:41 INFO: Checking Internet Connection ...
2017-06-21 00:00:41 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-06-21 00:00:43 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2017-06-21 00:00:43 INFO: Connection OK
2017-06-21 00:00:43 INFO: AVS token: Requesting a new one
2017-06-21 00:00:43 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-06-21 00:00:44 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 977
2017-06-21 00:00:44 INFO: AVS token: Obtained successfully
2017-06-21 00:00:44 DEBUG: Stopping audio play
2017-06-21 00:00:44 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 55, in thread
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, self._config['sound']['input_device'])
ALSAAudioError: Input/output error [plughw:CARD=Microphone,DEV=0]

PeterH888, 2017-06-21:
@EmerickH   So I have done as suggested and installed pulse audio and this is what I get from here.

pi@raspberrypi:/tmp $ sudo journalctl -u AlexaPi.service
-- Logs begin at Wed 2017-06-21 18:46:28 AEST, end at Wed 2017-06-21 18:47:30 AEST. --
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
[... cut ...]
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error
Jun 21 18:46:50 raspberrypi pulseaudio[846]: [alsa-source-USB Audio] alsa-source.c: Failed to set hardware parameters: Input/output error

renekliment, 2017-10-08:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment

AydinSyed, 2019-01-19:
did you guys ever solve this?