alexa-pi/AlexaPi #254: AlexaToken

muneerkk66, 2017-06-08:
Hi Team,

 I'm trying to integrate Alexa app on my IOS. I have integrated Amazon login option for end user to authorise the account, after authorisation I got the accesstoken from Amazon API. I am sending that token for calling AVS service this is working as expected.I already link the user to my own system successfully. 

   Now the issue is when i receive the request from AWS lambda function i received the another token. Why it is like this ?

EmerickH, 2017-06-13:
Hello,
You're on the AlexaPi project, **we're not Amazon**!
I see that you've already been "dropped" from another project: https://github.com/rlisle/Patriot/issues/8
Please close this issue and ask to **the Amazon team, not OpenSource projects** for your questions!

Emerick

renekliment, 2017-06-13:
Yup, not our issue. Moreover, we don't do anything with Lambda (so at least I don't know). Ask in an appropriate place. Cheers.

closed 2017-06-13 by renekliment

muneerkk66, 2017-06-13:
Thanks Team