alexa-pi/AlexaPi #255: PulseAudio: Unable to connect: Connection refused

Hinoy, 2017-06-08:
**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

http://www.alsa-project.org/db/?f=163054592e7d42c08ebf5e5a29bdcb57f7bad82d

**Your OS (including version) where you are running AlexaPi:**

Raspbian GNU/Linux 8.0 (jessie)
(Linux raspberrypi 4.9.24-v7+)
Running AlexaPi in /opt

**Your hardware platform and model you are running on:**

Raspberry Pi 2 Model B

**Python release (`python2 --version`):**

Python 2.7.9

**Description of problem:**
After a few seconds where Alexa speaks and hear me it stops (The mirror still works too) but in the debugging it still shows the "heatbeat" sucess)
Basically she just stops hearing me (therefore stops responding)
Here is the incriminated log :
```
2017-06-08 15:46:51 DEBUG: Setting up playback handler: VlcHandler
[01a845e0] pulse audio output error: PulseAudio server connection failure: Connection refused
[01a80588] pulse audio output error: PulseAudio server connection failure: Connection refused
[...]
2017-06-08 15:46:59 DEBUG: Sending MM Heatbeat
2017-06-08 15:47:00 DEBUG: Response: {"status":"success"}
2017-06-08 15:47:00 DEBUG: Stopping audio play
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

2017-06-08 15:47:01 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-06-08 15:47:01 DEBUG: Player State: State.Opening
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 55, in thread
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, self._config['sound']['input_device'])
ALSAAudioError: Connection refused [pulse]

2017-06-08 15:47:01 DEBUG: Player State: State.Playing
2017-06-08 15:47:01 DEBUG: Started play.
2017-06-08 15:47:01 DEBUG: Indicate Start Playing
2017-06-08 15:47:01 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=playback
2017-06-08 15:47:01 DEBUG: Response: {"status":"success"}
[741032a0] pulse audio output error: PulseAudio server connection failure: Connection refused
2017-06-08 15:47:04 DEBUG: Sending MM Heatbeat
```

**Expected:**
Should say yes when I call for her, like she does when it's still booting up. Shouldn't see any audio related issues...

**Problem-relevant `config.yaml` entries:**
```
Possibly the asound config ?
```

**Steps to reproduce:**

No idea really

**Traceback (if applicable):**
```

```

**Additional info:**
```
Pretty sure the audio debug log will be the key (I might have messed up some audio settings)
```

EmerickH, 2017-06-10:
Hello,
And are you shure to be in the audio group?
```
adduser pi audio
sudo reboot
```
If it's not working, are you shure that the pulseaudio server is running?

Emerick

Hinoy, 2017-06-11:
Pi user is already part of "audio" group.
Not sure about the pulseaudio server, I'll check on that today and get back to you  (`sudo apt-get install pulseaudio` again etc)

EmerickH, 2017-06-13:
Hello,
How do you start AlexaPi?
You must start it as `pi` user, not as `root`.

Emerick

renekliment, 2017-06-13:
@Hinoy @EmerickH AlexaPi (the service) is run under the user `alexapi` so you need to make sure you do `sudo adduser alexapi pulse-access`.

Hinoy, 2017-06-13:
@EmerickH I do start it as Pi, sadly not helping
@renekliment I tried that, didn't do much sadly
Just to let you guys know, I figured it might not come from how my sounds cards and all because Alexa starts, works for a bit, stops working but 2 HOURS later she finally react and become fluid again answering to everything smoothly 
`juin 12 22:18:43 raspberrypi python[789]: INFO: AVS token: Obtained successfully
juin 12 22:18:43 raspberrypi python[789]: ERROR: URLError: [Errno 111] Connection refused
juin 12 22:18:44 raspberrypi python[789]: ERROR: URLError: [Errno 111] Connection refused
juin 13 14:02:44 raspberrypi python[789]: ERROR: URLError: [Errno 111] Connection refused
juin 13 14:02:44 raspberrypi python[789]: ERROR: URLError: [Errno 111] Connection refused
juin 13 14:02:44 raspberrypi python[789]: ERROR: URLError: [Errno 111] Connection refused
juin 13 14:02:54 raspberrypi python[789]: INFO: AVS token: Requesting a new one
juin 13 14:03:04 raspberrypi python[789]: INFO: AVS token: Obtained successfully
`

Hinoy, 2017-06-13:
Eveything now works in debug mode. But not in AlexaPi service... Whats the difference between the two?

EmerickH, 2017-06-13:
Yes @renekliment, you're right.
Effectively the AlexaPi service uses `alexapi` user and when you start it manually, it's with the `pi` user, that's the difference.
@Hinoy, try `sudo adduser alexapi audio`

Emerick

Hinoy, 2017-06-13:
I figured how to fix the issue and I guess it's linked to what you guys said ! 
Basically when the Raspberry boots up it fails but when I do a restart of the service it's back to normal (like in debug everything works) 
I'll do that in the next hour or so and come back before, hopefully, closing the issue (thanks to both of you)

Hinoy, 2017-06-13:
alexapi was already part of both pulse-access and audio. I even added to every group pi as except users, not enough, same error. I m going to keep digging on my side.

EmerickH, 2017-06-14:
@Hinoy can you give us your `AlexaPi.service` file?
It should be in `/etc/systemd/system/`

Emerick

Hinoy, 2017-06-17:
I am sorry there is no such thing here ... I have a .service (including pulseaudio) but no alexapi.service
BUT I have a alexapi.service.d (a folder), which is empty.
(FYI the installer was set to default location)
But anyway I tried a locate AlexaPi.service and found those : 
`pi@raspberrypi:~ $ sudo updatedb
pi@raspberrypi:~ $ locate AlexaPi.service
/etc/systemd/system/AlexaPi.service.d
/etc/systemd/system/default.target.wants/AlexaPi.service
/opt/AlexaPi/src/scripts/AlexaPi.service
/usr/lib/systemd/system/AlexaPi.service`
I assumed it was the second one so here it is : 
https://pastebin.com/WsFiZt2Y
Shall I modify the user to pi to make things easier ? Or is alexapi user really usefull ? 
Thanks for everything by the way (Thanks to you, and what you told me about user perm, I got alexa working for my college exam with a service restart as pi)

EmerickH, 2017-06-17:
@Hinoy, no sorry it's the `/usr/lib/systemd/system/AlexaPi.service` one 

First try:
```
sudo systemctl daemon-reload
sudo systemctl enable AlexaPi.service
sudo reboot
```

If it's not working, look at theses lines in `/usr/lib/systemd/system/AlexaPi.service`:
```
User=??
Group=??
```
If ?? is `alexapi`, try to change it to `pi`, if it's not change to `alexapi`.
Don't forget to do `systemctl daemon-reload` after changing!

Emerick

Hinoy, 2017-06-17:
It's okay ! :)
1. Not working (same error as before).
2. I didn't know what to put in group (It was alexapi for both user and group) so I put pi in both. Not working with same error as before.
3. Not working :
`juin 17 17:18:12 raspberrypi systemd[1]: Started Alexa client for all your devices.
juin 17 17:18:18 raspberrypi python[850]: Home directory not accessible: Permission denied
juin 17 17:18:18 raspberrypi python[850]: Home directory not accessible: Permission denied
juin 17 17:18:46 raspberrypi python[850]: Home directory not accessible: Permission denied
juin 17 17:18:46 raspberrypi python[850]: Home directory not accessible: Permission denied
juin 17 17:18:46 raspberrypi python[850]: INFO: Magic Mirror HTTP Server - :5055
juin 17 17:18:46 raspberrypi python[850]: INFO: Checking Internet Connection ...
juin 17 17:18:47 raspberrypi python[850]: INFO: Connection OK
juin 17 17:18:47 raspberrypi python[850]: INFO: AVS token: Requesting a new one
juin 17 17:18:49 raspberrypi python[850]: INFO: AVS token: Obtained successfully
juin 17 17:18:49 raspberrypi python[850]: ERROR: URLError: [Errno 111] Connection refused
juin 17 17:18:49 raspberrypi python[850]: Home directory not accessible: Permission denied
juin 17 17:18:50 raspberrypi python[850]: ERROR: URLError: [Errno 111] Connection refused
juin 17 17:18:50 raspberrypi python[850]: Home directory not accessible: Permission denied
` That's a new error.
I am incredibly busy right now but wanted to go back to you as soon as possible, I'll edit this with more details tomorrow

renekliment, 2017-10-08:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment

jaserjasar, 2017-10-29:
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

2017-10-29 09:18:40,222 listen.py:19 [DEBUG] Awaiting user input.