alexa-pi/AlexaPi #257: pocketsphinxtrigger error

ezar, 2017-06-11:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 763, in run
  File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 73, in thread
  File "/usr/local/lib/python2.7/dist-packages/pocketsphinx/pocketsphinx.py", line 285, in process_raw
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'Decoder_process_raw'

EmerickH, 2017-06-13:
Hello,

> Thank you for reporting an issue with **AlexaPi**.
> 
> In order to help troubleshooting, be sure to include the following information:
> 
> * **Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**
> 
> * **Your OS (including version) where you are running AlexaPi**
> 
> * **Your hardware platform and model you are running on**
> 
> * **Python release (`python2 --version`)**
>
> * **Your `config.yaml` (be sure to remove your Amazon keys)**
>
> * **Steps to reproduce**
> 
Thank you,
Emerick

ezar, 2017-06-13:
**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

http://www.alsa-project.org/db/?f=c06c9e25569d46621bb6cb0c92c08edf318ea32c

**Your OS (including version) where you are running AlexaPi:**
```
Raspbian Jessie Lite
```

**Your hardware platform and model you are running on:**
```
Raspberry 2
```

**Python release (`python2 --version`):**
```
Python 2.7.9
```

EmerickH, 2017-06-13:
Hello,
this seems to be a pocketsphinx problem, not AlexaPi
Try:
```
pip install --upgrade pip setuptools wheel
pip install --upgrade pocketsphinx
 ```

If it's not working, try to update to python3:
```
sudo apt-get remove --purge python*
sudo apt-get install python3
```
And relaunch the AlexaPi install script

Emerick

renekliment, 2017-06-13:
Don't purge python2 as they can co-exist and python2 might be a requirement for some system packages.

ezar, 2017-06-13:
2017-06-13 21:41:38 DEBUG: Initialising Magic Mirror platorm
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]                  [DEFLT]         [VALUE]
-agc                    none            none
-agcthresh              2.0             2.000000e+00
-allphone                               
-allphone_ci            no              no
-alpha                  0.97            9.700000e-01
-ascale                 20.0            2.000000e+01
-aw                     1               1
-backtrace              no              no
-beam                   1e-48           1.000000e-48
-bestpath               yes             yes
-bestpathlw             9.5             9.500000e+00
-ceplen                 13              13
-cmn                    live            batch
-cmninit                40,3,-1         41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17-compallsen             no              no
-debug                                  0
-dict                                   /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase               no              no
-dither                 no              no
-doublebw               no              no
-ds                     1               1
-fdict                                  
-feat                   1s_c_d_dd       1s_c_d_dd
-featparams                             
-fillprob               1e-8            1.000000e-08
-frate                  100             100
-fsg                                    
-fsgusealtpron          yes             yes
-fsgusefiller           yes             yes
-fwdflat                yes             yes
-fwdflatbeam            1e-64           1.000000e-64
-fwdflatefwid           4               4
-fwdflatlw              8.5             8.500000e+00
-fwdflatsfwin           25              25
-fwdflatwbeam           7e-29           7.000000e-29
-fwdtree                yes             yes
-hmm                                    /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us
-input_endian           little          little
-jsgf                                   
-keyphrase                              alexa
-kws                                    
-kws_delay              10              10
-kws_plp                1e-1            1.000000e-01
-kws_threshold          1               1.000000e-10
-latsize                5000            5000
-lda                                    
-ldadim                 0               0
-lifter                 0               22
-lm                                     
-lmctl                                  
-lmname                                 
-logbase                1.0001          1.000100e+00
-logfn                                  
-logspec                no              no
-lowerf                 133.33334       1.300000e+02
-lpbeam                 1e-40           1.000000e-40
-lponlybeam             7e-29           7.000000e-29
-lw                     6.5             6.500000e+00
-maxhmmpf               30000           30000
-maxwpf                 -1              -1
-mdef                                   
-mean                                   
-mfclogdir                              
-min_endfr              0               0
-mixw                                   
-mixwfloor              0.0000001       1.000000e-07
-mllr                                   
[... 93 lines cut ...]
2017-06-13 21:41:42 INFO: AVS token: Obtained successfully
2017-06-13 21:41:42 DEBUG: Starting Magic Mirror platform HTTP Server
2017-06-13 21:41:42 DEBUG: Starting Magic Mirror heartbeat with 5 second interval
2017-06-13 21:41:42 DEBUG: Sending MM Heatbeat
2017-06-13 21:41:42 DEBUG: Response: {"status":"success"}2017-06-13 21:41:42 DEBUG: Stopping audio play
2017-06-13 21:41:43 DEBUG: Stopped play. No SoX process to stop
2017-06-13 21:41:43 DEBUG: Indicate Stop Playing
2017-06-13 21:41:43 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
2017-06-13 21:41:43 DEBUG: Response: {"status":"success"}2017-06-13 21:41:43 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-06-13 21:41:43 DEBUG: Started play. sox -q /opt/AlexaPi/src/resources/hello.mp3 -t pulseaudio default vol -6 dB pad 0 0
2017-06-13 21:41:43 DEBUG: Indicate Start Playing
2017-06-13 21:41:43 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=playback
2017-06-13 21:41:43 DEBUG: Response: {"status":"success"}2017-06-13 21:41:45 DEBUG: Finished play. 
2017-06-13 21:41:45 DEBUG: Indicate Stop Playing
2017-06-13 21:41:45 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
2017-06-13 21:41:46 DEBUG: Response: {"status":"success"}2017-06-13 21:41:46 DEBUG: Indicating Success
2017-06-13 21:41:46 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=success
2017-06-13 21:41:46 DEBUG: Response: {"status":"success"}2017-06-13 21:41:47 DEBUG: Sending MM Heatbeat
2017-06-13 21:41:47 DEBUG: Response: {"status":"success"}^C2017-06-13 21:41:48 DEBUG: Cleaning up Magic Mirror platform
2017-06-13 21:41:48 DEBUG: Cleaning up playback handler
2017-06-13 21:41:48 DEBUG: Stopped play. [Errno 3] No such process
2017-06-13 21:41:48 DEBUG: Indicate Stop Playing
2017-06-13 21:41:48 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
2017-06-13 21:41:49 DEBUG: Response: {"status":"success"}INFO: cmn_live.c(88): Update from < 41.00 -5.29 -0.12  5.09  2.48 -4.07 -1.37 -1.78 -5.08 -2.05 -6.45 -1.42  1.17 >
INFO: cmn_live.c(105): Update to   < 44.82 -11.86 -7.02 -2.72 -7.96  0.93  3.32 -5.41 -9.14  0.32 -1.12 -0.98  2.14 >

EmerickH, 2017-06-14:
Same thing: https://github.com/alexa-pi/AlexaPi/issues/242#issuecomment-308348936

renekliment, 2017-10-14:
Closing due to inactivity. If there is still a problem:
- try latest Raspbian
- try the snowboy trigger (currently a PR)

closed 2017-10-14 by renekliment