alexa-pi/AlexaPi #265: [6c0004d8] alsa audio output error: cannot estimate delay: Input/output error

wiredwrx, 2017-06-19:
Getting the following error
"[6c0004d8] alsa audio output error: cannot estimate delay: Input/output error"

**Your OS (including version) where you are running AlexaPi:**
Raspbian Jessie/Pixel.  Updated today with apt-get update and apt-get install update

**Your hardware platform and model you are running on:**
Raspberry PI 3

**Python release (`python2 --version`):**
2.7.9

**Description of problem:**

Getting the following error
"[6c0004d8] alsa audio output error: cannot estimate delay: Input/output error"

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
sound:
  # Name of your ALSA microphone/soundcard in arecord -L
  input_device: "pulse"
  # This is here if none of the listed device: "raspberrypi"
  # This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false

  playback_handler: "vlc"

   # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "pulse"
  # output device: "raspberrypi"
  # currently supports ALSA only
  output_device: ""

  # if left empty, won't be passed to the underlying library / command if it su$
  # if it doesn't 100% will be used
  default_volume: 80

  # Means the same as above, but is used for media (playing music for example)
  # Leave blank to use the same setting as for speech content
  media_output:
  media_output_device:
  media_default_volume:

  # Some device: "raspberrypi"
  # value is the number of seconds of silence to pad the beginning and end of t$
  # the sox handler supports this configuration option.
  playback_padding: 0

**Additional info:**
This is a brand new install following the instructions form here "http://lifehacker.com/the-simplest-way-to-build-a-raspberry-pi-powered-amazon-1794218212"

wiredwrx, 2017-06-19:
Forgot to add this.  So, here it is.

!!################################
!!ALSA Information Script v 0.4.64
!!################################

!!Script ran on: Mon Jun 19 21:19:25 UTC 2017

!!Linux Distribution
!!------------------

ID_LIKE=debian

!!DMI Information
!!---------------

Manufacturer:      
Product Name:      
Product Version:   
Firmware Version:  

!!Kernel Information
!!------------------

Kernel release:    4.9.28-v7 
Operating System:  GNU/Linux
Architecture:      armv7l
Processor:         unknown
SMP Enabled:       Yes

!!ALSA Version
!!------------

Driver version:     k4.9.28-v7 
Library version:    1.0.28
Utilities version:  1.0.28

!!Loaded ALSA modules
!!-------------------

snd_bcm2835
snd_usb_audio

!!Sound Servers on this system
!!----------------------------

Pulseaudio:
      Installed - Yes (/usr/bin/pulseaudio)
      Running - Yes

Jack:
      Installed - Yes (/usr/bin/jackd)
      Running - No

!!Soundcards recognised by ALSA
!!-----------------------------

 0 [ALSA           ]: bcm2835 - bcm2835 ALSA
                      bcm2835 ALSA
 1 [Device         ]: USB-Audio - USB PnP Sound Device
                      C-Media Electronics Inc. USB PnP Sound Device at usb-3f980000.usb-1.4, full spe

!!Modprobe options (Sound related)
!!--------------------------------

snd_pcsp: index=-2
snd_usb_audio: index=-2
snd_atiixp_modem: index=-2
snd_intel8x0m: index=-2
snd_via82xx_modem: index=-2
[... 253 lines cut ...]
snd_rawmidi
snd_seq_device
evdev
brcmfmac
brcmutil
cfg80211
rfkill
snd_bcm2835
snd_pcm
snd_timer
snd
bcm2835_gpiomem
uio_pdrv_genirq
fixed
uio
i2c_dev
fuse
ipv6

!!ALSA/HDA dmesg
!!--------------

[    3.326020] hid-generic 0003:8086:0808.0003: input,hidraw2: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device] on usb-3f980000.usb-1.4/input2
[    3.521818] usbcore: registered new interface driver snd-usb-audio
[    3.806542] systemd-journald[142]: Received request to flush runtime journal from PID 1

EmerickH, 2017-06-20:
See https://github.com/alexa-pi/AlexaPi/issues/100#issuecomment-264739025
Excepting this error, is it working?

Emerick

wiredwrx, 2017-06-20:
It is not working.  When it boots, the unit does not say "Hello" nor does it respond to the hotword.

kis4karma, 2017-07-19:
I had the same exact issue. This got mine to work.
First try following this:
https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging
but specifically
1.  If you're using VLC handler set
sudo nano /etc/opt/AlexaPi/config.yaml
input_device: "pulse"
output: "pulse"
output_device: ""
2. Download VLC manually using this 'sudo apt-get install vlc'
3. Reboot

EmerickH, 2017-07-20:
Hello,
Maybe it's your pulse that's not correctly configured.
If you have a graphical interface, you can use pavucontrol. Else there's a CLI package but I don't remember the name.

Emerick

renekliment, 2017-10-08:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment