alexa-pi/AlexaPi #266: AlexaPi plays only first 2 seconds of Alexa response

Sailor676, 2017-06-20:
I have AlexaPi functioning on a RPi3 with Google AIY hat. All works perfectly except only the first 2 seconds of the Alexa response is actually played. The full response *is* received and saved correctly and the debug log [incorrectly?] shows player state timings that should play the entire response.

```
Jun 20 05:37:10 raspberrypi python[622]: DEBUG: Playing audio: file:///tmp/AlexaPi-runtime-G0vc7i/WeatherPrompt.175f1126-b234-4a47-9d3f-d9deaa5ef4b5_1001413565.mp3
Jun 20 05:37:10 raspberrypi python[622]: DEBUG: Player State: State.Opening
Jun 20 05:37:10 raspberrypi python[622]: DEBUG: Player State: State.Playing
Jun 20 05:37:10 raspberrypi python[622]: DEBUG: Started play.
Jun 20 05:37:10 raspberrypi python[622]: DEBUG: indicate_playback True
Jun 20 05:37:17 raspberrypi python[622]: DEBUG: Player State: State.Ended
Jun 20 05:37:17 raspberrypi python[622]: DEBUG: Finished play.
```

Any ideas?

Many thanks.

EmerickH, 2017-06-20:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi** before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**

**Python release (`python2 --version`):**

**Problem-relevant `config.yaml` entries:**
```
  input_device: "???"
  allow_unlisted_input_device: ???
  playback_handler: "???"
  output: "???"
  output_device: "???"
```

Emerick

Sailor676, 2017-06-20:
Thank you for a fast response. I appreciate any pointers.

OS: Raspbian GNU/Linux 8 (jessie)
Python2: Python 2.7.9 (default, Sep 17 2016, 20:26:04)
Relevant config.yaml entries:
```
sound:
  input_device: "default"
  allow_unlisted_input_device: false
  playback_handler: "vlc"
  output: "alsa"
  output_device: "softvol"
  default_volume: 60
```

Audio debugging script output:
[alsa-info.txt](https://github.com/alexa-pi/AlexaPi/files/1088670/alsa-info.txt)
Full config.yaml (anonomised):
[config.yaml.txt](https://github.com/alexa-pi/AlexaPi/files/1088677/config.yaml.txt)

maso27, 2017-06-20:
Don't know if this helps or not, but at one point we discovered that the AIY hat uses some pins that AlexaPi has designated as GPIO.

You might try changing platform to "dummy" and see if that helps.

Sailor676, 2017-06-20:
Thank you maso27. Yes, the platform is set to "dummy" in my config.yaml to avoid the GPIO conflict you refer to. (It didn't work at all with it set otherwise).

renekliment, 2017-10-14:
If this is still a problem, you can:
- try Raspbian Strech with a clean install of AlexaPI,
- try SoX.

closed 2017-10-14 by renekliment

Sailor676, 2017-10-15:
For information, I stopped using AlexaPi and instead used the Amazon javascript client which works flawlessly (once you discover how to autostart it - see YouTube).

shadowscott, 2018-03-14:
@Sailor676 can you provide instruction on how you got Alex's to run in the Google aiy voice kit? I'm currently using GassistPi to run google assistant but I want to also run Alexa if I can