alexa-pi/AlexaPi #268: How would you display alexa display cards on your screen?

SeanOMik, 2017-06-22:
When I ask a question to AlexaPi, I notice a new box (in the new update) and it fills with json code when alexa begins to speak. I was wondering how you could display that json data in a non json data form, and in an actual presentable form. 

PS: I know this probably isn't an issue, but there isn't any other place to ask this kind of stuff. Thanks any way

SeanOMik, 2017-06-29:
So seeing what EmerickH added I'm guessing there is no use for this JSON code yet? You should have it show in that json code box and have a button to show the json code.

alfredtv, 2018-03-07:
I'm new to GitHub and am searching so if this is not the right place, please let me know. I was able to build my first AlexaPi the other day and it's running fine. I have a 7 screen that is connected and am currently running DAKboard for visuals but just came across some items related to the Display Cards being available on new devices. I went into my Alexa setting on the developer site and added display card capabilities but was wondering if this can actually work or not?