alexa-pi/AlexaPi #272: Bluetooth Speaker with Mic Input used with AlexaPi

Jasonb11, 2017-06-26:
Hello,
I am trying to get a [iSound twist speaker with a mic ](https://www.amazon.com/Rechargeable-Wireless-Advanced-Rejecting-Microphone/dp/B007FETPC8) working on Alexa Pi. Have tried multiple PulseAudio 5 combinations in the config.yaml but have not gotten any success with the speaker's mic input.
I read that Pulse audio >= 6 will support headsets but I didn't find any tutorials on how to install v6 to test. I also hear BlueALSA will supposedly do something what I am looking to do.
Has anyone gotten this working and would be willing to share how they did it? If it is even possible...

EmerickH, 2017-07-21:
Hello,
You have 2 solutions:
 * You can use pulse, and so the config is done with `pavucontrol` for example, not in `config.yaml`, just:
```
input_device: "pulse"
output: "pulse"
output_device: ""
```
 * Or you can use alsa, and the config is done in `config.yaml`:
```
input_device: "YOUR INPUT PCM"
output: "alsa"
output_device: "YOUR OUTPUT PCM"
```
The PCMs can be obtained by `aplay -L` for output and `arecord -L` for input.
It's a thing like this `plughw:XXX,Dev:X` or just `plughw:XXX`

Emerick

Jasonb11, 2017-08-19:
This kind of makes sense. Except, what is pavucontrol? How is this not controlled in config.yaml?

EmerickH, 2017-08-20:
http://lmgtfy.com/?q=pavucontrol

> PulseAudio Volume Control (pavucontrol) is a simple GTK based volume control tool ("mixer") for the PulseAudio sound server.

Jasonb11, 2017-08-20:
I'm still having trouble with audio input. 

I just saw that raspbian was upgraded to stretch. They removed pulseaudio. 

Maybe upgrading will fix my issue?

renekliment, 2017-10-14:
In `pavucontrol` this would work.
Doing it in ALSA is a little more hardcore, but should work. Especially since we've updated to PyAudio and don't use ALSA directly.
You can give it a shot, but be sure to upgrade AlexaPi.