alexa-pi/AlexaPi #273: Alexa Responds only 3 times then stops + Choppy Audio

LoganPi3, 2017-06-27:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
Raspbian Pixel (Newest as of 6/25/17)

**Your hardware platform and model you are running on:**
Raspberry Pi 3 Model B

```

**Python release (`python2 --version`):**
Newest Version (Asof 6/25/17)

```

**Description of problem:**
When I boot it up, I say Alexa into the usb microphone, and she replies with "Yes?" so I say another command like "What time is it?" and she doesn't respond and I try "Alexa" again and I get nothing, so I open the LXTerminal and I type "sudo systemctl status AlexaPi.service" and I get the whole service thing says that its running and working, but the last line is "raspberrypi python[605]: [6e6004d8] alsa audio output error: cannot estimate delay: Input/output error"
Plus, when she says Yes? or Hello, it is either choppy or delayed. Can you help? Thanks

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

EmerickH, 2017-07-21:
Duplicate of #100:

> Hey!
> 
> `alsa audio output error: cannot estimate delay: Input/output error`
> This is normal, just ignore it.
> 
> Check your configuration.yaml to make sure you have correct audio devices set up. There is currently a bug in master that sets it up incorrectly. See #98 (comment)

Please provide more information like your `config.yaml`.

Emerick

renekliment, 2017-10-14:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-14 by renekliment