alexa-pi/AlexaPi #276: Alexa not responding to voice command after default 'Hello' greeting

uditparikh93, 2017-07-06:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of main.py script with -d flag**
```
2017-07-06 19:14:26 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 977
2017-07-06 19:14:26 INFO: AVS token: Obtained successfully
2017-07-06 19:14:26 DEBUG: after_setup
2017-07-06 19:14:26 DEBUG: Stopping audio play
2017-07-06 19:14:26 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-07-06 19:14:26 DEBUG: Player State: State.Opening
2017-07-06 19:14:26 DEBUG: Player State: State.Playing
2017-07-06 19:14:26 DEBUG: Started play. 
2017-07-06 19:14:26 DEBUG: indicate_playback True
2017-07-06 19:14:28 DEBUG: Player State: State.Ended
2017-07-06 19:14:28 DEBUG: Finished play. 
2017-07-06 19:14:28 DEBUG: indicate_playback False
2017-07-06 19:14:28 DEBUG: setup_complete
INFO: cmn_live.c(88): Update from < 41.00 -5.29 -0.12  5.09  2.48 -4.07 -1.37 -1.78 -5.08 -2.05 -6.45 -1.42  1.17 >
INFO: cmn_live.c(105): Update to   < 78.08  5.60 -6.66  6.14 -7.31 -3.57 -3.90 -9.23  0.77  0.79  2.91  4.19  0.32 >
INFO: cmn_live.c(88): Update from < 78.08  5.60 -6.66  6.14 -7.31 -3.57 -3.90 -9.23  0.77  0.79  2.91  4.19  0.32 >
INFO: cmn_live.c(105): Update to   < 77.71  4.46 -8.15  6.79 -7.29 -3.38 -2.82 -8.97  2.68  1.62  2.43  4.39 -0.87 >
INFO: cmn_live.c(88): Update from < 77.71  4.46 -8.15  6.79 -7.29 -3.38 -2.82 -8.97  2.68  1.62  2.43  4.39 -0.87 >
INFO: cmn_live.c(105): Update to   < 76.57  5.37 -8.18  5.39 -6.46 -3.91 -2.55 -8.71  2.14  3.75  3.22  3.51 -0.10 >
```

**Your OS (including version) where you are running AlexaPi:**
```
Ubuntu 14.04
```

**Your hardware platform and model you are running on:**
```
On local PC running Ubuntu 14.04
```

**Python release (`python2 --version`):**
```
Python 2.7.6
```

**Description of problem:**
```
On running main.py, Alexa is greeting with "Hello" message but, it is not responding to voice instructions.
The logs I have provided are obtained by running main.py with -d flag. FYI, I am using my headphones for audio input/output. Any guidance will be helpful. Thanks in advance.
```

**Problem-relevant `config.yaml` entries:**
```
After running setup.sh, I got config.template.yaml got edited. I do not see any config.yaml. Is this an issue?
```

hellad, 2017-07-12:
I've got the same problem on my Raspberry Pi2 when just installed AlexaPi first time. It looks like AlexaPi does not recognize the keyword (Alexa) but can hear some sounds. I'd suggest to check sound record quality first. In my case I had to switch over to [snowboy](https://github.com/Kitt-AI/snowboy) trigger since it is more tolerant for non-native speakers. It did the trick.

EmerickH, 2017-07-20:
@uditparikh93 the config file is located in /etc/opt/AlexaPi 
Like hellad has said, check your microphone, you can adjust its volume with `alsamixer`.
Can you provide us your config?

Emerick

maxavent, 2017-09-17:
Were you able to solve your issue ? I'm facing exactly the same one.
I'm able to record and play correctly my voice using arecord and aplay, the output of the alexa service is clean : 

```
Sep 17 15:45:53 raspberrypi systemd[1]: Started Alexa client for all your devices.
Sep 17 15:45:56 raspberrypi python[2142]: INFO: Checking Internet Connection ...
Sep 17 15:45:57 raspberrypi python[2142]: INFO: Connection OK
Sep 17 15:45:57 raspberrypi python[2142]: INFO: AVS token: Requesting a new one
Sep 17 15:45:58 raspberrypi python[2142]: INFO: AVS token: Obtained successfully
```

The output of the debug mode is the same kind as your : 
```
2017-09-17 15:50:36 DEBUG: Setting up playback handler: SoxHandler
2017-09-17 15:50:36 DEBUG: setup
2017-09-17 15:50:36 INFO: Checking Internet Connection ...
2017-09-17 15:50:36 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-09-17 15:50:37 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2017-09-17 15:50:37 INFO: Connection OK
2017-09-17 15:50:37 INFO: AVS token: Requesting a new one
2017-09-17 15:50:37 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-09-17 15:50:38 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 960
2017-09-17 15:50:38 INFO: AVS token: Obtained successfully
2017-09-17 15:50:38 DEBUG: after_setup
2017-09-17 15:50:38 DEBUG: Stopping audio play
2017-09-17 15:50:38 DEBUG: Stopped play. No SoX process to stop
2017-09-17 15:50:38 DEBUG: indicate_playback False
2017-09-17 15:50:38 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-09-17 15:50:38 DEBUG: Started play. sox -q /opt/AlexaPi/src/resources/hello.mp3 -t pulseaudio default vol -6 dB pad 0 0
2017-09-17 15:50:38 DEBUG: indicate_playback True
2017-09-17 15:50:40 DEBUG: Finished play.
2017-09-17 15:50:40 DEBUG: indicate_playback False
2017-09-17 15:50:40 DEBUG: setup_complete
^[[AINFO: cmn_live.c(88): Update from < 41.00 -5.29 -0.12  5.09  2.48 -4.07 -1.37 -1.78 -5.08 -2.05 -6.45 -1.42  1.17 >
INFO: cmn_live.c(105): Update to   < 24.17  6.42 -1.42  0.25 -4.16 -2.70 -5.77 -1.18  1.72  6.26  1.24  5.80  1.56 >
^[[AINFO: cmn_live.c(88): Update from < 24.17  6.42 -1.42  0.25 -4.16 -2.70 -5.77 -1.18  1.72  6.26  1.24  5.80  1.56 >
INFO: cmn_live.c(105): Update to   < 32.99 12.03  0.12  0.19 -6.66 -7.80 -4.13 -4.97  1.36  5.26  1.12  6.57 -1.83 >
INFO: cmn_live.c(88): Update from < 32.99 12.03  0.12  0.19 -6.66 -7.80 -4.13 -4.97  1.36  5.26  1.12  6.57 -1.83 >
```
Some help would be more than appreciated :-)

maxavent, 2017-09-21:
Did anyone face the same kind of issue ?
Since 4 days I :

- tried with PA and sox => same result
- bough a ps eye to change the mic => same result
- reinstalled the whole raspbian OS => same result

So i'm starting to doubt that this is caused by my configuration....

maso27, 2017-09-21:
Have you tried activating it via button press?  Any success with that?

A common issue is setting the volume level of your microphone too high, so you could try turning that down.

Also, you can try changing the trigger word (listed as "phrase" in _config.yaml_) to something different.  Pocketsphinx isn't the best at hearing "alexa" with differing accents.

maxavent, 2017-10-08:
Hello,

I found the solution : I changed the default sensibility of the hotword detection and it works now much better.

Regards.

closed 2017-10-08 by renekliment