alexa-pi/AlexaPi #277: Install won't complete - Amazon error: "We're sorry! An error occurred"

nuentes, 2017-07-12:
I can't seem to complete the install... I'm stuck at the very last step. The amazon page will not load either on the local device or through the network, I always get [this error message](https://i.imgur.com/zMZbFwE.jpg). From searching this issues list, it seems it's possible this is related to the .asoundrc file, which I did modify to:

```
pcm.!default {
        type hw
        card 0
        device 1
}

ctl.!default {
        type hw
        card 0
}
```

This was working fine while I was using the alexa-avs-sample-app, but I have decided to migrate to this fork. Here is my install log:

```
$ cd /opt
pi@PI-HOLE:/opt $ sudo git clone https://github.com/alexa-pi/AlexaPi.git
Cloning into 'AlexaPi'...
remote: Counting objects: 903, done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 903 (delta 0), reused 0 (delta 0), pack-reused 900
Receiving objects: 100% (903/903), 521.80 KiB | 440.00 KiB/s, done.
Resolving deltas: 100% (565/565), done.
Checking connectivity... done.
pi@PI-HOLE:/opt $ sudo ./AlexaPi/src/scripts/setup.sh
TIP: When there is a value in brackets like [default_value], hit Enter to use it.

Which operating system are you using?
         debian - Debian, Raspbian, Armbian, Ubuntu or other Debian-based
      archlinux - Arch Linux or Arch Linux-based
Your OS [debian]: 
Which device are you using?
           chip - C.H.I.P.
       hyperion - Integrate with Hyperion Ambient Lightning Software
    magicmirror - Integrating with Magic Mirror project (MMM-AlexaPi)
       orangepi - Orange Pi or another H3 based board
          other - other SBCs, desktops, or anything else
    raspberrypi - all Raspberry Pi variants
         serial - an Arduino controled device for example; used for Teddy Ruxpin
Your device [raspberrypi]: 
Do you want AlexaPi to run on boot?
You have these options: 
0 - NO
1 - yes, use systemd (default, RECOMMENDED and awesome)
2 - yes, use a classic init script (for a very old PC or an embedded system)
Which option do you prefer? [1]: 1
Creating a user to run AlexaPi under ... done.
gpio:x:997:pi
audio:x:29:pi,pulse
Created symlink from /etc/systemd/system/default.target.wants/AlexaPi.service to /usr/lib/systemd/system/AlexaPi.service.
Would you like to also install Airplay support (Y/n)? n
Get:1 http://mirrordirector.raspbian.org jessie InRelease [14.9 kB]
Hit http://ppa.launchpad.net trusty InRelease                              
Get:2 http://archive.raspberrypi.org jessie InRelease [22.9 kB]            
Hit http://ppa.launchpad.net trusty/main Sources             
Hit http://ppa.launchpad.net trusty/main armhf Packages                    
Get:3 http://mirrordirector.raspbian.org jessie/main armhf Packages [9,533 kB] 
Hit http://ppa.launchpad.net trusty/main Translation-en                        
Get:4 http://archive.raspberrypi.org jessie/main armhf Packages [170 kB]       
Hit http://archive.raspberrypi.org jessie/ui armhf Packages                    
Ign http://archive.raspberrypi.org jessie/main Translation-en_US               
Ign http://archive.raspberrypi.org jessie/main Translation-en                  
Ign http://archive.raspberrypi.org jessie/ui Translation-en_US                 
Ign http://archive.raspberrypi.org jessie/ui Translation-en                    
Get:5 http://mirrordirector.raspbian.org jessie/contrib armhf Packages [43.3 kB]
Get:6 http://mirrordirector.raspbian.org jessie/non-free armhf Packages [84.2 kB]
Get:7 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1,356 B]   
[... 242 lines cut ...]
      Successfully uninstalled chardet-2.3.0
  Found existing installation: urllib3 1.9.1
    Uninstalling urllib3-1.9.1:
      Successfully uninstalled urllib3-1.9.1
  Found existing installation: requests 2.4.3
    Uninstalling requests-2.4.3:
      Successfully uninstalled requests-2.4.3
  Running setup.py install for python-vlc ... done
  Running setup.py install for pyalsaaudio ... done
  Running setup.py install for webrtcvad ... done
  Running setup.py install for pyyaml ... done
  Running setup.py install for pocketsphinx ... done
  Running setup.py install for future ... done
Successfully installed CherryPy-11.0.0 certifi-2017.4.17 chardet-3.0.4 cheroot-5.7.0 coloredlogs-7.0 future-0.16.0 humanfriendly-4.3 idna-2.5 monotonic-1.3 pocketsphinx-0.1.3 portend-2.1.2 pyalsaaudio-0.8.4 python-vlc-1.1.2 pytz-2017.2 pyyaml-3.12 requests-2.18.1 tempora-1.8 urllib3-1.21.1 webrtcvad-2.0.10
Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages

Creating configuration file /etc/opt/AlexaPi/config.yaml ...
Enter your Device Type ID []: Raspberry_Pi
Enter your Security Profile Description []: Default
Enter your Security Profile ID []: amzn1.application.xxx101b1381c48599b796aa3xxxxxxx
Enter your Client ID []: amzn1.application-oa2-client.xxx528b1156d475ebc8a21149xxxxxxx
Enter your Client Secret []: xxxxxxxc5426aa4430bd88a76762c7b019be7348d764f6490fc375xxxxxxx
Ready goto http://192.168.1.11:5050 or http://localhost:5050  to begin the auth process
(Press Ctrl-C to exit this script once authorization is complete)
```

chixinmu, 2017-07-12:
@nuentes it is because the URLs is not correct, refer to the installation of guide, it's http://ALEXA.DEVICE.IP.ADDRESS:5050/code    where different to the alexa-avs-sample-app install guide.

nuentes, 2017-07-12:
I'm going to the addresses provided by alexa-pi in the 2nd to last line of my last post:

- http://192.168.1.11:5050
- http://localhost:5050

both give the error I mentioned above.

- http://192.168.1.11:5050/code
- http://localhost:5050/code

Both give a 500 Internal Server error.

hellad, 2017-07-16:
Try to use not local IP address, but rather your full.domain.name:5050 .  Ensure port 5050 on your Raspberry is available from internet for Amazon servers. 
For me it works.

nuentes, 2017-07-17:
I've tried all the following:
192.168.1.11:5050 (tested on both host and client)
localhost:5050 (tested on host only)
www.mydomain.com:5050 (tested on host and client)

All of them result in the Amazon error which I displayed. Isn't receiving the Amazon error proof that the connection is being made, anyway? Why else would all these tests resolve to Amazon.com? The connection can't possibly be the issue.

I've turned off my Pi's VPN entirely, and the issue persists. Port 5050 is not being used by any other service, as far as I know. If it were, I doubt I'd be able to load the page at all.

maso27, 2017-07-17:
So just to be sure, 
You went to developer.amazon.com and logged in.  
Navigated to the settings for your alexa device.
Under Security Profile / Web settings, you entered:

- Allowed Origins: **http://192.168.1.11:5050** (Assuming that's your Pi's address, of course)
- Allowed Return URLs: **https://192.168.1.11:5050/code**

Now, edit your config file with `sudo nano /etc/opt/AlexaPi/config.yaml`
Go down to the Amazon Alexa settings, and verify that you have the right things entered for:

- Client_ID
- Client_Secret
- Device_type_ID
- Security_Profile_Description
- Security_Profile_ID

They should match everything in your developer.amazon.com account from above.  Exit and save changes.

Next, run `sudo /opt/AlexaPi/src/auth_web.py`.
From a separate computer, on the same network, navigate to 192.168.1.11:5050 and log in with your amazon credentials.
-----------------------------------------

That should work.  If it doesn't, go back and make sure you haven't mis-typed something.  Every time I've run into this problem I've fat-fingered something.

nuentes, 2017-07-18:
Ah - there we go. I forgot to change the Allowed Origins and Allowed Return URLs in the Developer console from alexa-avs-sample-app. That took care of it. My bad. I'm an idiot.

closed 2017-07-18 by nuentes

maso27, 2017-07-18:
No worries!  I've previously lost a couple days on a space where it shouldn't have been.
:)

rbaudanza, 2018-03-31:
I am getting the same error.  Checked config.yaml, it matches everything I have for my security profile.  The one difference is in the web settings it doesn't allow to use http for anything other than localhost.  Here are my entries 

Allowed Origins | http://localhost:5050
https://192.186.86.40:5050
-- | --
Allowed Return URLs | http://localhost:5050/code
https://192.186.86.40:5050/code

Here is the error I'm getting from amazon.  It's weird because I think it needs to have the return URL set to http://my.pi.ip.address:5050 but it won't let me add that on the amazon developer page.  It only accepts https. 

Error Summary
400 Bad Request
The redirect URI you provided has not been whitelisted for your application. Please add your redirect URI in the 'Allowed Return URLs' section under 'Web Settings' for your Security Profile on Amazon Developer Portal.
Request Details
scope=alexa%3Aall
redirect_uri=http%3A%2F%2F192.168.86.40%3A5050%2Fcode
response_type=code
client_id=I redacted this but verified it is correct in config.yaml.
scope_data=%7B%22alexa%3Aall%22%3A+%7B%22productInstanceAttributes%22%3A+%7B%22deviceSerialNumber%22%3A+%22721d24bab2580c6cbe4fdb8f610b0d88c10144b6f4c88db74b216699a595deb2%22%7D%2C+%22productID%22%3A+%22Grace%22%7D%7D

bekirdag, 2018-09-28:
In my case it was a cherrpy issue. I installed the 10.x version, but later installing the 17.4.0 version fixed the issue. 

`sudo pip install CherryPy==17.4.0`

**Details:**
When I tried to see if it was a cherrpy issue and tested a simple server, I started to get the below error:
`Module_six_moves_urllib_parse' object has no attribute 'unquote_to_bytes`

I have upgraded the package called `six` (from v10 to v11) but it didn't solve the issue. 

**Why not upgrading to the latest version of cheerypy?**
Because <18.x requires <=python 3.5