alexa-pi/AlexaPi #278: AlexaPi doesn't respond after TuneIn Radio playback started

hellad, 2017-07-12:
In order to help troubleshooting, be sure to include the following information:
**Your OS (including version) where you are running AlexaPi:**
Linux adhassbian 4.9.35-v7+ #1014
**Your hardware platform and model you are running on:**
Pi 2 Model B v1.1 1GB a21041 (Embest, China)
**Python release (`python2 --version`):**
Python 2.7.9
**Description of problem:**
Alexa stop responding after TuneIn Radio SomaFM Groove salad playback started. Music is being played endlessly until I restart the service manually.

**Expected:**
Ability to interact with Alexa and control playback

config.yaml
```
  input_device: ""
  playback_handler: "vlc"
  output: "alsa"
  output_device: "default"
  snowboy:
    enabled: true
```

**Steps to reproduce:**

1. Alexa, play SomaFM Groove salad

**Additional info:**
I use development branch of AlexaPi (sudo git clone -b dev https://github.com/alexa-pi/AlexaPi.git) with pulseaudio service stopped. Only in this case Alexa works smoothly with my Onkyo AV receiver via HDMI as ALSA output. I tried to use pulse without success. 

pi@adhassbian:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@adhassbian:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Q9000 [QuickCam Pro 9000], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

Debug log of the response to my command:

июл 12 20:50:49 adhassbian python[2210]: DEBUG: TUNE IN URL = https://opml.radiotime.com/Tune.ashx?serial=AFOMESROFC6THLDWTMPUJPIOWDYA&formats=aac,mp3&id=s2591&partnerId=4JqugguZ
июл 12 20:50:49 adhassbian python[2210]: DEBUG: Starting new HTTPS connection (1): opml.radiotime.com
июл 12 20:50:50 adhassbian python[2210]: DEBUG: https://opml.radiotime.com:443 "GET /Tune.ashx?serial=AFOMESROFC6THLDWTMPUJPIOWDYA&formats=aac,mp3&id=s2591&partnerId=4JqugguZ HTTP/1.1" 200 200
июл 12 20:50:50 adhassbian python[2210]: DEBUG: Extracting URIs from http://api.somafm.com/groovesalad130.pls
июл 12 20:50:50 adhassbian python[2210]: DEBUG: Starting new HTTP connection (1): api.somafm.com
июл 12 20:50:55 adhassbian python[2210]: DEBUG: http://api.somafm.com:80 "GET /groovesalad130.pls HTTP/1.1" 200 202
июл 12 20:50:55 adhassbian python[2210]: DEBUG: http://api.somafm.com/groovesalad130.pls has content-type: audio/x-scpls
июл 12 20:50:55 adhassbian python[2210]: DEBUG: Got [u'http://ice1.somafm.com/groovesalad-128-aac', u'http://ice2.somafm.com/groovesalad-128-aac']
июл 12 20:50:55 adhassbian python[2210]: DEBUG: Playing audio: http://ice1.somafm.com/groovesalad-128-aac
июл 12 20:50:55 adhassbian python[2210]: DEBUG: Player State: State.Opening
июл 12 20:50:55 adhassbian python[2210]: Exception in thread Thread-6:
июл 12 20:50:55 adhassbian python[2210]: Traceback (most recent call last):
июл 12 20:50:55 adhassbian python[2210]: File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
июл 12 20:50:55 adhassbian python[2210]: self.run()
июл 12 20:50:55 adhassbian python[2210]: File "/usr/lib/python2.7/threading.py", line 763, in run
июл 12 20:50:55 adhassbian python[2210]: self.__target(*self.__args, **self.__kwargs)
июл 12 20:50:55 adhassbian python[2210]: File "/opt/AlexaPi/src/main.py", line 542, in trigger_process
июл 12 20:50:55 adhassbian python[2210]: alexa_speech_recognizer(audio_stream)
июл 12 20:50:55 adhassbian python[2210]: File "/opt/AlexaPi/src/main.py", line 351, in alexa_speech_recognizer
июл 12 20:50:55 adhassbian python[2210]: process_response(resp)
июл 12 20:50:55 adhassbian python[2210]: File "/opt/AlexaPi/src/main.py", line 453, in process_response
июл 12 20:50:55 adhassbian python[2210]: if directive['namespace'] == 'SpeechSynthesizer':
июл 12 20:50:55 adhassbian python[2210]: KeyError: 'namespace'
июл 12 20:50:56 adhassbian python[2210]: [70500b28] http access: Raw-audio server found, m4a demuxer selected
июл 12 20:50:56 adhassbian python[2210]: [70523430] packetizer_mpeg4audio demux packetizer: AAC channels: 2 samplerate: 44100
июл 12 20:50:56 adhassbian python[2210]: DEBUG: Player State: State.Playing
июл 12 20:50:56 adhassbian python[2210]: DEBUG: Started play.
июл 12 20:50:56 adhassbian python[2210]: DEBUG: Sending Playback Progress Report Request...
июл 12 20:50:56 adhassbian python[2210]: DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com
июл 12 20:51:03 adhassbian python[2210]: DEBUG: https://access-alexa-na.amazon.com:443 "POST /v1/avs/audioplayer/playbackStarted HTTP/1.1" 200 None
июл 12 20:51:03 adhassbian python[2210]: WARNING: (alexa_playback_progress_report_request Response) <Response [200]>

EmerickH, 2017-07-20:
It seems to be a bug, thanks for reporting, we'll try to solve it.

Emerick

hellad, 2017-07-20:
I'm looking forward for your solution.

In meanwhile I can add one more station with the same problem - Alexa, play TuneIn BBC Radio One .

zespirit, 2017-07-21:
Agreed, same here, running raspberry pi, latest jessie, alexapi, magic mirror, all works perfect.
As soon as a video starts playing, i can no longer speak with alexa...
Plugged to an external Microphone Blue Snowball.

hellad, 2017-08-08:
I'd like to inform that after apt-get update and apt-get upgrade my Raspberry Pi 2, as well as upgrade to the latest version of Home-Assistant  running on the same board, the problem has gone. At least now I'm able to stop TuneIn Radio playback after it was started. Will keep testing it.