alexa-pi/AlexaPi #279: [Enhancement] Play wake word feedback in parallel to improve Alexa responsiveness

lap-tran, 2017-07-15:
This is an enhancement suggestion.
Currently we wait until wake work feedback (the "yes" sound) finishes then connect to AVS and then stream recorded voice. Connecting to AVS isn't always fast, can be seconds, and that causes a delay in Alexa respond - very often in my experience. Eventually, we can connect to AVS earlier and make Alexa feel more responsive.
Let's start playing "yes" in parallel in another thread while connecting to AVS immediately after the wake word is detected. Then the sound playback thread needs to join back before we start recording and send speech to AVS.

lap-tran, 2017-07-15:
I hacked it locally and Alexa feels a lot smoother.

EmerickH, 2017-07-21:
Very good idea, I think there's no problems to do that.
If you have done it, you can push your local git to Github and make a pull request.

Emerick

renekliment, 2017-10-14:
@lap-tran Can you open a PR with this?

Cole128, 2017-10-16:
This sounds like a great idea, it always seemed kinda laggy to respond.