alexa-pi/AlexaPi #282: Playback is failing

nuentes, 2017-07-20:
I'm getting the following error: mpgatofixed32 audio converter error: libmad error: bad main_data_begin pointer

I have a feeling it's due to the /home/pi/.asoundrc file which I had previously edited, and then edited again just a few minutes ago (I was getting an identical error before I changed that).

```
$ sudo systemctl status AlexaPi.service
? AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: active (running) since Thu 2017-07-20 18:13:29 EDT; 11min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 2145 (python)
   CGroup: /system.slice/AlexaPi.service
           ??2145 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

Jul 20 18:13:29 PI-HOLE systemd[1]: Started Alexa client for all your devices.
Jul 20 18:14:14 PI-HOLE python[2145]: INFO: Checking Internet Connection ...
Jul 20 18:14:16 PI-HOLE python[2145]: INFO: Connection OK
Jul 20 18:14:17 PI-HOLE python[2145]: INFO: AVS token: Requesting a new one
Jul 20 18:14:18 PI-HOLE python[2145]: INFO: AVS token: Obtained successfully
Jul 20 18:14:24 PI-HOLE python[2145]: [b410c620] mpgatofixed32 audio converter error: libmad error: bad main_data_begin pointer
Jul 20 18:14:24 PI-HOLE python[2145]: [b410c620] mpgatofixed32 audio converter error: libmad error: bad main_data_begin pointer
Jul 20 18:14:24 PI-HOLE python[2145]: [b410c620] mpgatofixed32 audio converter error: libmad error: bad main_data_begin pointer
Jul 20 18:14:24 PI-HOLE python[2145]: [b410c620] mpgatofixed32 audio converter error: libmad error: bad main_data_begin pointer
Jul 20 18:14:24 PI-HOLE python[2145]: [b410c620] mpgatofixed32 audio converter error: libmad error: bad main_data_begin pointer
```

result from

`cd ~ && wget https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info.sh && bash alsa-info.sh`

```
upload=true&script=true&cardinfo=
!!################################
!!ALSA Information Script v 0.4.64
!!################################

!!Script ran on: Thu Jul 20 22:16:34 UTC 2017

!!Linux Distribution
!!------------------

ID_LIKE=debian

!!DMI Information
!!---------------

Manufacturer:      
Product Name:      
Product Version:   
Firmware Version:  

!!Kernel Information
!!------------------

Kernel release:    4.9.35+
Operating System:  GNU/Linux
Architecture:      armv6l
Processor:         unknown
SMP Enabled:       No

!!ALSA Version
!!------------

Driver version:     k4.9.35+
Library version:    1.0.28
Utilities version:  1.0.28

!!Loaded ALSA modules
!!-------------------

snd_bcm2835
snd_usb_audio

!!Sound Servers on this system
!!----------------------------

Pulseaudio:
      Installed - Yes (/usr/bin/pulseaudio)
      Running - Yes

Jack:
      Installed - Yes (/usr/bin/jackd)
      Running - No

!!Soundcards recognised by ALSA
!!-----------------------------

 0 [ALSA           ]: bcm2835 - bcm2835 ALSA
                      bcm2835 ALSA
 1 [CameraB409241  ]: USB-Audio - USB Camera-B4.09.24.1
                      OmniVision Technologies, Inc. USB Camera-B4.09.24.1 at usb-20980000.usb-1.5, hi

!!Modprobe options (Sound related)
!!--------------------------------

snd_pcsp: index=-2
snd_usb_audio: index=-2
snd_atiixp_modem: index=-2
snd_intel8x0m: index=-2
snd_via82xx_modem: index=-2

!!Loaded sound module options
!!---------------------------

!!Module: snd_bcm2835
	force_bulk : N

!!Module: snd_usb_audio
	autoclock : Y
	device_setup : 0,0,0,0,0,0,0,0
	enable : Y,Y,Y,Y,Y,Y,Y,Y
	id : (null),(null),(null),(null),(null),(null),(null),(null)
	ignore_ctl_error : N
	index : -2,-1,-1,-1,-1,-1,-1,-1
	pid : -1,-1,-1,-1,-1,-1,-1,-1
	quirk_alias : (null),(null),(null),(null),(null),(null),(null),(null)
	vid : -1,-1,-1,-1,-1,-1,-1,-1

!!USB Mixer information
!!---------------------
--startcollapse--

USB Mixer: usb_id=0x14152000, ctrlif=1, ctlerr=0
Card: OmniVision Technologies, Inc. USB Camera-B4.09.24.1 at usb-20980000.usb-1.5, hi
  Unit: 3
    Control: name="Mic Capture Volume", index=0
    Info: id=3, control=2, cmask=0xf, channels=4, type="S16"
    Volume: min=0, max=1, dBmin=0, dBmax=0
--endcollapse--

!!ALSA Device nodes
!!-----------------

crw-rw----+ 1 root audio 116,  0 Jul 20 17:17 /dev/snd/controlC0
crw-rw----+ 1 root audio 116, 32 Jul 20 17:48 /dev/snd/controlC1
crw-rw----+ 1 root audio 116, 16 Jul 20 18:01 /dev/snd/pcmC0D0p
crw-rw----+ 1 root audio 116, 17 Jul 20 17:17 /dev/snd/pcmC0D1p
crw-rw----+ 1 root audio 116, 56 Jul 20 17:48 /dev/snd/pcmC1D0c
crw-rw----+ 1 root audio 116,  1 Jul 20 17:17 /dev/snd/seq
crw-rw----+ 1 root audio 116, 33 Jul 20 17:17 /dev/snd/timer

/dev/snd/by-id:
total 0
drwxr-xr-x 2 root root  60 Jul 20 17:48 .
drwxr-xr-x 4 root root 220 Jul 20 17:48 ..
lrwxrwxrwx 1 root root  12 Jul 20 17:48 usb-OmniVision_Technologies__Inc._USB_Camera-B4.09.24.1-01 -> ../controlC1

/dev/snd/by-path:
total 0
drwxr-xr-x 2 root root  80 Jul 20 17:48 .
drwxr-xr-x 4 root root 220 Jul 20 17:48 ..
lrwxrwxrwx 1 root root  12 Jul 20 17:48 platform-20980000.usb-usb-0:1.5:1.1 -> ../controlC1
lrwxrwxrwx 1 root root  12 Jul 20 17:17 platform-soc:audio -> ../controlC0

!!ALSA configuration files
!!------------------------

!!User specific config file (~/.asoundrc)

pcm.!default {
	type pulse
	card PCH
}

ctl.!default {
	type pulse          
	card PCH
}

!!Aplay/Arecord output
!!--------------------

APLAY

**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

ARECORD

**** List of CAPTURE Hardware Devices ****
card 1: CameraB409241 [USB Camera-B4.09.24.1], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

!!Amixer output
!!-------------

!!-------Mixer controls for card 0 [ALSA]

Card hw:0 'ALSA'/'bcm2835 ALSA'
  Mixer name	: 'Broadcom Mixer'
  Components	: ''
  Controls      : 6
  Simple ctrls  : 1
Simple mixer control 'PCM',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback 0 [96%] [0.00dB] [on]

!!-------Mixer controls for card 1 [CameraB409241]

amixer: Mixer load hw:1 error: Invalid argument
Card hw:1 'CameraB409241'/'OmniVision Technologies, Inc. USB Camera-B4.09.24.1 at usb-20980000.usb-1.5, hi'
  Mixer name	: 'USB Mixer'
  Components	: 'USB1415:2000'
  Controls      : 2
amixer: Mixer hw:1 load error: Invalid argument

!!Alsactl output
!!--------------

--startcollapse--
--endcollapse--

!!All Loaded Modules
!!------------------

Module
bnep
bluetooth
tun
cfg80211
rfkill
snd_usb_audio
snd_hwdep
snd_usbmidi_lib
snd_rawmidi
snd_seq_device
gspca_ov534
gspca_main
v4l2_common
videodev
media
ip6table_filter
ip6_tables
xt_tcpudp
xt_comment
xt_conntrack
iptable_filter
ipt_MASQUERADE
nf_nat_masquerade_ipv4
xt_nat
iptable_nat
nf_conntrack_ipv4
nf_defrag_ipv4
nf_nat_ipv4
nf_nat
nf_conntrack
ip_tables
x_tables
snd_bcm2835
snd_pcm
snd_timer
snd
bcm2835_gpiomem
uio_pdrv_genirq
uio
fixed
evdev
i2c_dev
fuse
ipv6

!!ALSA/HDA dmesg
!!--------------

[   16.383093] usb 1-1.5: 3:1: cannot get min/max values for control 2 (id 3)
[   16.391463] usbcore: registered new interface driver snd-usb-audio
[   20.208849] tun: Universal TUN/TAP device driver, 1.6
```

Before this, I was getting the below error, which cleared up after I changed input and output to 'pulse' in /etc/opt/AlexaPi/config.yaml:

```
$ sudo systemctl status AlexaPi.service
? AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Thu 2017-07-20 17:49:16 EDT; 8s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 1539 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 1539 (code=exited, status=1/FAILURE)

Jul 20 17:49:16 PI-HOLE python[1539]: surround50:CARD=CameraB409241,DEV=0
Jul 20 17:49:16 PI-HOLE python[1539]: surround51:CARD=CameraB409241,DEV=0
Jul 20 17:49:16 PI-HOLE python[1539]: surround71:CARD=CameraB409241,DEV=0
Jul 20 17:49:16 PI-HOLE python[1539]: iec958:CARD=CameraB409241,DEV=0
Jul 20 17:49:16 PI-HOLE python[1539]: dmix:CARD=CameraB409241,DEV=0
Jul 20 17:49:16 PI-HOLE python[1539]: dsnoop:CARD=CameraB409241,DEV=0
Jul 20 17:49:16 PI-HOLE python[1539]: hw:CARD=CameraB409241,DEV=0
Jul 20 17:49:16 PI-HOLE python[1539]: plughw:CARD=CameraB409241,DEV=0
Jul 20 17:49:16 PI-HOLE systemd[1]: AlexaPi.service: main process exited, c...RE
Jul 20 17:49:16 PI-HOLE systemd[1]: Unit AlexaPi.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.
```

EmerickH, 2017-07-21:
Hello,
You have to change your recording device like said at the end the setup script:
edit `/var/opt/AlexaPi/config.yaml` and change
`input_device: "hw:CARD=CameraB409241,DEV=0"`
or
`input_device: "plughw:CARD=CameraB409241,DEV=0"`
if it's not working.

Emerick

nuentes, 2017-07-21:
I had changed both input and output to "pulse", as I said in my prior message. I just changed that field, and restarted AlexaPi, but I just keep getting Connection Errors every 20 seconds, and it never successfully authenticated, so I assume there is something else wrong, as I haven't changed anything, so I'll just wait longer.

EmerickH, 2017-07-22:
Hello,
If you changed to pulse, it seems thar alea can't find it, are you shure it is installed?
sudo apt-get install pulseaudio

Emerick

nuentes, 2017-07-26:
Yes, pulseaudio was already installed. I've tweaked some things, and can confirm that it's recording. When I look in PulseAudio volume control, I can see AlexaPi recording in the Recording section. When saying "alexa, what time is it", I see the recording close out, and a new one start a few seconds later. So recording is working. I've played with several settings, and gotten varied results, but I'm yet to get an audio response.

One thing that's probably noteworthy - it's a raspberry pi B+ running raspbian. I have the display connected via HDMI, but I'm trying to send sound out the 3.5mm.

EmerickH, 2017-07-27:
Hello,
With old raspberry's you can tweak 3.5/hdmi output with `raspi-config` (in Audio section).
Maybe it's your output device that's not correctly configured, please give us the complete log:
```
sudo systemctl stop AlexaPi.service
/opt/AlexaPi/src/main.py -d
```

Emerick

nuentes, 2017-07-27:
Ok, so running `raspi-config` is leading me down a rabbit hole. When changing audio output to force 3.5mm, raspi-config throws the following error "there was an error running option A4 Audio". So I escape out and see the following:

```
$ sudo raspi-config
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied

amixer: Control default open error: Connection refused
```

That led me to [this post](https://github.com/alexa-pi/AlexaPi/issues/182), so I ran `speaker-test -c2 --test=wav -w /usr/share/sounds/alsa/Front_Center.wav`, and I heard nothing.

Next, I found [this](http://www.pclinuxos.com/forum/index.php?topic=133522.0), and so I ran `sudo rm -rf /home/pi/.config/pulse` and rebooted, with no luck.

I've tried a few other things, but not having any luck getting sound out to my speakers...

EmerickH, 2017-07-29:
Hello,
have you read this?
https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#running-pa-in-system-wide-mode-recommended

Emerick

nuentes, 2017-07-31:
Yes, that is the guide I used to install pulseaudio. That doesn't seem to be relevent to my issue - I can't seem to change the audio device to the 3.5mm. I disconnected the camera and rebooted as a last-ditch effort, but I still can't change the default device.

I'm feeling about ready to abandon this end of the project. I thought this would be simple, but it's proving to be a pain, and I feel I'm 1 response from hearing "reinstall raspbian". Even if I got this working, I still have to find a good workaround for playing music using kodi. At this point, it's cheaper and easier for me to just buy another echo dot.

renekliment, 2017-10-08:
Have you tried SoX?

Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment