alexa-pi/AlexaPi #283: Err#111 - Connection Refused

kevoliveir, 2017-07-21:
After installing ALEXA PI the first time, everything worked just fine. But after a moment Alexa stopped responding and although AlexaPi.service was running she did not answer to any command. Now, it only works on debugging mode.

Log:
_Jul 20 19:47:37 KOl714 python[477]: ERROR: URLError: [Errno 111] Connection...ed
Jul 20 19:47:40 KOl714 python[477]: ERROR: URLError: [Errno 111] Connection...ed
Jul 20 19:47:40 KOl714 python[477]: ERROR: URLError: [Errno 111] Connection...ed
[...]_

Log on dubugging:
_2017-07-20 19:49:50 DEBUG: Sending MM Heatbeat
2017-07-20 19:49:50 ERROR: URLError: [Errno 111] Connection refused
2017-07-20 19:49:50 DEBUG: Player State: State.Ended
2017-07-20 19:49:50 DEBUG: Finished play. 
2017-07-20 19:49:50 DEBUG: Indicate Stop Playing
2017-07-20 19:49:50 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
2017-07-20 19:49:50 ERROR: URLError: [Errno 111] Connection refused
2017-07-20 19:49:50 DEBUG: Setting up recording
2017-07-20 19:49:50 DEBUG: Start recording
2017-07-20 19:49:50 DEBUG: Indicate Start Recording
2017-07-20 19:49:50 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=recording
2017-07-20 19:49:50 ERROR: URLError: [Errno 111] Connection refused
2017-07-20 19:49:50 DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com
2017-07-20 19:49:51 DEBUG: Start sending speech to Alexa Voice Service
2017-07-20 19:49:55 DEBUG: Sending MM Heatbeat
2017-07-20 19:49:55 ERROR: URLError: [Errno 111] Connection refused
2017-07-20 19:49:55 DEBUG: End recording
2017-07-20 19:49:55 DEBUG: Indicate Stop Recording
2017-07-20 19:49:55 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
2017-07-20 19:49:55 ERROR: URLError: [Errno 111] Connection refused_

I've tried to reinstall it, restart services (daemon, alexa, ...), running as root, well I tried everything the manual says but I still have the same issue.

Do you guys have any idea why this happens?

securefoo, 2017-08-25:
I'm running into the exact same issue. Lots of searching and troubleshooting have turned up nothing. I can get Alexa to work for a few mins - I run a few commends and then it stops working. I'm running Alexapi in debugging mode and can't get anything more than "URLError: [Errno 111] Connection refused". It seems like the fact that it works sometimes means that it's not a configuration issue with the allowed IPs on the Amazon Dev site. Netstat shows 5050 as open: 
"sudo netstat -tlnpu | grep 5050
tcp        0      0 0.0.0.0:5050            0.0.0.0:*               LISTEN"

Any ideas would be much appreciated.

renekliment, 2017-10-08:
`URLError: [Errno 111] Connection refused` is a warning when MagicMirror isn't set up properly.

Don't see anything relevant.
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment