alexa-pi/AlexaPi #284: Alexa not starting.

rites1978, 2017-07-21:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

http://www.alsa-project.org/db/?f=d3ee9ff7e33664da614e53c495e75abebd30f3f0

**Your OS (including version) where you are running AlexaPi:**
```Raspberry Pi3
```

**Your hardware platform and model you are running on:**
```Raspberry Pi3

```

**Python release (`python2 --version`):**
```python 2.7.9

```

**Description of problem:**
```I am a beginner and followed all the steps mentioned on various feeds. I am unable to set up USB mic as  default in Asla. I have tried using Putty but cannot save anything as there is only option to OPEN. 
ALexa service seems to be running but i cant wake her up with any command. May be it because of the usb mic. I have not seen anything happening after following all the steps.  

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```bash: config.yaml: command not found

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

EmerickH, 2017-07-21:
@ Your issue is incomplete, we can't he you! Please provide more information.
- You config.yaml
- The output of `aplay -L` and `arecord -L`

But first have you tried [this](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#checking-if-your-system-audio-works)? 

Emerick

renekliment, 2017-10-08:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment