alexa-pi/AlexaPi #285: Cannot get microphone to work

L0hr, 2017-07-22:
OS: Raspbian
Hardware: Raspberry Pi Zero

Hi, I cannot get my PI to use a usb mic. I have an external USB DAC plugged into a powered USB hub, with a 3.5mm cable going to a Bose speaker for output.

pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
pi@raspberrypi:~ $

Shows nothing listed....

pi@raspberrypi:~ $ lsusb
Bus 001 Device 008: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
Bus 001 Device 026: ID 08bb:2702 Texas Instruments Speakers
Bus 001 Device 007: ID 214b:7000  
Bus 001 Device 002: ID 214b:7000  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ 

Doesn't appear to list the usb mic either? I plug it into my mac though, and my Mac detects it.. 

Do I need to do more editing in alsa.conf? I cannot figure this out, been messing with it for hours. Finally got audio to work, but no mic...

EmerickH, 2017-07-23:
See https://github.com/alexa-pi/AlexaPi/issues/281#issuecomment-316814222

Emerick

L0hr, 2017-07-24:
Thanks. I had spent several days playing with it, and just ended up instead spending the $40ish and buying an Echo Dot. I'll save this raspberry pi for another project I suppose, or maybe put Alexapi on it for another room. Anyways, thanks for the answer.

closed 2017-10-08 by renekliment