alexa-pi/AlexaPi #287: Cannot Recieve AVS refresh-token

sw1mm3r, 2017-07-25:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
Raspbian "Jessie" 
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi 2 Model B
```

**Python release (`python2 --version`):**
```
2.7
```

**Description of problem:**
```
Made new config.yaml to solve my audio input problem, tried to start up Alexa but was met with a message saying I was missing an AVS refresh token from my config file. Tried to use the web_auth.py but the Amazon page said it could not process my request.
```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. Copy the config.yaml template to a config.yaml file
2. Make the file writeable using "sudo chmod 777 config.yaml"
3. Attempt to initiate service using "python main.py"

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

EmerickH, 2017-07-25:
Hello,
 You need the amazon tokens in the config file, just run the setup.sh again to obtain them automatically.
The config file must ne in /var/opt/AlexaPi/config.yaml.

Emerick

sw1mm3r, 2017-07-25:
So I am also trying to use a broken headset as a mic, doesn't replacing the config file lead to that being reset to the default mic?

EmerickH, 2017-07-27:
Hello,
First, a Raspberry Pi doesn't have any audio input, so you can use only a USB mic.
Secondly you **can EDIT** your config, but **you must keep all the others parameters that are already here** like the Amazon tokens.

Emerick

renekliment, 2017-10-08:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment