alexa-pi/AlexaPi #288: After Hello and "YES" Alexa do not say nothing. USB mic works well!

masteruan, 2017-07-26:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
Raspbian JESSIE

**Your hardware platform and model you are running on:**
```
USB Kinobo mic
analog stereo output
```

**Python release (`python2 --version`):**
```
python2
```

**Description of problem:**
```
When I send a request to alexa alexa do not respond
```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```
/opt/AlexaPi/src/main.py -d

DEBUG:

2017-07-26 15:27:17 DEBUG: Setting up playback handler: VlcHandler
2017-07-26 15:27:18 INFO: Checking Internet Connection ...
2017-07-26 15:27:18 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-07-26 15:27:23 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2017-07-26 15:27:23 INFO: Connection OK
2017-07-26 15:27:23 INFO: AVS token: Requesting a new one
2017-07-26 15:27:23 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-07-26 15:27:29 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 979
2017-07-26 15:27:29 INFO: AVS token: Obtained successfully
2017-07-26 15:27:29 DEBUG: Stopping audio play
2017-07-26 15:27:29 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-07-26 15:27:29 DEBUG: Player State: State.Opening
2017-07-26 15:27:30 DEBUG: Player State: State.Playing
2017-07-26 15:27:30 DEBUG: Started play. 
2017-07-26 15:27:32 DEBUG: Player State: State.Ended
2017-07-26 15:27:32 DEBUG: Finished play. 

I say: "Alexa" -- YES -- "What time is it?"

2017-07-26 15:28:16 DEBUG: Stopping audio play
2017-07-26 15:28:16 DEBUG: Playing audio: /opt/AlexaPi/src/resources/alexayes.mp3
2017-07-26 15:28:16 DEBUG: Player State: State.Opening
2017-07-26 15:28:16 DEBUG: Player State: State.Playing
2017-07-26 15:28:16 DEBUG: Started play. 
2017-07-26 15:28:16 DEBUG: Player State: State.Ended
2017-07-26 15:28:16 DEBUG: Finished play. 
2017-07-26 15:28:16 DEBUG: Setting up recording
2017-07-26 15:28:17 DEBUG: Start recording
2017-07-26 15:28:17 DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com
2017-07-26 15:28:22 DEBUG: Start sending speech to Alexa Voice Service
2017-07-26 15:28:30 DEBUG: End recording
2017-07-26 15:28:30 DEBUG: Finished sending speech to Alexa Voice Service
2017-07-26 15:28:31 DEBUG: Processing Request Response...
2017-07-26 15:28:31 DEBUG: Request Response is null (This is OKAY!)

EmerickH, 2017-07-27:
Hello,
see [my new audio setup guide)[https://emerickh.github.io/AlexaPi-config/setup/alsa.html].

Emerick

masteruan, 2017-07-27:
I've followed your audio guide...but nothing... :-(

EmerickH, 2017-07-27:
Try to use [pulse](https://emerickh.github.io/AlexaPi-config/setup/pulse.html).

Emerick

masteruan, 2017-07-27:
After installation of pulse method, I open the pavucontrol panel, but I do not see the input device. After with alsa configuration when I saw "Alexa", Alexa say yes now with pulse I can hear "hello" but when I talk "Alexa" do not respond me and in the debug console nothing appears.

This mistake where are? 
<<there was a mistake in the guide, it's not /var/AlexaPi/config.yaml but /etc/AlexaPi/config.yaml !>>
In the pulse guide? or general installation Alexa guide?
Thanks

EmerickH, 2017-07-27:
@masteruan I changed everything, it was in the Alsa guide.
Can you give me the output of `arecord -L`?

Emerick

masteruan, 2017-07-27:
@EmerickH this is the output of arecord -L
pi@raspberrypi:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
default
    Playback/recording through the PulseAudio sound server
sysdefault:CARD=Device
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions

bkems, 2017-08-02:
I have the exact same issue. I was wondering if the audio is in the expected format and if not how i'd fix that. I've also followed the steps in the guide, but I get the same issue still. It would be nice to be able to verify that the audio I am sending is ok to rule that out. It's odd though, because the wakeword works fine - saying 'alexa" wakes it up reliably.

maso27, 2017-08-03:
Wakeword detection and Alexa's responses are entirely separate software.

Have you tried "Checking if your system audio works" in the [audio debugging section of the wiki](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging)?

My bet is you have the microphone sensitivity up too high and your voice is getting distorted.  So when the audio is sent to amazon's servers it doesn't know what to do with it.

There's also a step right below that, called "Checking AlexaPi's Recordings".

MJ12358, 2017-09-09:
I am having similar issues. The wake word detection works but on any captured audio I receive the following error using debug:

Exception in thread Thread-4:
Traceback (most recent call last):
  File "/user/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/AlexaPi/src/main.py", line 542, in trigger_process
    alexa_speech_recognizer(audio_stream)
  File "/opt/AlexaPi/src/main.py", line 351, in alexa_speech_recognizer
    process_response(resp)
  File "/opt/AlexaPi/src/main.py", line 442, in process_response
    with open(filename, 'wb') as f:
IOError: [Errno 2] No such file or directory: '/tmp/AlexaPi-runtime-######/$STRINGOFJUNK.mp3'

Playback of the recording.wav file within /tmp/AlexaPi-runtime-###### plays without distortion!

Any thoughts?
Thanks

bkems, 2017-09-09:
An issue I had was the absence of an MP3 decoder. https://www.virtuit.com.au/2017/08/22/creating-alexa-device-raspberry-pi/ has a description of my experiences that might help.

This e-mail is confidential and may be read, copied and used only by the intended recipient. If you have received it in error, please contact the sender immediately by return e-mail. Please then delete the e-mail and do not disclose its contents to any other person.

renekliment, 2017-10-08:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment

Gandalf1783, 2020-03-27:
Well then i'd like to repopen the conversation.

Im Using Raspi 4, an Behringer Xenyx 302 usb which i confirmed working with 
`aplay /usr/share/sounds/alsa/Front_Center.wav`
Now my problem:
I dont know which format i should use...
Thanks everybody ^^

btw, this is the output of aplay -L:

> pi@raspberrypi:~ $ arecord -L
> null
>     Discard all samples (playback) or generate zero samples (capture)
> default
> output
> sysdefault:CARD=CODEC
>     USB Audio CODEC, USB Audio
>     Default Audio Device
> front:CARD=CODEC,DEV=0
>     USB Audio CODEC, USB Audio
>     Front speakers
> surround21:CARD=CODEC,DEV=0
>     USB Audio CODEC, USB Audio
>     2.1 Surround output to Front and Subwoofer speakers
> surround40:CARD=CODEC,DEV=0
>     USB Audio CODEC, USB Audio
>     4.0 Surround output to Front and Rear speakers
> surround41:CARD=CODEC,DEV=0
>     USB Audio CODEC, USB Audio
>     4.1 Surround output to Front, Rear and Subwoofer speakers
> surround50:CARD=CODEC,DEV=0
>     USB Audio CODEC, USB Audio
>     5.0 Surround output to Front, Center and Rear speakers
> surround51:CARD=CODEC,DEV=0
>     USB Audio CODEC, USB Audio
>     5.1 Surround output to Front, Center, Rear and Subwoofer speakers
> surround71:CARD=CODEC,DEV=0
>     USB Audio CODEC, USB Audio
>     7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
> iec958:CARD=CODEC,DEV=0
>     USB Audio CODEC, USB Audio
>     IEC958 (S/PDIF) Digital Audio Output
> dmix:CARD=CODEC,DEV=0
>     USB Audio CODEC, USB Audio
>     Direct sample mixing device
> dsnoop:CARD=CODEC,DEV=0
>     USB Audio CODEC, USB Audio
>     Direct sample snooping device
> hw:CARD=CODEC,DEV=0
>     USB Audio CODEC, USB Audio
>     Direct hardware device without any conversions
> plughw:CARD=CODEC,DEV=0
>     USB Audio CODEC, USB Audio
>     Hardware device with all software conversions
>