alexa-pi/AlexaPi #289: Alexa will not respond after 3 attempts [NEW]

AustinErben, 2017-07-26:
**Your OS (including version) where you are running AlexaPi:**
Ubuntu Mate 16.04

**Your hardware platform and model you are running on:**
Raspberry Pi 3 Model B

```

**Python release (`python2 --version`):**
Python 2.7.12

```

**Description of problem:**
Alexa answers and responds twice and then on third attempt does not respond. We then check on Alexa's status in the terminal and when we close the terminal she will respond to the question and work for another two questions.

```

**Expected:**
She can continually answer questions without use of the terminal or rebooting

```

**Problem-relevant `config.yaml` entries:**

  GNU nano 2.5.3                  File: /etc/opt/AlexaPi/config.yaml                                            

# See https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging
# to get a better idea of how to set things in here

# Level of log output
# default: INFO
# set to DEBUG for verbose logging
logging: INFO

sound:
  # Name of your ALSA microphone/soundcard in arecord -L
  input_device: "pulse"
  output: "pulseaudio"
  output_device: ""
  # This is here if none of the listed device: "raspberrypi"
  # This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false

  playback_handler: "sox"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "alsa"
  # output device: "raspberrypi"
  # currently supports ALSA only
  output_device: "default"

  # if left empty, won't be passed to the underlying library / command if it supports it
  # if it doesn't 100% will be used
  default_volume: 80

```

**Steps to reproduce:**

1.  Alexa starts at boot
2.  Ask 2 questions, get 2 answers
3. Ask third question, get no response

**Traceback (if applicable):**
n/a

```

**Additional info:**
We have ran PA system wide and have already ran all audio setups and configured our config.yaml file.

```

EmerickH, 2017-07-27:
Hello,
First you have twice `output:` and `output_device:` in your config, delete one:
`output: "pulseaudio"`
`output_device: "default"`
~~`output: "alsa"`~~
~~`output_device: ""`~~
And restart AlexaPi:
```
sudo systemctl restart AlexaPi.service
```

If it's not working, please provide the complete log:
```
sudo systemctl stop AlexaPi.service
/opt/AlexaPi/src/main.py -d
```

Emerick

AustinErben, 2017-07-31:
Hello,

Thank you for your response. Now she seems to be a bit more random. She will respond to some questions and not others. Sometimes I can ask her more questions before she stops responding and other times I can get nothing or maybe one to two questions. 

Here is the complete log, thanks again. Let me know if there is anything else you need.

-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7.000000e-29
-fwdtree		yes		yes
-hmm					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us
-input_endian		little		little
-jsgf					
-keyphrase				alexa
-kws					
-kws_delay		10		10
-kws_plp		1e-1		1.000000e-01
-kws_threshold		1		1.000000e-10
-latsize		5000		5000
-lda					
-ldadim			0		0
-lifter			0		22
-lm					
-lmctl					
-lmname					
-logbase		1.0001		1.000100e+00
-logfn					
-logspec		no		no
-lowerf			133.33334	1.300000e+02
-lpbeam			1e-40		1.000000e-40
-lponlybeam		7e-29		7.000000e-29
-lw			6.5		6.500000e+00
-maxhmmpf		30000		30000
-maxwpf			-1		-1
-mdef					
-mean					
-mfclogdir				
-min_endfr		0		0
-mixw					
-mixwfloor		0.0000001	1.000000e-07
-mllr					
-mmap			yes		yes
-ncep			13		13
-nfft			512		512
-nfilt			40		25
[... 308 lines cut ...]
INFO: dict.c(333): Reading main dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
INFO: dict.c(213): Dictionary size 134723, allocated 1016 KiB for strings, 1679 KiB for phones
INFO: dict.c(336): 134723 words read
INFO: dict.c(358): Reading filler dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/noisedict
INFO: dict.c(213): Dictionary size 134728, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -225, delay 10)
2017-07-30 19:26:53 DEBUG: Setting up playback handler: SoxHandler
2017-07-30 19:26:53 INFO: Checking Internet Connection ...
2017-07-30 19:26:53 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-07-30 19:26:54 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2017-07-30 19:26:54 INFO: Connection OK
2017-07-30 19:26:54 INFO: AVS token: Requesting a new one
2017-07-30 19:26:54 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-07-30 19:26:54 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 978
2017-07-30 19:26:54 INFO: AVS token: Obtained successfully
2017-07-30 19:26:54 DEBUG: Stopping audio play
2017-07-30 19:26:54 DEBUG: Stopped play. No SoX process to stop
2017-07-30 19:26:54 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-07-30 19:26:54 DEBUG: Started play. sox -q /opt/AlexaPi/src/resources/hello.mp3 -t pulseaudio vol -6 dB pad 0 0
2017-07-30 19:26:57 DEBUG: Finished play.

renekliment, 2017-10-14:
It's probably the tricky voice triggering. 
Having better mic / settings is one way to go. Trying snowboy is the other.

Closing due to inactivity.

closed 2017-10-14 by renekliment