alexa-pi/AlexaPi #290: Button is not triggering Alexa

newm-newm, 2017-07-30:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**
```
-- Logs begin at Sat 2017-07-29 20:13:15 PDT, end at Sat 2017-07-29 20:24:59 PDT. --
Jul 29 20:13:19 Bacon321 systemd[1]: Starting Alexa client for all your devices...
Jul 29 20:13:19 Bacon321 systemd[1]: Started Alexa client for all your devices.
Jul 29 20:13:28 Bacon321 python[531]: INFO: Checking Internet Connection ...
Jul 29 20:13:28 Bacon321 python[531]: INFO: Connection OK
Jul 29 20:13:28 Bacon321 python[531]: INFO: AVS token: Requesting a new one
Jul 29 20:13:29 Bacon321 python[531]: INFO: AVS token: Obtained successfully
```

**Your OS (including version) where you are running AlexaPi:**
```
```Linux Bacon321 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux

**Your hardware platform and model you are running on:**
```
raspberryPi 3
```

**Python release (`python2 --version`):**
```
Python 2.7.9

```

**Description of problem:**
```
I installed a button and it is not triggering Alexa
```

**Expected:**
```
Alexa should trigger
```

**Problem-relevant `config.yaml` entries:**
```

triggers:
  platform:
    enabled: true
    voice_confirm: false

    # one of: oneshot-vad, continuous, continuous-vad
    event_type: "oneshot-vad"

    # only for "continuous" event_types
    long_press:
      # system command to run after pressing the button for _duration_ number of seconds
      # empty to disable
      command: ""

      # duration in seconds
      duration: 10

      # CURRENTLY DISABLED
      # optional audio file to play before executing the command
      # you might wanna use full path to make sure AlexaPi finds it
      # you can use {resources_path} for the path to the AlexaPi "resources" directory
      audio_file: ""

  pocketsphinx:
    enabled: false
    voice_confirm: false

    phrase: "alexa"
    threshold: 1e-10

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

newm-newm, 2017-07-30:
I've actually got the button working but now neither of the LEDs are working.

asktoby, 2017-08-27:
I wonder if you made the same mistake I did.
I was having the same issue.
Turned out I hadn't realised that GPIO 18 is not the same as pin 18.
**GPIO 18 is pin 12.**
So you have to connect pin 12 (GPIO18) to pin 6 (GND), not pin 18 to pin 6.

newm-newm, 2017-08-29:
Thank you! But I've actually figured it out and got it working! The issue was actually (difficult to admit), that the LEDs were dead. Should have tested them but I wasn't at all familiar with electronics. Bought some brand new LEDs and I actually got a different button as the button stopped working shortly after that post. Plugged it all in to the correct pin headers and it worked perfectly!

renekliment, 2017-10-08:
Seems resolved.

closed 2017-10-08 by renekliment