alexa-pi/AlexaPi #291: Installation script doesn't work

gingermcninja, 2017-08-04:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
Raspbian (latest)
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi 3
```

**Python release (`python2 --version`):**
```
Python 2.7.9
```

**Description of problem:**
```
Running the setup.sh script results in a number of 403 errors when accessing http://dl.bintray.com. This breaks the installation.
```

**Expected:**
```
The installation works without the 403 errors
```

**Problem-relevant `config.yaml` entries:**
```
config.yaml isn't generated
```

**Steps to reproduce:**

1. Run the setup script

**Traceback (if applicable):**
```
pi@retropie:/opt $ sudo ./AlexaPi/src/scripts/setup.sh 
TIP: When there is a value in brackets like [default_value], hit Enter to use it.

Which operating system are you using?
         debian - Debian, Raspbian, Armbian, Ubuntu or other Debian-based
      archlinux - Arch Linux or Arch Linux-based
Your OS [debian]: 
Which device are you using?
           chip - C.H.I.P.
       hyperion - Integrate with Hyperion Ambient Lightning Software
    magicmirror - Integrating with Magic Mirror project (MMM-AlexaPi)
       orangepi - Orange Pi or another H3 based board
          other - other SBCs, desktops, or anything else
    raspberrypi - all Raspberry Pi variants
         serial - an Arduino controled device for example; used for Teddy Ruxpin
Your device [raspberrypi]: 
Do you want AlexaPi to run on boot?
You have these options: 
0 - NO
1 - yes, use systemd (default, RECOMMENDED and awesome)
2 - yes, use a classic init script (for a very old PC or an embedded system)
Which option do you prefer? [1]: 
Creating a user to run AlexaPi under ... user already exists. That's cool - using that.
gpio:x:997:pi,alexapi
audio:x:29:pi,kodi,alexapi
Would you like to also install Airplay support (Y/n)? n
Ign http://dl.bintray.com ./ InRelease
Ign http://dl.bintray.com ./ InRelease                                                                                                           
Hit http://archive.raspberrypi.org jessie InRelease                                                                                              
Hit http://mirrordirector.raspbian.org jessie InRelease
Ign http://dl.bintray.com ./ Release.gpg       
Ign http://dl.bintray.com ./ Release.gpg          
Ign http://dl.bintray.com ./ Release              
Ign http://dl.bintray.com ./ Release                                                             
Hit http://archive.raspberrypi.org jessie/main armhf Packages                                                       
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages                                                   
Hit http://archive.raspberrypi.org jessie/ui armhf Packages
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages                                                                
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages                                                               
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages                                              
Err http://dl.bintray.com ./ Packages                                                                                                   
  403  Forbidden
Ign http://dl.bintray.com ./ Translation-en_GB                                                                                          
Ign http://dl.bintray.com ./ Translation-en                                                                       
Err http://dl.bintray.com ./ Packages                                                                             
  403  Forbidden
Ign http://dl.bintray.com ./ Translation-en_GB                                                                    
Ign http://dl.bintray.com ./ Translation-en                                                                       
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB                                                  
Ign http://archive.raspberrypi.org jessie/main Translation-en                                                     
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
W: Failed to fetch http://dl.bintray.com/pipplware/dists/jessie/main/binary/./Packages  403  Forbidden                                                      

W: Failed to fetch http://dl.bintray.com/pipplware/dists/jessie/armv7/binary/./Packages  403  Forbidden

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

**Additional info:**
```

```

maso27, 2017-08-04:
This is not a problem with the script.
There is something wrong with your update profile.

Try:
`sudo apt-get update`
Then
`sudo apt-get upgrade`

And see if the problem exists there.

renekliment, 2017-08-12:
Yup, not a problem w/ AlexaPi.

closed 2017-08-12 by renekliment