alexa-pi/AlexaPi #294: raspberrypi python[7072]: ERROR: URLError: Not Found

rites1978, 2017-08-05:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
```

**Your hardware platform and model you are running on:**
```

```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
```

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

OutsourcedGuru, 2017-08-28:
I'm guessing that someone would want you to fill in the information in the reporting template.

dikkedimi, 2017-10-04:
I'm having the same error...

`Oct 04 23:33:19 raspberrypi python[32239]: INFO: Magic Mirror HTTP Server - :5055
Oct 04 23:33:19 raspberrypi python[32239]: INFO: Checking Internet Connection ...
Oct 04 23:33:20 raspberrypi python[32239]: INFO: Connection OK
Oct 04 23:33:20 raspberrypi python[32239]: INFO: AVS token: Requesting a new one
Oct 04 23:33:20 raspberrypi python[32239]: INFO: AVS token: Obtained successfully
Oct 04 23:33:20 raspberrypi python[32239]: ERROR: URLError: Not Found
Oct 04 23:33:20 raspberrypi python[32239]: Exception in thread Thread-1:
Oct 04 23:33:20 raspberrypi python[32239]: Traceback (most recent call last):
Oct 04 23:33:20 raspberrypi python[32239]:   File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
Oct 04 23:33:20 raspberrypi python[32239]:     self.run()
Oct 04 23:33:20 raspberrypi python[32239]:   File "/usr/lib/python2.7/threading.py", line 754, in run
Oct 04 23:33:20 raspberrypi python[32239]:     self.__target(*self.__args, **self.__kwargs)
Oct 04 23:33:20 raspberrypi python[32239]:   File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 55, in thread
Oct 04 23:33:20 raspberrypi python[32239]:     inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, self._config['sound']['input_device'])
Oct 04 23:33:20 raspberrypi python[32239]: ALSAAudioError: Device or resource busy [iec958:CARD=C920,DEV=0]
Oct 04 23:33:21 raspberrypi python[32239]: ERROR: URLError: Not Found`

renekliment, 2017-10-08:
`URLError: Not Found` is basically a warning when you don't have MagicMirror configured properly.

`raspberrypi python[32239]: ALSAAudioError: Device or resource busy [iec958:CARD=C920,DEV=0]`
Obviously, the audio isn't set up properly. Read up on that (our Wiki and google Linux audio). Try now with AlexaPi 1.6 where we use PyAudio now. If that doesn't help, set up PulseAudio for god's sake.

closed 2017-10-08 by renekliment