alexa-pi/AlexaPi #295: Question about WakeWord

DJS2017, 2017-08-07:
Dear friend, I am wondering if your github has implemented WakeWord (automatically detected) or is this more like a push a button and talk?

The WakeWord Agent that I am referring to is [here](https://github.com/alexa/alexa-avs-sample-app/tree/master/samples/wakeWordAgent/src).

If this part has been implemented, is it possible to tell me which python program has it, and does it stream onto a buffer or save the audio into a .wav file? Thanks!

renekliment, 2017-08-09:
https://github.com/alexa-pi/AlexaPi/wiki/Triggers

closed 2017-08-09 by renekliment

DJS2017, 2017-08-09:
Thanks for your reply.

After wake word triggers , does it call capture.silence to record the audio?

https://github.com/alexa-pi/AlexaPi/wiki/Triggers